openwrt/openwrt #15741: kernel memory corruption on Turris Omnia

M95D, 2024-06-18:
### Describe the bug

When compat.ko, mac80211.ko, cfg80211.ko, and preferably ath*.ko are loaded, various memory corruption errors occur randomly, sometimes crashing the kernel completely.

I found a somewhat reliable way to reproduce the crash, but it doesn't always work if ath modules are not loaded. However, some errors sometimes appear with only compat, mac and cfg modules loaded.

Bug present in kernel v6.6 (snapshot) and v6.1.
I can't reproduce it with v5.15.

### OpenWrt version

r26290+30-dc9220f748

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Turris Omnia

### Image kind

Official downloaded image

### Steps to reproduce

1) Download and install the official snapshot image for Turris Omnia.
2) Enable wireless by removing `disabled 1` lines from `/etc/config/wireless`
3) Install stress-ng package.
4) Remove all modules from /lib/modules/<version>, except compat.ko, cfg80211.ko, mac80211.ko and ath*.ko.
5) Reboot
6) Run `stress-ng --abort --klog-check --timeout 10m --bigheap 2 --bigheap-bytes 800M`

My Omnia has 2GB RAM with aprox. 1.9GB available after boot. Adjust bytes according to your own router to avoid OOM.
The test will cause a memory error in over 90% of cases within 1 minute, but sometimes it takes longer.

### Actual behaviour

A less serious error without ath driver loaded. A second test run without rebooting completed without error.
```
root@GRAPHRT:/# ls /lib/modules/6.6.33
drwxr-xr-x    3 root     root          3488 2024-06-18 11:23:35 +0300 .
drwxr-xr-x    1 root     root          3488 2024-06-18 10:25:19 +0300 ..
drwxrwxrwx    2 root     root          3488 2024-06-18 11:23:35 +0300 bak
-rw-r--r--    1 root     root        364604 2024-06-17 01:10:53 +0300 cfg80211.ko
-rw-r--r--    1 root     root          3176 2024-06-17 01:10:53 +0300 compat.ko
-rw-r--r--    1 root     root        760688 2024-06-17 01:10:53 +0300 mac80211.ko
root@GRAPHRT:/# stress-ng --abort --klog-check --timeout 10m --bigheap 2 --bighe
ap-bytes 800M
stress-ng: info:  [3765] setting to a 10 mins, 0 secs run per stressor
stress-ng: info:  [3765] dispatching hogs: 2 bigheap
[  680.460676] BUG: Bad rss-counter state mm:0208c878 type:MM_FILEPAGES val:-13
[  680.467806] BUG: Bad rss-counter state mm:0208c878 type:MM_ANONPAGES val:13
stress-ng: info:  [3766] klog-check: kernel cmdline: 'console=ttyS0,115200 root=/dev/mmcblk0p2'
stress-ng: info:  [3766] proc: root      3765  1049 S stress-ng --abort --klog-check --timeout 10m --bigheap 2 --bigheap-bytes 800M
stress-ng: info:  [3766] proc: root      3766  3765 R stress-ng-klog [monitoring]
stress-ng: info:  [3766] proc: root      3767  3765 S stress-ng-bigheap [wait]
stress-ng: info:  [3766] proc: root      3769  3767 R stress-ng-bigheap [deinit]
stress-ng: error: [3766] klog-check: alert: [680.460676] 'BUG: Bad rss-counter state mm:0208c878 type:MM_FILEPAGES val:-13'
stress-ng: error: [3766] klog-check: alert: [680.467806] 'BUG: Bad rss-counter state mm:0208c878 type:MM_ANONPAGES val:13'
[  680.627052] BUG: Bad rss-counter state mm:e3c0ed4e type:MM_FILEPAGES val:-16
[  680.634160] BUG: Bad rss-counter state mm:e3c0ed4e type:MM_ANONPAGES val:16
stress-ng: error: [3766] klog-check: alert: [680.627052] 'BUG: Bad rss-counter state mm:e3c0ed4e type:MM_FILEPAGES val:-16'
stress-ng: error: [3766] klog-check: alert: [680.634160] 'BUG: Bad rss-counter state mm:e3c0ed4e type:MM_ANONPAGES val:16'
stress-ng: info:  [3765] skipped: 0
[... 157 lines cut ...]
[ 3003.666364]  ipi_handler from handle_percpu_devid_irq+0x78/0x134
[ 3003.666377]  handle_percpu_devid_irq from generic_handle_domain_irq+0x28/0x38
[ 3003.666388]  generic_handle_domain_irq from gic_handle_irq+0x74/0x88
[ 3003.666395]  gic_handle_irq from generic_handle_arch_irq+0x34/0x44
[ 3003.666407]  generic_handle_arch_irq from call_with_stack+0x18/0x20
[ 3003.666418]  call_with_stack from __irq_svc+0x80/0x98
[ 3003.666424] Exception stack(0xc0e01f30 to 0xc0e01f78)
[ 3003.666429] 1f20:                                     00000003 00000001 01774d9c 40000000
[ 3003.666433] 1f40: c0e04f68 c0e08600 00000000 c0e04fa4 00000000 00000000 c0e04f08 00000000
[ 3003.666437] 1f60: c10a8838 c0e01f80 c096017c c0960890 60000013 ffffffff
[ 3003.666440]  __irq_svc from default_idle_call+0x2c/0x30
[ 3003.666450]  default_idle_call from do_idle+0x188/0x1d8
[ 3003.666466]  do_idle from cpu_startup_entry+0x28/0x2c
[ 3003.666477]  cpu_startup_entry from kernel_init+0x0/0x12c
[ 3003.666489]  kernel_init from arch_post_acpi_subsys_init+0x0/0x8
[ 3003.795382] Rebooting in 3 seconds..
```

### Additional info

A kernel with `memtest=17` doesn't find anything wrong with the RAM. I ran it multiple times.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

M95D, 2024-06-18:
PS:
Is this right for kernel v6.6.33?
`Loading modules backported from Linux version v6.6.15-0-g51f354b815c4`

mrkiko, 2024-06-18:
Mark to follow.

M95D, 2024-11-10:
I tried the new 10 LEDs recovery - RAM compatibility mode - which reduces RAM speed to 1333 MHz and it seems to resolve the issues.

closed 2024-11-10 by M95D