openwrt/openwrt #15759: “WARNING: your configuration is out of sync. Please run make menuconfig, oldconfig or defconfig!” on every build when using `make defconfig`

markmentovai, 2024-06-19:
### Describe the bug

On every build past 9a52ec4fa092 (@Ansuel as part of #15543), this warning is produced on each build:

```
WARNING: your configuration is out of sync. Please run make menuconfig, oldconfig or defconfig!
```

This is even happening for official builds on [buildbot.openwrt.org](https://buildbot.openwrt.org/). For example, [x86/generic](https://buildbot.openwrt.org/images/#/builders/38) [build 269](https://buildbot.openwrt.org/images/#/builders/38/builds/269) shows this on many steps starting with [(28) cleanbase](https://buildbot.openwrt.org/images/#/builders/38/builds/269/steps/28/logs/stdio) and persisting in others like [(29) tools](https://buildbot.openwrt.org/images/#/builders/38/builds/269/steps/29/logs/stdio), [(30) toolchain](https://buildbot.openwrt.org/images/#/builders/38/builds/269/steps/30/logs/stdio), [(31) kmods](https://buildbot.openwrt.org/images/#/builders/38/builds/269/steps/31/logs/stdio), etc.

I see this although I have properly generated a `.config` with `make defconfig`, as the buildbots do ([(17) newconfig](https://buildbot.openwrt.org/images/#/builders/38/builds/269/steps/17/logs/stdio), [(18) defconfig](https://buildbot.openwrt.org/images/#/builders/38/builds/269/steps/18/logs/stdio)):

```
% git clone https://git.openwrt.org/openwrt/openwrt.git openwrt
% cd openwrt
% cat > .config << __EOF__
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
__EOF__
% make defconfig
% make
[…]
make[2]: Entering directory '…/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '…/openwrt/scripts/config'
make[4]: Entering directory '…/openwrt/target/linux'
make[5]: Entering directory '…/openwrt/target/linux/x86'
make[6]: Entering directory '…/openwrt/target/linux/x86/image'
make[6]: Leaving directory '…/openwrt/target/linux/x86/image'
make[5]: Leaving directory '…/openwrt/target/linux/x86'
make[4]: Leaving directory '…/openwrt/target/linux'
time: target/linux/prereq#0.06#0.07#0.15
WARNING: your configuration is out of sync. Please run make menuconfig, oldconfig or defconfig!
 make[1] world
^C
```

If I run `make menuconfig`, it “fixes” the problem, and shows a diff in `.config`:

```
% cp .config .config.def
% make menuconfig  # Don’t make changes, just exit and save
% diff -u .config.def .config
--- .config.def	2024-06-19 11:25:46
+++ .config	2024-06-19 11:26:35
@@ -100,6 +100,7 @@
 CONFIG_DEFAULT_ppp=y
 CONFIG_DEFAULT_ppp-mod-pppoe=y
 CONFIG_DEFAULT_procd=y
+CONFIG_DEFAULT_procd-seccomp=y
 CONFIG_DEFAULT_procd-ujail=y
 CONFIG_DEFAULT_uci=y
 CONFIG_DEFAULT_uclient-fetch=y
```

`make defconfig` alone is producing a `.config` that doesn’t contain `CONFIG_DEFAULT_procd-seccomp` and that triggers these warnings.

If I revert 381021944135 and 9a52ec4fa092, I am able to use `make defconfig` to produce a `.config` that does not trigger these warnings. The generated `.config` is identical to the one that `make defconfig` produces past 9a52ec4fa092: it also doesn’t contain `CONFIG_DEFAULT_procd-seccomp`. However in that pre-9a52ec4fa092 case, the `.config` produced by `make menuconfig` also doesn’t contain `CONFIG_DEFAULT_procd-seccomp`.

I see this when building for various targets, including x86/64, [ipq806x/generic](https://buildbot.openwrt.org/images/#/builders/185/builds/267/steps/28/logs/stdio), [ipq40xx/mikrotik](https://buildbot.openwrt.org/images/#/builders/234/builds/271/steps/28/logs/stdio), and [ath79/generic](https://buildbot.openwrt.org/images/#/builders/219/builds/257/steps/28/logs/stdio).

### OpenWrt version

r26710-3b83d393ddcd

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Steps to reproduce

```
% git clone https://git.openwrt.org/openwrt/openwrt.git openwrt
% cd openwrt
% cat > .config << __EOF__
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
__EOF__
% make defconfig
% make
```

### Actual behaviour

This warning is produced early during `make`:

```
WARNING: your configuration is out of sync. Please run make menuconfig, oldconfig or defconfig!
```

### Expected behaviour

`make defconfig` was used to create a valid `.config`. The warning about the configuration being out of sync should not appear.

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_MOLD=y
CONFIG_MOLD=y
# CONFIG_USE_MOLD is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-06-19:
https://github.com/openwrt/openwrt/commit/847fad476f3d3652e75daa98822662f654bfd225 should fix it

markmentovai, 2024-06-19:
https://github.com/openwrt/openwrt/issues/15759#issuecomment-2179117302:
> 847fad476f3d3652e75daa98822662f654bfd225 should fix it

I synced to that revision and I’m still seeing the warnings.

CharlesMengCA, 2024-07-10:
Same warning when built for Linksys E8450 on Arch Linux.

Spudz76, 2024-07-11:
"Me too" on ramips/mt7621 (Linksys EA7300v1, `mipsel`/`mipsel_24kc`) although I did not do all the tracing as in OP, `menuconfig` and exit/save also fixes it.  Menuconfig adds the same missing line (below).
``` diff
+CONFIG_DEFAULT_procd-seccomp=y

```
Unsure why https://github.com/openwrt/openwrt/commit/847fad476f3d3652e75daa98822662f654bfd225 didn't fix it.