openwrt/openwrt #15766: Banana Pi R4 RTC not work

MashiroYae, 2024-06-21:
### Describe the bug

```
root@OpenWrt:~#  dmesg | grep rtc
[    3.899885] FIT:          flat_dt sub-image 0x00599000..0x0059911c "fdt-mt7988a-bananapi-bpi-r4-rtc" (ARM64 OpenWrt bananapi_bpi-r4 device tree overlay mt7988a-bananapi-bpi-r4-rtc)
root@OpenWrt:~# hwclock --verbose
hwclock from util-linux 2.39.3
System Time: 1718961079.062589
Trying to open: /dev/rtc0
Trying to open: /dev/rtc
Trying to open: /dev/misc/rtc
No usable clock interface found.
hwclock: Cannot access the Hardware Clock via any known method.

```

### OpenWrt version

r26733-2b6772c82c

### OpenWrt release

snapshots

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kvic-z, 2024-06-29:
The RTC interface seems disabled on purpose. Even if the interface is enabled, it still doesn't work for me in Linux. However, I'm quite sure the RTC works offline. For example, if you disable ntp in OpenWrt, and unplug BPI R4 for an hour and then power on, the time is still current. If you want to try the RTC interface yourself, you can apply this patch and recompile:

````
--- a/target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4.dtsi
+++ b/target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4.dtsi
@@ -251,7 +251,7 @@
                        pcf8563: rtc@51 {
                                compatible = "nxp,pcf8563";
                                reg = <0x51>;
-                               status = "disabled";
+                               status = "okay";
                        };
                };

````

Hope we will figure out or understand why the interface doesn't work in Linux.

kvic-z, 2024-06-29:
A better fix should be in 'target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4-rtc.dtso'. Somehow the existing code seems not able to enable the device:
````
fragment@0 {
     target = <&pcf8563>;
     __overlay__ {
       status = "okay";
     };
};
````

kvic-z, 2024-06-29:
Okay. I think understand it now. The root cause is that there is actually no issue. By design, the RTC is an opt-in feature to be enabled by user. I guess developers expect users to install a battery to ensure the RTC works properly. And the proper way to enable the RTC is inside OpenWrt and issue the following command:

````
$ fw_setenv bootconf_extra mt7988a-bananapi-bpi-r4-rtc
````

Reboot. Then you'll see the RTC at /dev/rtc0.

closed 2024-06-30 by MashiroYae