openwrt/openwrt #15775: ttyS ttyS0: 1 input overrun(s)

csjtl, 2024-06-22:
### Describe the bug

`ttyS ttyS0: 1 input overrun(s)`
Keyboard not working after entering openwrt
I don't know if it's a problem with my device. I have an img file from a long time ago and it works fine.

### OpenWrt version

r26742-d648ee4c58

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Gigabyte Technology Co., Ltd. H77N-WIFI

### Image kind

Self-built image

### Steps to reproduce

Use the serial port device to start openwrt on the x86 host. The up and down keys on the keyboard are selected incorrectly in the gnu interface. After entering the openwrt system, it cycles repeatedly. Using securecrt and Ubuntu's minicom have the same effect. At the same time, the keyboard does not work after entering openwrt. The monitor shows `ttyS ttyS0: 1 input overrun(s)`. ssh works normally.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-06-22:
Just press repeatedly same character to resync terminal, like ctrl-d to log off and r#set it

csjtl, 2024-06-23:
https://github.com/openwrt/openwrt/assets/55336802/5dcb85e0-c0c0-40b1-bf08-69564ae49308

Keyboard ↑↓ keys cannot be selected, → keys enter openwrt, the keyboard can not be used. Device external monitor output`ttyS ttyS0: 1 input overrun(s)`

[image]

csjtl, 2024-06-23:
> Just press repeatedly same character to resync terminal, like ctrl-d to log off and r#set it

Thanks, I tried it and it doesn't work

csjtl, 2024-06-30:
error can't find command 'search'

csjtl, 2024-07-01:
kmod-usb-hid

closed 2024-07-01 by csjtl