openwrt/openwrt #15814: serial console speed for Radxa E25 should be 115200bps instead of 1.5Mbps

RadxaNaoki, 2024-06-27:
### Describe the bug

CH340B, USB-UART chip, used on E25 for serial console is unstable with 1.5Mbps.
for example, if you press and hold enter key on the shell, you can see some letters from command prompt are missing.

### OpenWrt version

r0+26800-843b51b26d

### OpenWrt release

r0+26800-843b51b26d

### OpenWrt target/subtarget

rockchip

### Device

Radxa E25 Carrier Board

### Image kind

Self-built image

### Steps to reproduce

1. connect PC USB port and E25 Type-C port
2. access serial console with 1.5Mbps
3. press and hold enter key on the shell to repeat command prompt

### Actual behaviour

some letters are missed

### Expected behaviour

no letter should be missed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

RadxaNaoki, 2024-06-27:
TPL binary also need to be modified to use 115200bps

1715173329, 2024-07-03:
> TPL binary also need to be modified to use 115200bps

However that would be little bit hard to do it on OpenWrt side since it requires proprietary binary tool from rockchip.
Could you add a patched TPL binary in radxa's rkbin repo?

RadxaNaoki, 2024-07-04:
https://github.com/radxa/rkbin/blob/5696fab20dcac57c1458f72dc7604ba60e553adf/bin/rk35/rk3568_ddr_1560MHz_uart2_m0_115200_v1.21.bin

closed 2024-07-10 by openwrt-bot