openwrt/openwrt #15820: Packet Steering in a WRT1200AC 23.05.03

ja16779, 2024-06-27:
Hi team, I have a Linksys WRT1200AC router working as Bridge router, and the Packet Steering does not work, the status is the following

/etc/init.d/packet_steering status
active with no instances

Do i need to perform any additional task?
Thanks

khanjui, 2024-06-27:
It also says active with no instances on my device but I'm pretty sure it's working. Install htop and monitor it while doing a speedtest, turn packet steering off and on to see if there are any differences.

ja16779, 2024-06-27:
whit packet off it is working on do nothing

httpstorm, 2024-07-12:
@ja16779 
I just enabled packet steering out of curiosity, and thought I might as well share the results:
WRT3200ACM send to MBP2019 benchmark using [g-net-speed](https://httpstorm.com/download/g-net-speed/)
Speed in megabytes per second: 96 MB/s = 768 Mbit/s
``` bash
/etc/init.d/packet_steering status
active with no instances # during benchmark

[  1 GB   46 MB][  82 MB/s] 12.9 s [1448-2097152 46886] # steering off, connections 1
[  1 GB    5 MB][  86 MB/s] 11.8 s [416-2097152 38196]  # steering off, connections 2
[  1 GB   33 MB][  84 MB/s] 12.5 s [416-602368 60496]   # steering on, connections 1
[  1 GB   58 MB][  89 MB/s] 12.0 s [416-346072 45335]   # steering on, connections 2
[  1 GB   18 MB][  89 MB/s] 11.7 s [1448-2097152 63247] # steering all CPUs, connections 1
[  1 GB   44 MB][  96 MB/s] 11.0 s [416-1831720 42178]  # steering all CPUs, connections 2
```

brada4, 2024-07-13:
@httpstorm try this https://github.com/openwrt/firewall4/commit/a5553dae70439c7e4fa910490fcf12a1ffff5bd2 - edit 2 lines on system essentially.

httpstorm, 2024-07-13:
@brada4 
Do you want me to compile without your commit?
I forgot to mention I'm on OpenWRT branch main, `firewall4-2024.05.21~4c01d1eb`, which includes this patch. Hardware and software offload is disabled on the firewall, because they didn't make any difference.

brada4, 2024-07-13:
Ok, you get maximum in this case, undoing will dramatically rise input/output chain cpu usage... no overlap with forward/offload in particular test.
Can you try nginx with auto workers auto affinity and reuseport? Especially if it is balanced...

httpstorm, 2024-07-15:
@brada4 
> Ok, you get maximum in this case, undoing will dramatically rise input/output chain cpu usage... no overlap with forward/offload in particular test.

Do I need to have `Software flow offloading` and `Hardware flow offloading` enabled for your patch to make a difference? Currently they are disabled. All benchmarks I've done so far are between my laptop and the router over the 5GHz radio. Do I need to test NAT performance to asses the effect of your patch?

I edited `/usr/share/firewall4/templates/ruleset.uc` directly on the router back to `iifname` and `oifname`, and then `/etc/init.d/firewall restart` I didn't notice any difference. Then I restarted the router, still no difference. Either I need to pay close attention to the CPU usage, which is hard when it is spread over two cores, or the CPU is powerful enough to keep up with the radio, so it makes no difference.

Here is `htop` on `r26916-0ff2d0070e` WRT3200ACM, 768 Mbit/s router to MBP2019, steering all CPUs, connections 2
Note: sometimes the load on the cores is not symmetrical
``` bash
    0[||||||||||||||||||||||||                          41.8%] Tasks: 28, 132 thr, 82 kthr; 0 running
    1[|||||||||||||||||||||||                           41.2%] Load average: 0.59 0.57 0.30 
  Mem[||||||||||||||||||||||||||||||||              208M/494M] Uptime: 00:16:28
```

> Can you try nginx with auto workers auto affinity and reuseport? Especially if it is balanced...

I'm not sure if `nginx` will compile on macOS host. Does it work well as a replacement to `uhttpd` as the router web interface? On the other hand I'm curious to compare the performance between the two web servers.
I can make tests with the [httpstorm.com](https://httpstorm.com) web server, and [g-net-speed](https://httpstorm.com/download/g-net-speed/).
Note: the throughput depends on the size of content being served, in this case index.htm is 4694 bytes.
``` bash
# HTTP, 2 connections, queue 8 requests, GET /
gns 192.168.192.11 83 -c2 -g 192.168.192.11 / 8
:: client_00.working  active 01  RAW
:: client_01.working  active 02  RAW
[  9 GB  898 MB][  64 MB/s]  2:35.8 mm:ss [ 24-2097152 22221]

    0[|||||||||||||||||||||||||||||||||||||||||||||||||100.0%] Tasks: 32, 132 thr, 82 kthr; 0 running
    1[|||||||||||||||||||||||||||||||||||||||||||||||||100.0%] Load average: 2.62 1.65 0.98 
  Mem[|||||||||||||||||||||||||||||||               206M/494M] Uptime: 00:44:27

  PID USER       PRI  NI  VIRT   RES   SHR S  CPU%-MEM%   TIME+  Command
 3567 root        20   0 33636  6844  3968 R  57.7  1.4  1:04.54 /usr/sbin/httpstorm -d:httpstorm
 3569 root        20   0 33636  6844  3968 R  57.1  1.4  1:04.30 /usr/sbin/httpstorm -d:httpstorm
 3568 nobody      20   0 33636  6844  3968 R  40.4  1.4  0:46.10 /usr/sbin/httpstorm -d:httpstorm
 3570 nobody      20   0 33636  6844  3968 S  40.4  1.4  0:45.97 /usr/sbin/httpstorm -d:httpstorm
```

``` bash
# HTTPS, 2 connections, queue 8 requests, GET /
gns 192.168.192.11 83 -c2 -g 192.168.192.11 / -t
OpenSSL 3.3.1 4 Jun 2024
:: client_00.working  active 01  TLS
:: client_01.working  active 02  TLS
[  9 GB  288 MB][  62 MB/s]  2:32.5 mm:ss [4990-4990 4990]

    0[|||||||||||||||||||||||||||||||||||||||||||||||||100.0%] Tasks: 32, 132 thr, 82 kthr; 0 running
    1[|||||||||||||||||||||||||||||||||||||||||||||||||100.0%] Load average: 2.18 0.92 0.62 
  Mem[|||||||||||||||||||||||||||||||               204M/494M] Uptime: 00:40:35

  PID USER       PRI  NI  VIRT   RES   SHR S  CPU%-MEM%   TIME+  Command
 3548 nobody      20   0 33908  6588  3968 R  70.3  1.3  1:05.52 /usr/sbin/httpstorm -d:httpstorm
 3550 nobody      20   0 33908  6588  3968 R  70.3  1.3  1:05.72 /usr/sbin/httpstorm -d:httpstorm
 3549 root        20   0 33908  6588  3968 R  28.4  1.3  0:26.36 /usr/sbin/httpstorm -d:httpstorm
 3547 root        20   0 33908  6588  3968 S  27.7  1.3  0:26.23 /usr/sbin/httpstorm -d:httpstorm
```

brada4, 2024-07-15:
Try to add cpu detail like usermode/irq/softirq split CPU usage

httpstorm, 2024-07-15:
@brada4 
Alas they all report NAN.
Is there any particular package that needs to be compiled for this information to be available in `htop`?
```
PSI full IRQ:      nan%   nan%   nan% 
```

brada4, 2024-07-15:
It is option  "detailed cpu time (...)" after htop f2-setup

httpstorm, 2024-07-15:
@brada4 
Thanks for the clarification. I enabled it, but I don't see any difference on the main screen. Would you like to provide your configuration file for `htop`? 
```
[x]    Detailed CPU time (System/IO-Wait/Hard-IRQ/Soft-IRQ/Steal/Guest)

  PID USER       PRI  NI  VIRT   RES   SHR S  CPU%-MEM%   TIME+  Command
 6360 root        20   0  3520  3072   896 R   1.3  0.6  0:00.24 htop
```

brada4, 2024-07-15:
```
hide_kernel_threads=0
detailed_cpu_time=1
```

httpstorm, 2024-07-15:
@brada4 
So what do my results show?

``` bash
# with patch
gns 192.168.192.11 81
:: client_00.working  active 01  RAW
[ 10 GB  977 MB][  88 MB/s]  2:06.6 mm:ss [416-2097152 61465]

    0[||||||||||||||||                                  27.9%] Tasks: 30, 132 thr, 80 kthr; 0 running
    1[|||||||||||||||||||||||||||                       46.8%] Load average: 0.58 0.59 0.37 
  Mem[|||||||||||||||||||||||||||||||               203M/494M] Uptime: 03:50:14

  PID USER       PRI  NI  VIRT   RES   SHR S  CPU%-MEM%   TIME+  Command
 6603 root        20   0 15736  5120  3840 R  20.8  1.0  0:22.92 gns -lp 81 -w
   20 root        20   0     0     0     0 R  13.0  0.0  1:39.77 ksoftirqd/1
 6602 root        20   0 15736  5120  3840 R   2.6  1.0  0:03.12 gns -lp 81 -w
 6500 root        20   0     0     0     0 I   1.9  0.0  0:13.03 kworker/u4:2-phy1
 6658 root        20   0  2532  2176   896 R   1.3  0.4  0:13.40 htop
   14 root        20   0     0     0     0 S   0.6  0.0  3:45.86 ksoftirqd/0
 2949 network     20   0  6404  4992  4096 S   0.6  1.0  0:27.67 /usr/sbin/hostapd -s -g /var/run/hostapd/global
 3801 root        20   0  6448  5376  4608 S   0.6  1.1  0:54.71 /usr/sbin/openvpn --syslog openvpn(g_client_tu) --status /va
 6671 root        20   0     0     0     0 R   0.6  0.0  0:01.39 kworker/u4:0+phy1
    1 root        20   0  1384   768   640 S   0.0  0.2  0:03.28 /sbin/procd
    2 root        20   0     0     0     0 S   0.0  0.0  0:00.00 kthreadd
    3 root        20   0     0     0     0 S   0.0  0.0  0:00.00 pool_workqueue_release
    4 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-rcu_g
    5 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-rcu_p
    6 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-slub_
    7 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-netns
    9 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/0:0H-events_highpri
   12 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-mm_pe
   13 root        20   0     0     0     0 I   0.0  0.0  0:00.00 rcu_tasks_trace_kthread
   15 root        20   0     0     0     0 I   0.0  0.0  0:02.91 rcu_sched
   16 root        RT   0     0     0     0 S   0.0  0.0  0:00.00 migration/0
   17 root        20   0     0     0     0 S   0.0  0.0  0:00.00 cpuhp/0
   18 root        20   0     0     0     0 S   0.0  0.0  0:00.00 cpuhp/1
   19 root        RT   0     0     0     0 S   0.0  0.0  0:00.00 migration/1
   25 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-inet_
   27 root        20   0     0     0     0 S   0.0  0.0  0:00.00 oom_reaper
   28 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-write
   29 root        20   0     0     0     0 S   0.0  0.0  0:00.25 kcompactd0
   30 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-pencr
   31 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-pdecr
   32 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-crypt
   33 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-kbloc
   34 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-ata_s
   35 root       -51   0     0     0     0 S   0.0  0.0  0:00.00 watchdogd
   43 root        20   0     0     0     0 S   0.0  0.0  0:00.00 kswapd0
   44 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-kthro
   45 root         0 -20     0     0     0 I   0.0  0.0  0:00.98 kworker/0:1H-kblockd
  127 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-nvme-
  148 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-nvme-
  150 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-nvme-
  159 root        20   0     0     0     0 S   0.0  0.0  0:00.01 scsi_eh_0
  167 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-scsi_
  171 root        20   0     0     0     0 S   0.0  0.0  0:00.00 scsi_eh_1
  172 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-scsi_
  368 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/u7:0-hci0
  370 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/u8:0-MWIFIEX_WORK_QUEUE
  371 root        20   0     0     0     0 I   0.0  0.0  0:01.35 kworker/0:5-events
  372 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/u9:0-MWIFIEX_WORK_QUEUE
  373 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-sdhci
  374 root       -51   0     0     0     0 S   0.0  0.0  0:00.00 irq/50-mmc0
  379 root       -51   0     0     0     0 S   0.0  0.0  0:00.00 irq/51-f1090000.crypto
  381 root       -51   0     0     0     0 S   0.0  0.0  0:00.00 irq/52-f1090000.crypto
  393 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-mld
  396 root        20   0     0     0     0 I   0.0  0.0  0:00.14 kworker/u6:2-events_unbound
  399 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-ipv6_
  400 root         0 -20     0     0     0 I   0.0  0.0  0:00.78 kworker/1:1H-kblockd
[... 572 lines cut ...]
  170 root        20   0     0     0     0 S   0.0  0.0  0:00.00 scsi_eh_1
  171 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-scsi_
  190 root        20   0     0     0     0 I   0.0  0.0  0:00.14 kworker/u5:4-events_unbound
  368 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/u7:0-hci0
[... cut ...]
  623 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/1:1H
  628 root       -51   0     0     0     0 S   0.0  0.0  0:00.00 irq/53-gpio-keys
  629 root       -51   0     0     0     0 S   0.0  0.0  0:00.00 irq/54-gpio-keys
  637 root         0 -20     0     0     0 I   0.0  0.0  0:00.00 kworker/R-uas
  751 root        20   0     0     0     0 I   0.0  0.0  0:00.03 kworker/u6:2-events_unbound
  796 root        20   0     0     0     0 S   0.0  0.0  0:00.01 ubifs_bgt0_1
  807 root        20   0     0     0     0 S   0.0  0.0  0:00.00 ubi_bgt1d
```

brada4, 2024-07-15:
You need colour terminal...

httpstorm, 2024-07-15:
@brada4 
The terminal does have colour. But I don't think I can preserve it when I copy and paste the text as a code block here.
I didn't know you wanted screenshots.

brada4, 2024-07-15:
Either way ksoftirqd/0 of cpu0 does the iptables, so work is not balanced, but most clog is in userland process. At least that what linux kernel accounts.

httpstorm, 2024-07-15:
@brada4 
There you go. Plenty o'colour, packet-steering on/off your patch with/without, TCP 1/2 connections HTTP/HTTPS.
Feedback on my web site is welcome 😎
https://httpstorm.com/share/.openwrt/test/2024-07-15_steering/

brada4, 2024-07-15:
Patch works against softirq, wait for 23.05.4 ;-) Rest looks quite balanced.