openwrt/openwrt #15821: [RPi4] skb_warn_bad_offload with kernel 6.6

unterd0g, 2024-06-27:
### Describe the bug

Logread/dmesg shows skb_warn_bad_offload kernel trace and packet routing performance is degraded.
WAN is on a kmod-usb-net-rtl8152 device (TP-Link UE300). ISP requires using a VLAN (12).

Issues started ever since kernel 6.6 was introduced, kernel 6.1 works without issue.

```
Thu Jun 27 19:08:45 2024 user.notice firewall: Reloading firewall due to ifup of wan6 (eth1.12)
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.278094] skb len=1365 headroom=130 headlen=1292 tailroom=0
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.278094] mac=(130,14) net=(144,40) trans=184
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.278094] shinfo(txflags=0 nr_frags=0 gso(size=1230 type=393216 segs=2))
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.278094] csum(0x8fa8 ip_summed=0 complete_sw=0 valid=0 level=0)
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.278094] hash(0xf7086e0d sw=1 l4=1) proto=0x86dd pkttype=0 iif=6
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.307736] dev name=eth1.12 feat=0x02000006401d5849
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.312710] skb linear:   00000000: 00 03 fa 00 00 01 5c a6 e6 b6 07 3d 86 dd 60 01
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.320371] skb linear:   00000010: 37 5f 05 1f 11 3f 20 01 08 a0 dd 00 87 00 60 84
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.328029] skb linear:   00000020: f6 b3 92 2b ba f8 2a 00 14 50 40 01 08 0f 00 00
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.335686] skb linear:   00000030: 00 00 00 00 20 0a ed 99 01 bb 05 1f dc 68 c3 00
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.343343] skb linear:   00000040: 00 00 01 08 0e 8d 29 d7 d2 92 62 60 00 40 46 00
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.350999] skb linear:   00000050: aa 82 b6 18 c6 9f 50 c9 12 1e fe 09 84 55 c1 0c
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.358655] skb linear:   00000060: 99 98 87 f9 65 c1 9b d7 5c d7 a6 16 e1 3c 82 1a
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.366311] skb linear:   00000070: 41 53 83 09 62 0c 27 81 25 64 9b bc 4a b7 93 89
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.373967] skb linear:   00000080: be a5 ca 3d 22 65 4a 5e 33 32 df 46 0a aa 7c 8f
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.381623] skb linear:   00000090: 59 2b b3 0a dd 44 75 f2 c5 52 cc ae 96 db 8e bb
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.389279] skb linear:   000000a0: 11 93 38 5a 71 3d a1 7b d1 9d f8 da 3b 80 45 11
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.396936] skb linear:   000000b0: 3d a3 73 0a cd e8 a0 ec cc 8a 81 46 0b b0 f6 c7
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.404592] skb linear:   000000c0: 82 87 b2 92 57 10 3b 90 5e f3 90 17 2b 86 2f 6e
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.412247] skb linear:   000000d0: 4e ef fa 11 ea 23 26 65 87 db 4c a1 00 10 25 f4
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.419903] skb linear:   000000e0: 8d 09 b4 3f c1 33 fd c0 cd d2 09 dd 12 89 da 93
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.427560] skb linear:   000000f0: 21 bf 65 c5 b4 4f aa 77 28 e7 74 90 50 5b 0f 42
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.435215] skb linear:   00000100: 88 83 9f 56 43 85 cd 40 36 84 6f 24 c2 20 6d 59
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.442871] skb linear:   00000110: 6e 11 7d ea 47 7c 0f 6a d1 f6 60 a6 ae f7 e5 65
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.450527] skb linear:   00000120: 57 f9 ec a9 56 26 6a 32 af 03 db 4c 02 cc 01 e2
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.458184] ------------[ cut here ]------------
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.462799] eth1.12: caps=(0x02000006401d5849, 0x0000000000000000)
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.469016] WARNING: CPU: 0 PID: 0 at net/core/dev.c:3313 skb_warn_bad_offload+0x7c/0x114
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.477206] Modules linked in: brcmfmac_wcc(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet brcmfmac(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack cfg80211(O) cdc_ncm cdc_ether usbnet usbhid tcp_bbr slhc sch_cake r8169 r8152 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c lan78xx hid_generic crc_ccitt compat(O) brcmutil(O) sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact snd_bcm2835(C) hid evdev ifb snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 nls_iso8859_1 nls_cp437 vfat fat realtek microchip
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.560086] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C O       6.6.35 #0
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.567483] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.573313] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.580274] pc : skb_warn_bad_offload+0x7c/0x114
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.584889] lr : skb_warn_bad_offload+0x7c/0x114
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.589503] sp : ffffffc080003770
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.592813] x29: ffffffc080003770 x28: 0000000000000000 x27: ffffffc080b7a9a0
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.599954] x26: 0000000000000000 x25: ffffff804468c610 x24: 0000000000000036
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.607094] x23: ffffff8040da3400 x22: 0000000000000000 x21: ffffff8041e37000
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.614234] x20: ffffff8041e37000 x19: ffffff80445d7c00 x18: 0000000000000141
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.621374] x17: 6320323020633420 x16: 6264203330206661 x15: ffffffc080b89ec8
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.628513] x14: 00000000000003c3 x13: 0000000000000141 x12: 00000000ffffffea
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.635653] x11: 00000000ffffefff x10: ffffffc080be1ec8 x9 : ffffffc080b89e70
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.642792] x8 : ffffffc080b79000 x7 : 0000000000000001 x6 : ffffffc080b791c0
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.649931] x5 : 0000000000000001 x4 : 0000000000000001 x3 : 0000000000010005
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.657070] x2 : 0000000000000000 x1 : 0000000000000005 x0 : 0000000000000036
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.664209] Call trace:
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.666651]  skb_warn_bad_offload+0x7c/0x114
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.670919]  __skb_gso_segment+0xe8/0x14c
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.674926]  validate_xmit_skb.isra.0+0x128/0x328
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.679628]  __dev_queue_xmit+0x1a4/0xd68
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.683635]  neigh_resolve_output+0xfc/0x1f0
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.687901]  ip6_finish_output2+0x22c/0x640
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.692083]  ip6_finish_output+0x128/0x370
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.696177]  ip6_output+0x64/0x138
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.699576]  ip6_forward+0x5dc/0xa90
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.703148]  ipv6_rcv+0xdc/0x110
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.706374]  __netif_receive_skb_one_core+0x48/0x58
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.711249]  netif_receive_skb+0x58/0xf4
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.715168]  br_pass_frame_up+0x14c/0x1a4
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.719179]  br_handle_frame_finish+0x494/0x5ac
Thu Jun 27 19:08:48 2024 kern.warn kernel: [   20.723709]  br_handle_frame+0x334/0x42c
[... 96 lines cut ...]
```

### OpenWrt version

r26805-b43c0a915f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

unterd0g, 2024-06-27:
I believe the bot set the invalid tag because the info was not in code tags. updated.

rany2, 2024-06-27:
> I believe the bot set the invalid tag because the info was not in code tags. updated.

It's because of extra whitespace, the bot doesn't remove the leading/trailing whitespaces when validating.

FilHarr, 2024-07-03:
Seeing the same issue when compiling a snapshot build.

Getting the following in the logs:

`Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.573308] skb len=1113 headroom=148 headlen=1019 tailroom=0
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.573308] mac=(148,0) net=(148,40) trans=188
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.573308] shinfo(txflags=0 nr_frags=0 gso(size=971 type=393216 segs=2))
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.573308] csum(0xffff9158 ip_summed=2 complete_sw=0 valid=0 level=0)
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.573308] hash(0xaee95d88 sw=1 l4=1) proto=0x86dd pkttype=0 iif=2
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.603123] dev name=6in4-henet feat=0x02000006401d5869
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.608377] skb linear:   00000000: 60 04 e7 5d 04 31 11 3f 20 01 04 70 6e 03 da f7
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.616038] skb linear:   00000010: 88 5e d9 05 a0 cd e5 61 2a 00 14 50 40 09 08 21
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.623702] skb linear:   00000020: 00 00 00 00 00 00 20 0e f4 b6 01 bb 04 31 33 fe
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.631359] skb linear:   00000030: dc 00 00 00 01 08 54 ac 43 5c bb b8 da 42 00 43
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.639015] skb linear:   00000040: ba c2 5a 94 bd 8c 81 61 81 2d e4 29 f6 c6 a3 95
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.646666] skb linear:   00000050: f2 10 9d 38 b1 74 e0 3d 0b 76 e1 97 fa 5c 60 27
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.654320] skb linear:   00000060: e1 9b 12 28 1b aa 9a c5 95 5d 41 25 a9 03 25 13
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.661975] skb linear:   00000070: 3e 5b 77 2a e9 83 43 d2 9f 8e 95 cb e8 59 c1 92
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.669631] skb linear:   00000080: 0d 5b 7e 10 0c fe 24 76 e5 86 55 12 95 e6 8d cc
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.677290] skb linear:   00000090: c8 aa 5c 4f de b4 33 3d 26 33 e6 6d 26 35 e1 59
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.684940] skb linear:   000000a0: fd dd 43 28 fb b3 9d e3 51 9b 0d b2 a8 44 c5 6b
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.692597] skb linear:   000000b0: 6f 56 11 f0 f2 76 92 b8 58 62 4a a8 91 13 03 62
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.700256] skb linear:   000000c0: fd b2 28 6c 1f 42 dd 10 58 c7 22 32 8d de 3f 50
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.707914] skb linear:   000000d0: 0f 0d ae a2 0f 05 e7 6d 93 a7 f9 c5 2d ec 74 74
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.715572] skb linear:   000000e0: 10 24 78 7f 6b a4 29 ac 27 7e 43 7e e7 91 c6 63
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.723246] skb linear:   000000f0: a4 ec d2 ca 22 77 89 05 88 98 9c bd 20 b1 f0 a0
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.730917] skb linear:   00000100: f8 cf 3e 10 ed 21 52 f5 e5 c8 22 b6 d3 88 92 1e
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.738589] skb linear:   00000110: 87 4a d7 39 20 d7 53 71 8d f9 83 a8 f5 33 bd e2
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.746261] skb linear:   00000120: 6d cb 17 8e e7 80 b9 af a1 4d 95 e2 12 07 4a 22
Tue Jul  2 00:34:20 2024 kern.warn kernel: [ 1856.753933] ------------[ cut here ]------------
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.758560] 6in4-henet: caps=(0x02000006401d5869, 0x0000000000000000)
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.765044] WARNING: CPU: 0 PID: 0 at net/core/dev.c:3313 skb_warn_bad_offload+0x7c/0x114
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.773252] Modules linked in: brcmfmac_wcc(O) nft_fib_inet nf_flow_table_inet brcmfmac(O) wireguard nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack libchacha20poly1305 ip_tables chacha_neon cfg80211(O) cdc_ncm cdc_ether x_tables usbnet usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda r8169 r8152 poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha lan78xx hid_generic crc_ccitt compat(O) brcmutil(O) snd_bcm2835(C) hid evdev ip6_udp_tunnel udp_tunnel sit tunnel4 ip_tunnel snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore autofs4 nls_utf8 nls_iso8859_1 nls_cp437 vfat fat realtek microchip
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.863816] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC O       6.6.35 #0
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.871226] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.877064] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.884036] pc : skb_warn_bad_offload+0x7c/0x114
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.888664] lr : skb_warn_bad_offload+0x7c/0x114
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.893289] sp : ffffffc080003a60
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1856.896605] x29: ffffffc080003a60 x28: 0000000000000000 x27: ffffffc080003e28
[... cut ...]
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1857.070855]  cpu_startup_entry+0x34/0x3c
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1857.074782]  kernel_init+0x0/0x1e0
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1857.078191]  arch_post_acpi_subsys_init+0x0/0x8
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1857.082727]  start_kernel+0x45c/0x520
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1857.086393]  __primary_switched+0xb4/0xbc
Tue Jul  2 00:34:21 2024 kern.warn kernel: [ 1857.090412] ---[ end trace 0000000000000000 ]---`

Have rolled back to 23.05.3 and no issues.

unterd0g, 2024-07-06:
Massive thanks to @everything411 who dug deeper and found the issue is caused by commit b5c5384.
In short, the issue is triggered after enabling the NETIF_F_GRO_FRAGLIST. The root cause is likely an unforeseen condition in the kernel (upstream) that was never discovered because this feature is disabled by default.

You can workaround this issue using `ethtool` to disable `rx-gro-list` in all your interfaces. That is, for instance, `ethtool -K eth0 rx-gro-list off`.
_caveat emptor_ only a few hours have passed since I introduced this workaround in my RPi4, and I still need further testing to validate if the kernel panics disappear entirely.

BrainSlayer, 2024-08-10:
> ethtool -K eth0 rx-gro-list off

this workaround will not work. in fact it immediatly triggers the kernel crash

everything411, 2024-08-10:
use this patch to disable gro completely https://github.com/openwrt/openwrt/commit/b5c53848c3792355a2dd4ff0e06bf4d230e1f5fb#commitcomment-143548442
t think this can be a more complex but better workaround for gro bug

unterd0g, 2024-08-11:
Because a quick `git revert` will have conflicts and everything is easily contained in patch files, the easiest course of action is tracking down the offending patches in K6.6 and manually reverting b5c5384 (and its later fixes).

That way, we keep the upstream GRO/GSO, and things just work _(that's what I've done)_.
Unless the issue gets fixed in time for the next release, there is no way OpenWRT can call this "stable". So, maybe a similar PR will inevitably appear around that time.

unterd0g, 2024-08-16:
Two new commits may change the behavior (17f0704 and 79fce42), especially the [611-03-v6.11-udp-Fall-back-to-software-USO-if-IPv6-extension-head.patch](https://github.com/openwrt/openwrt/commit/79fce424046fed521026df7c7f838441fb46add8#diff-30167fe8fd32a3bc967ab826308f6be994c98cfa940b8f9eb71b85977b221ca3) may be relevant to my issue. Building from source and testing :)

unterd0g, 2024-08-16:
Cannot reproduce this issue anymore, looks like it is fixed after 17f0704 and 79fce42.

closed 2024-08-16 by unterd0g