openwrt/openwrt #15829: realtek: SFP ports on GS1900-48 don't function reliably

jwh, 2024-06-28:
### Describe the bug

On the GS1900-48, port 49 and 50 don't work reliably as one or more of the following happens:

* Upon link state change (or even accessing them with ethtool) - a "REBOOT" is printed and the switch resets after some OOPS in gpiolib, I haven't managed to track down where the reboot gets triggered from, so for now I've just disabled the gpio reset definitions in the DTS to avoid rebooting almost every time.
* No reset, but the link does not come up - debugfs entries show there is link however the kernel never reports link up, nor does ethtool.

On this switch, ports 49 and 50 are connected to SerDes, but I can't see anything that would set this up properly.

Issuing `rtk network on` before booting allows the link to come up (but sometimes the reboot will happen regardless, so it might be poling the wrong thing) - without gpio resets this doesn't happen and the port works as expected.

Given the SerDes and that `rtk network on` probably sets up the PHYs properly, there is likely some setup missing however I'm not familiar at all with this hardware so some suggestions on how to diagnose this would be good.

### OpenWrt version

HEAD

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

realtek/rtl839x

### Device

Zyxel GS1900-48

### Image kind

Self-built image

### Steps to reproduce

Boot snapshot, plug in SFP (of any brand/type) and witness reboots

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-05:
Crucial detail - if and what transceiver is connected to each.

jwh, 2024-07-05:
It makes no difference, SFP or not, but they're all spec compliant and all report correct link state etc to the driver available via debugfs - given that there is no code to even setup these ports, sfp brand is probably not crucial :)

brada4, 2024-07-05:
Can you answer the question? Do you get ghost transceiver when nothing is plugged? Is it zyxel transceeiver? Other major brand? 10 bucks from ali?

jwh, 2024-07-07:
I did, "it makes no difference" and "debugfs entries show there is link however the kernel never reports link up, nor does ethtool." - 200$ or 20$ sfp is irrelevant, it is missing setup.

closed 2024-08-29 by jwh

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442