openwrt/openwrt #15834: Raspberry Pi 5 Status LED is Amber/Orange Instead of Green

dkachel7258, 2024-06-29:
### Describe the bug

When OpenWrt has finished booting, the status light on the Raspberry Pi 5 is orange/amber instead of green, as it used to be.

### OpenWrt version

r26730-17099f2760

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Steps to reproduce

Install a clean Snapshot release on a MicroSD card and boot.

### Actual behaviour

Prior to snapshot r26730-17099f2760, once the boot process finishes, the ACT (green) 
LED is on and the PWR (red) LED is off. In subsequent snapshot releases both LEDs are on.

### Expected behaviour

Once the boot process finishes, the ACT (green) LED is on and the PWR (red) LED is off.

### Additional info

WORKAROUND

Adding the following lines to /boot/config.txt seems to restore the former behavior
of the red PWR stat LED.

dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off

ADDITIONAL INFORMATION
The behavior started with the r26730-17099f2760 release (June 20, 2024).

Note that the behavior doesn't appear to be a power supply issue - measured at 5.12v, 0.64a, when the PWR LED is on.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aress31, 2024-07-01:
I have observed the same thing.

Noltari, 2024-07-02:
Before 6fdf3a32d97aca287c4a9f7121e589ecb1e64e18 OpenWrt was using "led0" or "led1", depending on the RPi model.
However, since these LEDs didn't exist, the "status light" wasn't actually a status light, but the MMC activity LED.
After 6fdf3a32d97aca287c4a9f7121e589ecb1e64e18, the LED "PWR" is the one used for OpenWrt diagnostics and therefore it's left powered on when booted.
I think this is the correct way to go since most OpenWrt devices use the power LED for the status.

dkachel7258, 2024-07-02:
My own understanding is that since the RPI 5 has a power button, the ACT and PWR LEDS indicate whether the system is running (green) or halted (red) - and thus whether pressing the power button to halt the system has succeeded.

As far as I know, the other models of the RPI don't offer this functionality and the PWR LED just indicates that the board has power, and one can't easily tell whether the system is actually running or halted.

(... unless the system is running without a MicroSD card in it, and then the green LED double-blinks as an ersatz indication that the system isn't halted. 😀)

Noltari, 2024-07-03:
Should be fixed with 37636a8752ecdc0ab399c3f38979c6cfb5576bc6

closed 2024-07-03 by Noltari