openwrt/openwrt #15835: Arcadyan ARV752DPW - Flash not erased

Yemtex, 2024-06-30:
### Describe the bug

Hi,

I'm currently trying to install version 23 of OpenWRT on my Arcadyan ARV752DPW (Easybox).
But every time I try to copy the squashfs sysupgrade image into flash memory (https://openwrt.org/toh/arcadyan/arv752dpw#permanent_openwrt), I get the following error 
```
ARV752DPW # crc32 81000000 6c0141
CRC32 for 81000000 ... 816c0140 ==> b4b8505c
ARV752DPW # protect off 0xB0040000 +6c0141
............................................................................................................. done
Un-Protected 109 sectors
ARV752DPW # erase 0xB0040000 +6c0141

............................................................................................................. done
Erased 109 sectors
ARV752DPW # cp.b 0x81000000 0xb0040000 6c0141
Copy to Flash... Flash not Erased
```

I think the flash memory on this device is damaged or has too many bad blocks, I would appreciate some help in proving my thesis right.

I'm a contributor for the https://openwrt.org/toh/arcadyan/arv752dpw documentation and I want to check if version 23 is running stable.
Just for your info, I added the warning in the documentation, about not installing version 23, this has already been solved https://github.com/openwrt/openwrt/issues/14283.

I previously bricked the device and restored it to a working u-boot in flash memory by following my steps in the current documentation (https://openwrt.org/toh/arcadyan/arv752dpw#uart_u-boot_bricked_device_only).
But now I'm stuck with the error mentioned above.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

lantiq/xway

### Device

Arcor 802

### Image kind

Official downloaded image

### Steps to reproduce

Following the steps in https://openwrt.org/toh/arcadyan/arv752dpw#permanent_openwrt using xmodem.

### Actual behaviour

```
ARV752DPW # cp.b 0x81000000 0xb0040000 6c0141
Copy to Flash... Flash not Erased
```

### Expected behaviour

Working copy from RAM to flash memory
```
ARV752DPW # cp.b 0x81000000 0xb0040000 6c0141
Copy to Flash... done
```

### Additional info

1. Unbricked device
2. Installed permanent u-boot into flash memory
3. Tried temporary OpenWRT (initramfs)
4. Initramfs OpenWRT is working
5. Tried installing permanent (squashfs-sysupgrade) OpenWRT
6. Copying sysupgrade image from RAM to flash memory failed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-01:
Have you read the warning on top of device page?

Yemtex, 2024-07-02:
Yes, I added the warning. We had a similiar conversation in the [issue linked](https://github.com/openwrt/openwrt/issues/14283#issuecomment-1869344027)
I want to test before I remove the warning. I got the initramfs image running, because I think my RAM is okey but my flash memory is not.

Yemtex, 2024-08-03:
Does anyone have any ideas?