openwrt/openwrt #15856: 最近几天编译的固件ipv6无法使用

uosbox, 2024-07-02:
### Describe the bug

四月初编译的固件ipv6上网正常
最近几天编译最新版的openwrt固件ipv6无法正常使用，上网方式pppoe模式，路由器可以获取到ipv6地址
连接到路由器的手机和电脑都能正常获取ipv6地址，但是无法通过ipv6上网，可以排除dns问题！
https://www.test-ipv6.com/地址测试不通过
[image]

[image]
[image]
[image]
[image]
[image]

### OpenWrt version

r26831+143-a787ebf716

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

nanopc-t4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

uosbox, 2024-07-02:
[image]

uosbox, 2024-07-02:
[image]

GoetzGoerisch, 2024-07-02:
Post the output of 
/etc/config/network
/etc/config/dhcp
/etc/config/firewall

Also please elaborate where you downloaded this firmware version as the current version is only:

`r26829-8ef4d78582` for this target

uosbox, 2024-07-03:
固件是拉取`https://github.com/openwrt/openwrt.git`编译的，main分支，`make menuconfig` 
4月初编译了一个固件没问题，可以正常使用，但是最近几天编译的固件就遇到这样的情况。相同的`.config`文件
/etc/config/network
`config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdb9:2c01:daea::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config device
	option name 'eth0.1'
	option macaddr '92:66:7c:66:35:3e'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.9.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device
	option name 'eth0.2'
	option macaddr '92:66:7c:66:35:3d'

config interface 'wan'
	option device 'eth0.2'
	option proto 'pppoe'
	option username 'xxxxxxx'
	option password 'xxxxx'
	option ipv6 'auto'

config interface 'wan6'
	option device 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'
	option enable_vlan4k '1'
	option max_length '3'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '7t 1 2 3 4 5'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '7t 6'

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

config device
	option type 'bridge'
	option name 'docker0'
`
/etc/config/dhcp
`config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option nonegcache '0'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '1232'
	option filter_aaaa '0'
	option filter_a '0'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	list dhcp_option '6,8.8.8.8'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4
`
/etc/config/firewall
`config defaults
	option syn_flood '1'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option network 'lan '

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config zone 'docker'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option name 'docker'
	list network 'docker'

config rule
	option name '80'
	list proto 'tcp'
	option src 'wan'
	option dest_port '80-88'
	option target 'ACCEPT'
`

uosbox, 2024-07-03:
固件是拉取`https://github.com/openwrt/openwrt.git`编译的，main分支，`make menuconfig` 
4月初编译了一个固件没问题，可以正常使用，但是最近几天编译的固件就遇到这样的情况。相同的`.config`文件
`config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdb9:2c01:daea::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config device
	option name 'eth0.1'
	option macaddr '92:66:7c:66:35:3e'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.9.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device
	option name 'eth0.2'
	option macaddr '92:66:7c:66:35:3d'

config interface 'wan'
	option device 'eth0.2'
	option proto 'pppoe'
	option username '18328620069a'
	option password '620069'
	option ipv6 'auto'

config interface 'wan6'
	option device 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'
	option enable_vlan4k '1'
	option max_length '3'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '7t 1 2 3 4 5'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '7t 6'

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

config device
	option type 'bridge'
	option name 'docker0'

`

GoetzGoerisch, 2024-07-05:
The host inside your lan has an IPv6 address, does it have a default route?
What does traceroute tell you from the router and a host in the `lan`?

What is the switch and docker doing in this setup?

uosbox, 2024-07-14:
找到问题原因了！似乎是由于 iptables 和 nftables 冲突导致的！
卸载掉ip6table 相关插件就正常了

我需要用docker 不知道nftables 是否正常用，默认docker编译自动选择了iptables

uosbox, 2024-07-14:
#24492#issuecomment-2211509498
这个补丁可以解决br-lan无法联网的问题！由新版docker引起的问题！

ramonalvesmodesto, 2024-09-07:
I also had problems with ipv6, in my case I had to make a script for manual configuration of [ipv6](https://github.com/ramonalvesmodesto/openwrt-config/tree/main/ipv6). Looking at your configuration, the slaac option is enabled on the luci interface, but not in the configuration file. In my research, adding `option ra_slaac '1'` in `/etc/config/network` in the `interface 'lan'` option may or may not solve the problem.

evanlucky, 2025-05-18:
> I also had problems with ipv6, in my case I had to make a script for manual configuration of [ipv6](https://github.com/ramonalvesmodesto/openwrt-config/tree/main/ipv6). Looking at your [...]

thank you! it works for me . you save my life

evanlucky, 2025-05-18:
我的openwrt是二级路由，通过设置为中继模式终于成功了。我参考了这个博客 [https://github.com/openwrt/openwrt/issues/15856](url)