openwrt/openwrt #15857: 6.6: NAT64 performance degradation

GoetzGoerisch, 2024-07-02:
### Describe the bug

As a follow up to #15848

Testing with the fix introduced by @Noltari :

Paketsteering: All CPU
Steering flows: Standard none
Software flow offloading: false
Hardware flow offloading: false

Single file download with wget
IPv4 NAT44 - 100Mibit/s  Load 2.0 - 10G file
IPv6 - 100Mibit/s - Load 1.7 - 10G file
**kernel NAT64 - 10Mibit/s - Load 0.5 - 10G file**

As soon as mixed traffic, e.g., surfing to a website with NAT64 and IPv6 traffic, the router crashes.

Last log:
```
Tue Jul  2 16:10:28 2024 [1719936628.569] kern.warn kernel: [ 9671.780974] skb len=193 headroom=102 headlen=125 tailroom=0
Tue Jul  2 16:10:28 2024 [1719936628.569] kern.warn kernel: [ 9671.780974] mac=(102,14) net=(116,20) trans=136
Tue Jul  2 16:10:28 2024 [1719936628.569] kern.warn kernel: [ 9671.780974] shinfo(txflags=0 nr_frags=0 gso(size=83 type=393216 segs=2))
Tue Jul  2 16:10:28 2024 [1719936628.569] kern.warn kernel: [ 9671.780974] csum(0x0 ip_summed=0 complete_sw=0 valid=0 level=1)
Tue Jul  2 16:10:28 2024 [1719936628.570] kern.warn kernel: [ 9671.780974] hash(0x511acf2e sw=1 l4=0) proto=0x0800 pkttype=0 iif=10
Tue Jul  2 16:10:28 2024 [1719936628.570] kern.warn kernel: [ 9671.811044] dev name=openwrt feat=0x020061164fdd19e9
Tue Jul  2 16:10:28 2024 [1719936628.570] kern.warn kernel: [ 9671.816171] skb linear:   00000000: 86 4d 34 db 27 ef d2 90 44 c3 ee 5a 08 00 45 00
Tue Jul  2 16:10:28 2024 [1719936628.592] kern.warn kernel: [ 9671.824106] skb linear:   00000010: 00 b3 63 5d 00 00 3e 11 65 75 c0 a8 a4 02 22 a0
Tue Jul  2 16:10:28 2024 [1719936628.593] kern.warn kernel: [ 9671.831963] skb linear:   00000020: 2c 1d fd d3 01 bb 00 9f 6b 86 ed 00 00 00 01 08
Tue Jul  2 16:10:28 2024 [1719936628.593] kern.warn kernel: [ 9671.839895] skb linear:   00000030: f3 d8 d4 70 22 ee f0 26 00 40 42 4a 74 c3 7b 92
Tue Jul  2 16:10:28 2024 [1719936628.600] kern.warn kernel: [ 9671.847804] skb linear:   00000040: 7a d9 eb 06 7d 85 63 d0 30 87 f3 02 8b a6 77 e3
Tue Jul  2 16:10:28 2024 [1719936628.616] kern.warn kernel: [ 9671.855727] skb linear:   00000050: 5c f0 87 64 93 d0 2a f5 27 ca c3 93 90 b1 4e 10
Tue Jul  2 16:10:28 2024 [1719936628.616] kern.warn kernel: [ 9671.863617] skb linear:   00000060: 12 21 f7 85 67 d8 4a 64 4d c4 0a c7 bf 21 7d e0
Tue Jul  2 16:10:28 2024 [1719936628.631] kern.warn kernel: [ 9671.871472] skb linear:   00000070: 0e 81 e2 2b 35 a1 df 10 08 4b 61 d2 48
Tue Jul  2 16:10:28 2024 [1719936628.631] kern.warn kernel: [ 9671.878613] ------------[ cut here ]------------
Tue Jul  2 16:10:28 2024 [1719936628.651] kern.warn kernel: [ 9671.883372] WARNING: CPU: 0 PID: 3957 at net/core/dev.c:3313 skb_warn_bad_offload+0x94/0x138
Tue Jul  2 16:10:28 2024 [1719936628.651] kern.warn kernel: [ 9671.892075] openwrt: caps=(0x020061164fdd19e9, 0x0000000000000000)
```

Related: #15847 #15821 

### OpenWrt version

r26830-a7f37b401b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bmips/bcm6368

### Device

Netgear DGND3800B

### Image kind

Official downloaded image

### Steps to reproduce

- Install SNAPSHOT
- [Setup NAT64](https://openwrt.org/docs/guide-user/network/ipv6/nat64?s%5B%5D=jool#option_2_-_running_jool_in_a_separate_network_namespace1)
- Measure throughput through NAT64
- surf the web

### Actual behaviour

crash and router reboot

### Expected behaviour

same performance and no crash like 5.15.158

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

GoetzGoerisch, 2024-07-06:
Retested with `6.6.36`

```
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.518221] skb len=199 headroom=102 headlen=125 tailroom=0
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.518221] mac=(102,14) net=(116,20) trans=136
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.518221] shinfo(txflags=0 nr_frags=0 gso(size=83 type=393216 segs=2))
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.518221] csum(0x0 ip_summed=0 complete_sw=0 valid=0 level=1)
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.518221] hash(0xc97a5a04 sw=1 l4=0) proto=0x0800 pkttype=0 iif=12
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.548277] dev name=nat64 feat=0x020061164fdd19e9
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.553217] skb linear:   00000000: a6 b6 fe 68 a9 b8 9e c4 42 14 eb 87 08 00 45 00
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.561139] skb linear:   00000010: 00 b9 d3 42 00 00 3e 11 f5 89 c0 a8 a4 02 22 a0
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.568995] skb linear:   00000020: 2c 1d bf 5c 01 bb 00 a5 04 ed e0 00 00 00 01 08
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.576929] skb linear:   00000030: e7 9e f8 91 09 05 4f 57 00 40 42 fb 66 84 0f 0d
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.584833] skb linear:   00000040: 7c 14 55 ed 53 0f 97 7b 8d 9b c9 1f de ab bb 92
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.592758] skb linear:   00000050: c8 47 48 e4 82 40 94 52 18 93 e7 ae 58 2f 96 a1
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.600682] skb linear:   00000060: 4c c8 9c 7b 3c 4f 6c 5e 3e f1 74 95 87 f5 4d 67
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.608557] skb linear:   00000070: 6e d4 c2 85 cc f2 1a 38 8e d8 3c dc f0
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.615673] ------------[ cut here ]------------
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.620449] WARNING: CPU: 0 PID: 0 at net/core/dev.c:3313 skb_warn_bad_offload+0x94/0x138
Sat Jul  6 10:20:33 2024 kern.warn kernel: [54585.628924] nat64: caps=(0x020061164fdd19e9, 0x0000000000000000)
```
`openwrt` in previous output `nat64` in this one

everything411, 2024-07-10:
This is a general issue affecting all targets and is related to fraglist GRO (rx-gro-list), see #15623.
cc @Noltari @nbd168

everything411, 2024-07-10:
@GoetzGoerisch Please try this https://github.com/openwrt/openwrt/issues/15623#issuecomment-2160486177 workaround and see if it works.

GoetzGoerisch, 2024-07-10:
Thanks  @everything411, but it does not change anything.

Set `ethtool -K switch.2 rx-gro-list off` and `ethtool -K switch.1 rx-gro-list off` but still 1/10th of the bandwidth of not NATed traffic and still crashing.

Crashlog with `rx-gro-list on`

```
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.342668] skb len=1441 headroom=102 headlen=1272 tailroom=0
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.342668] mac=(102,14) net=(116,20) trans=136
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.342668] shinfo(txflags=0 nr_frags=0 gso(size=1230 type=393216 segs=2))
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.342668] csum(0x0 ip_summed=0 complete_sw=0 valid=0 level=1)
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.342668] hash(0x7b820328 sw=1 l4=0) proto=0x0800 pkttype=0 iif=11
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.373215] dev name=nat64 feat=0x020061164fdd19e9
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.378131] skb linear:   00000000: da 64 02 11 5f 54 c2 f4 22 9f 77 07 08 00 45 00
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.386071] skb linear:   00000010: 05 93 d8 91 00 00 3e 11 95 18 c0 a8 a4 02 68 14
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.394044] skb linear:   00000020: 3c f1 92 0f 01 bb 05 7f 30 00 c0 00 00 00 01 14
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.402009] skb linear:   00000030: 01 d0 6f 05 84 19 69 58 4d d1 b8 04 ee 19 47 16
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.409886] skb linear:   00000040: 22 f9 29 dd 00 00 44 7c c6 72 61 54 a2 ef f6 34
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.417807] skb linear:   00000050: 4c b6 6b 85 f1 20 f1 9e e2 6c 8d a0 d3 f7 fd 14
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.425727] skb linear:   00000060: 3b 9d c0 b0 19 47 c7 16 ad ef 4f fa c7 97 40 5e
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.433647] skb linear:   00000070: b2 ad f1 cf e5 3d 5b 1b be 67 54 06 16 ba 04 38
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.441545] skb linear:   00000080: 28 7a e0 04 41 68 39 3b 29 2e d9 0a 78 01 f1 8e
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.449417] skb linear:   00000090: 93 ac 3c c3 35 8a 08 7f 17 fb 14 8e 25 88 cf 5c
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.457334] skb linear:   000000a0: 04 d0 f6 ce 34 da 55 4c f1 b9 f9 4f 26 23 4d 01
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.465252] skb linear:   000000b0: 28 d0 59 6f 23 78 3b 24 bb 25 da a0 c8 aa 21 3e
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.473168] skb linear:   000000c0: 81 35 32 01 2b 91 7b 33 6e 06 dc 56 33 b5 f8 55
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.481080] skb linear:   000000d0: db 5c 3a 3d d2 e5 54 fb 72 d0 37 17 eb db 7e 65
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.488958] skb linear:   000000e0: c5 04 42 fd 0c ce c0 ed 01 ca 8e a6 17 fb 2b 03
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.496875] skb linear:   000000f0: b2 cb 94 63 fc 20 1e 8b 49 8a 85 41 f1 f6 91 67
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.504790] skb linear:   00000100: 92 cb e1 e9 6f 34 af e9 5d 4a a9 ed 3b af 11 d3
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.512707] skb linear:   00000110: a2 8f 93 d2 f6 88 4d d3 93 53 fd 2b 93 32 9c 69
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.520634] skb linear:   00000120: 68 6a de e1 ee 20 8b 0c 99 65 76 cb 03 6e 68 30
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.528505] ------------[ cut here ]------------
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.533284] WARNING: CPU: 0 PID: 0 at net/core/dev.c:3313 skb_warn_bad_offload+0x94/0x138
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.541784] nat64: caps=(0x020061164fdd19e9, 0x0000000000000000)
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.547957] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet b43(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_nat nf_log_syslog nf_flow_table nf_conntrack libcrc32c crc_ccitt cordic compat(O) ledtrig_usbport jool_siit(O) jool(O) nf_defrag_ipv6 nf_defrag_ipv4 jool_common(O) veth sha512_generic seqiv sha3_generic jitterentropy_rng drbg md5 hmac geniv rng cmac leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) usbcore nls_base usb_common ssb crc32c_generic
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.620221] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O       6.6.36 #0
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.627816] Stack : 72633300 00000004 00000000 00000000 8180fb5c 80b70000 80a30000 808b98e8
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.636446]         809ad6d0 80a33437 00000000 00000000 00000000 10008f00 8180fb10 8183e180
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.645074]         00000000 00000000 808b98e8 8180fa40 00000000 00000000 ffffefff ffffffea
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.653693]         00000127 8180fa4c 00000127 809b2398 10008f01 808b98e8 8180fc10 805991a4
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.662322]         00000009 808f737c 809aa680 00000003 00000dd4 00000000 00000000 80b70000
Mon Jul  8 11:23:58 2024 kern.warn kernel: [ 5685.670943]         ...
[... cut ...]
Wed Jul 10 18:28:20 2024 kern.warn kernel: [ 2756.149302] skb linear:   00000040: 4d 08 00 26 ab 61 fb 40 91 94 e6 75 96 e1 1f 82
Wed Jul 10 18:28:20 2024 kern.warn kernel: [ 2756.157209] skb linear:   00000050: 7b fd
Wed Jul 10 18:28:20 2024 kern.warn kernel: [ 2756.161300] ------------[ cut here ]------------
Wed Jul 10 18:28:20 2024 kern.warn kernel: [ 2756.166090] WARNING: CPU: 0 PID: 0 at net/core/dev.c:3313 skb_warn_bad_offload+0x94/0x138
Wed Jul 10 18:28:20 2024 kern.warn kernel: [ 2756.174522] openwrt: caps=(0x020061164fdd19e9, 0x0000000000000000)

```

everything411, 2024-07-11:
also set `rx-gro-list off` on the switch interface `switch` itself besides the two vlans `switch.1` and `switch.2`

or just simply disable `rx-gro-list` on all of the interfaces listed in `ip link`

and see if it works?

GoetzGoerisch, 2024-07-11:
@everything411 switching it off at every interface, helps with the NAT performance but reduces also the not NATed traffic.
No crash so far.

everything411, 2024-07-11:
I only use NAT66 and the router is slow for NAT66 but works stable without crashing.

This indicates that `rx-gro-list` could improve bridging and NAT44 performance but harm NAT66 and NAT64 performance and crash NAT64 setups.

GoetzGoerisch, 2024-07-11:
It also harms not NAT performance, IPv6 forwarding.

IPv6 is down to about 75Mibit/s in the setting above.

GoetzGoerisch, 2024-08-17:
This affects also other targets like MT7981

CC: @hauke this is a blocking bug for 6.6, IMHO

Relates to: https://github.com/openwrt/openwrt/issues/15192

everything411, 2024-08-18:
https://github.com/openwrt/openwrt/commit/17f07040e7808a0c9cba2afc0e0b22e5589da133 and https://github.com/openwrt/openwrt/commit/79fce424046fed521026df7c7f838441fb46add8 fixes #15821. please test if this issue is also fixed with these two commits

GoetzGoerisch, 2024-08-18:
As I tested `r27160-b72c4b5386` on MT7891 (mediatek/filogic) / Cudy WR3000 v1, which includes the two fixes above, I can say, that this is not fixed.

With 5.15 the NAT performance is line speed, even for NAT64, but with 6.6 it is only 1/3.

nbd168, 2024-09-22:
Please try this patch for jool: https://nbd.name/p/f2a83bf4

GoetzGoerisch, 2024-09-22:
@nbd168 thanks for the patch.
Works, as it does the same like mentioned manually workaround https://github.com/openwrt/openwrt/issues/15857#issuecomment-2220732867

This is also related to NAT6& and needs to be fixed there as well.

nbd168, 2024-09-26:
Should be properly fixed in 774badd8a840be9888583266e12ba9d267201ee9 - please test.

GoetzGoerisch, 2024-09-28:
@nbd168 does improve the performance on the `mediatek/filogic` target with NAT64.
But does not show the same performance as with your `hotplug.d/net/90-jool-disable-fraglist-gro.sh` patch.

Unfortunately my `bmips` target is currently inoperative or possible dead.

nbd168, 2024-09-28:
@GoetzGoerisch  What's the difference in throughput between the different tests?

GoetzGoerisch, 2024-09-28:
with the hotplug.d/ patch - line speed 110MB/s
without - about 60% of line speed 60-65MB/s

everything411, 2024-09-28:
My school's ISP provide SLAAC now and I no longer use NAT66 so I'm not sure whether this commit also fixes NAT66 performance degrade

everything411, 2024-09-28:
I might give it a test during my holiday in October

GoetzGoerisch, 2024-10-04:
> @GoetzGoerisch What's the difference in throughput between the different tests?

@nbd168 I tested now also with a mt7621 target (r27648-71da76158e), NAT64 throughput is about the 50% of the 5.15 kernel throughput.

nbd168, 2024-10-04:
Thanks for testing. I've pushed the GRO-disable change to jool for now, until we find a better solution.

GoetzGoerisch, 2024-10-04:
Thanks, this provides a workaround for NAT64.

But what about NAT66 here it is currently even worse from my inital retesting? But asymmetrical, downstream is slower than upstream.

everything411, 2024-10-18:
use 23.05 please, nat66 works well for 23.05. @CTCD

devastgh, 2025-01-07:
Same issue on 24.10.0-rc4 with NAT66 running on x86/64.

reusu, 2025-01-09:
Mark, hopes to resolve the NAT66 issue during the RC phase.

GoetzGoerisch, 2025-01-17:
NAT64 performance is stable with `-rc2` `-rc5` with `mt7621` but still below 5.15.

@nbd168 could the jool patch also help with NAT66 as a workaround?

openwgcf, 2025-02-06:
The 24.10 stable series has already been released. Is there no plan to fix the NAT66 issue?

devastgh, 2025-02-13:
> The 24.10 stable series has already been released. Is there no plan to fix the NAT66 issue?

I agree, that this issue should've had a higher priority.

Noltari, 2025-02-14:
Friendly reminder that this is an open source project in which developers decide what they want to put their free time into.
Anyone who considers that this issue should have had a higher priority is simply wrong.
There are always lots of things to do in open source projects and stating things like this only discourages developers from contributing. See for example the recent asahi linux case with marcan.
Therefore I want to kindly ask everyone to avoid statements like this in the future.
Thanks for your understanding.

Headcrabed, 2025-02-15:
@nbd168 Can you do something with this?

nbd168, 2025-02-15:
I just ran some tests with IPv6 NAT enabled, and I couldn't reproduce any throughput/stability issues. Could somebody affected by this issue please send me some information about what network config + installed packages you're using?

GoetzGoerisch, 2025-02-16:
NAT64: Setup Option 2 like detailed [here](https://openwrt.org/docs/guide-user/network/ipv6/nat64?s[]=nat64). With your disabling of GRO, stable but below the throughput of kernel 5.15.
Don't have exact figures  between kernels here. Current performance see below.  

Last time I tryed NAT66 setup according to [here](https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6?s[]=nat66)
Performance was only 10% of NAT44 or IPv6 routing. 

Currently I have no HW/setup to conduct tests for NAT66.

NAT44/IPv6Routing/NAT64 comparison:

- Single http download 1GByte sparse file with wget on wired host behind router.

Target: MT7621 (HW flow offloading support)
OpenWrt: 24.10.0
PacketSteering: all CPU (2)
Steering Flows: 128

FlowOffloading: off
NAT44: 378 Mbps
IPv6 routing: 334 Mbps
NAT64: 163 Mbps

FlowOffloading: SW
NAT44: 402 Mbps
IPv6 routing: 459 Mbps
NAT64: 363 Mbps

FlowOffloading: HW
NAT44: 919 Mbps
IPv6 routing: 917 Mbps
NAT64: 316 Mbps

nbd168, 2025-02-16:
Are clients on wifi affected in a similar way, or just ethernet?

pedorich-n, 2025-02-16:
I am still experiencing this. However, I switched from NAT66 to Relay mode (my ISP only provides /64), so I'm not entirely sure it's the same issue. 

Here are some speedtest results (all IPv6):
1. Control result. Wired into my ISP's box: https://librespeed.org/results/?id=10e97qz (262Mbps / 455Mbps)
2. Wired to the router. Relay mode: https://librespeed.org/results/?id=0zd9ldl (37Mbps / 6.25Mbps)
3. Wired to the router. Relay mode. GRO off[^1]: https://librespeed.org/results/?id=0wqub21 (260Mbps / 7.3Mbps)

My client behaves about the same on both Ethernet and Wifi.

<details>

  <summary>Board Info</summary>

```
  {
        "kernel": "6.6.73",
        "hostname": "GL-MT6000",
        "system": "ARMv8 Processor rev 4",
        "model": "GL.iNet GL-MT6000",
        "board_name": "glinet,gl-mt6000",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "mediatek/filogic",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
   }
```

</details>

<details>

  <summary>/etc/config/network</summary>

```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd6f:3b40:15aa::/48'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'lan5'

config interface 'wan'
        option device 'eth1'
        option proto 'dhcp'

config interface 'wan6'
        option device 'eth1'
        option proto 'dhcpv6'
        option sourcefilter '0'
        option reqaddress 'try'
        option reqprefix 'auto'
        option norelease '1'

[... 526 lines cut ...]
ucode - 2025.02.10~a8a11aea-r1
ucode-mod-fs - 2025.02.10~a8a11aea-r1
ucode-mod-html - 1
ucode-mod-math - 2025.02.10~a8a11aea-r1
ucode-mod-nl80211 - 2025.02.10~a8a11aea-r1
ucode-mod-rtnl - 2025.02.10~a8a11aea-r1
ucode-mod-ubus - 2025.02.10~a8a11aea-r1
ucode-mod-uci - 2025.02.10~a8a11aea-r1
ucode-mod-uloop - 2025.02.10~a8a11aea-r1
uhttpd - 2023.06.25~34a8a74d-r4
uhttpd-mod-ubus - 2023.06.25~34a8a74d-r4
urandom-seed - 3
urngd - 2023.11.01~44365eb1-r1
usign - 2020.05.23~f1f65026-r1
vim - 9.0-r1
wifi-scripts - 1.0-r1
wireless-regdb - 2024.10.07-r1
wpad-basic-mbedtls - 2024.09.15~5ace39b0-r2
xtables-nft - 1.8.10-r1
zlib - 1.3.1-r1
```

</details>

[^1]: `for dev in eth0 eth1 br-lan; do ethtool -K $dev rx-gro-list off; done`

devastgh, 2025-02-16:
> I just ran some tests with IPv6 NAT enabled, and I couldn't reproduce any throughput/stability issues. Could somebody affected by this issue please send me some information about what network config + installed packages you're using?

I'm using x86, with intel i226v nics. I'm using pbr to direct specific traffic to a wireguard tunnel. The tunnel interface firewall settings have nat66 and mss clamping enabled. Whenever i try speedtest (ipv6) over this tunnel, i get roughly 40-50mbit, while if using ipv4 i get my full ~940mbit speed. Tried with iperf3 as well just for reference, and that was even worse, like ~10mbit per connection.

Shine-, 2025-02-18:
I just tested NAT66 (`uci set firewall.@zone[1].masq6=1`) in 24.10 @ 315facfce6dc13d6ec1993db1e16532cadcfcaaa in a DHCPv6-PD environment using iperf3 on a fresh setup (`firstboot -y; reboot`).

On ath79: 
- Issue does not occur, speed always as expected, regardless whether SW offload or no offloading, pure routing or NAT66 (masq6).

On mt7621:
- Pure routing: Wirespeed. regardless whether no offloading, SW or HW offload.
- NAT66 (masq6): <10MBit both directtions, regardless whether no offloading, SW or HW offload.

On mt7621 with `rx-gro-list off` on all I/F:
- Pure routing: Wirespeed. regardless whether no offloading, SW or HW offload.
- NAT66 (masq6) with HW offload: Download wirespeed, Upload ~400MBit/s
- NAT66 (masq6) with SW offload: Download wirespeed, Upload ~750MBit/s
- NAT66 (masq6) without offload: Download wirespeed, Upload ~400MBit/s

What's striking is that iperf3 will hang for about 30 seconds before starting, in all test cases, but only on mt7621, not on ath79.

nbd168, 2025-02-18:
Thanks for the update. When I find the time, I will run some tests with MT7621 instead of MT7622. Maybe I can reproduce it there.

everything411, 2025-02-19:
gmac1 muxing is not related to the perforamance degradation, my device has gmac1 muxing and I only observed NAT performance improvement, which is bidirectional wirespeed, compared to your results with `rx-gro-list off`, but with `rx-gro-list on`, NAT66 performance is poor, the same results as yours.

nbd168, 2025-02-24:
FYI, I was able to reproduce the issue on MT7621. I'm working on debugging/fixing it now.

closed 2025-02-24 by openwrt-bot