openwrt/openwrt #15885: unable to compile rust(无法编译rust)

forensaxgh, 2024-07-06:
### Describe the bug

Previously, it was possible to compile successfully, but suddenly one day after that, it prompted a compilation failure. Download url404
以前都是能编译成功的，突然有一天之后就提示编译失败，下载404

### OpenWrt version

v23.05.3

### OpenWrt release

v23.05.3

### OpenWrt target/subtarget

v23.05.3

### Device

x86_64

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

forensaxgh, 2024-07-06:
```
make[3]: Entering directory '/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/feeds/packages/lang/rust'
CARGO_HOME=/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/dl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/rust-1.75.0=rust-1.75.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" OPENWRT_RUSTC_BOOTSTRAP_CACHE=/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/dl/rustc python3 /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/x.py --build-dir /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/build --config /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
Building bootstrap
downloading https://ci-artifacts.rust-lang.org/rustc-builds/82e1608dfa6e0b5569232559e3d385fea5a93112/rust-dev-1.75.0-x86_64-unknown-linux-gnu.tar.xz
curl: (22) The requested URL returned error: 404
ERROR: failed to download llvm from ci

    HELP: old builds get deleted after a certain time
    HELP: if trying to compile an old commit of rustc, disable `download-ci-llvm` in config.toml:

    [llvm]
    download-ci-llvm = false
    
Build completed unsuccessfully in 0:00:00
make[3]: *** [Makefile:110: /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/.built] Error 1
make[3]: Leaving directory '/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.30#0.15#0.79
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:127: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt'
make[1]: *** [package/Makefile:123: /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt'
make: *** [/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/include/toplevel.mk:232: world] Error 2
```

1715173329, 2024-07-07:
Fixed by https://github.com/openwrt/packages/pull/24531.

forensaxgh, 2024-07-08:
> Fixed by [openwrt/packages#24531](https://github.com/openwrt/packages/pull/24531).
[Github Actions Job](https://github.com/Forensax/ActionsCompileOpenWRT/actions/runs/9830052534/job/27135800024
)new compile logs here,still report error when compile rust
`make[3]: Entering directory '/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/feeds/packages/lang/rust'
CARGO_HOME=/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/dl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/rust-1.75.0=rust-1.75.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" OPENWRT_RUSTC_BOOTSTRAP_CACHE=/home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/dl/rustc python3 /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/x.py --build-dir /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/build --config /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
Building bootstrap
downloading https://ci-artifacts.rust-lang.org/rustc-builds/82e1608dfa6e0b5569232559e3d385fea5a93112/rust-dev-1.75.0-x86_64-unknown-linux-gnu.tar.xz
curl: (22) The requested URL returned error: 404
ERROR: failed to download llvm from ci

    HELP: old builds get deleted after a certain time
    HELP: if trying to compile an old commit of rustc, disable `download-ci-llvm` in config.toml:

    [llvm]
    download-ci-llvm = false
    
Build completed unsuccessfully in 0:00:00
make[3]: *** [Makefile:110: /home/runner/work/ActionsCompileOpenWRT/ActionsCompileOpenWRT/openwrt/build_dir/target-x86_64_musl/host/rustc-1.75.0-src/.built] Error 1`

closed 2024-07-24 by forensaxgh