openwrt/openwrt #15888: Create lxc container failed on tar extract failure (Directory renamed before its status could be extracted)

Goodwu, 2024-07-06:
### Describe the bug

Add "--absolute-names" to tar command in /usr/share/lxc/templates/lxc-download can solve this problem.
Just like:
`tar --absolute-names --numeric-owner -xpJf "${LXC_CACHE_PATH}/rootfs.tar.xz" -C "${LXC_ROOTFS}"`

The following is part of the error message when I execute the original tar extract command manually.
```
tar: ./etc/modules-load.d: Directory renamed before its status could be extracted
tar: ./etc/dhcp/dhclient-exit-hooks.d: Directory renamed before its status could be extracted
tar: ./etc/dhcp/dhclient-enter-hooks.d: Directory renamed before its status could be extracted
tar: ./etc/dhcp: Directory renamed before its status could be extracted
tar: ./etc/alternatives: Directory renamed before its status could be extracted
tar: ./etc: Directory renamed before its status could be extracted
tar: Exiting with failure status due to previous errors
root@OpenWrt:/opt/lxc# which tar
/bin/tar
root@OpenWrt:/opt/lxc# ls -l /bin/tar
lrwxrwxrwx    1 root     root            20 Jul  6 00:01 /bin/tar -> /usr/libexec/tar-gnu
```

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

sunxi/cortexa53

### Device

FriendlyARM NanoPi NEO 2

### Image kind

Official downloaded image

### Steps to reproduce

Install luci-app-lxc and create debian container from debian bookworm template.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Goodwu, 2024-07-06:
Here's a reference: https://github.com/ot4i/ace-docker/issues/33

Goodwu, 2024-07-06:
> Invalid Version reported. `` Is this from a clean repository?

What's the problem? This is official release!

brada4, 2024-07-06:
```
ls -l /bin/tar
lrwxrwxrwx    1 root     root             7 Mar 22 22:09 /bin/tar -> busybox
```

Goodwu, 2024-07-07:
> root@OpenWrt:/opt/lxc# ls -l /bin/tar
> lrwxrwxrwx    1 root     root            20 Jul  6 00:01 /bin/tar -> /usr/libexec/tar-gnu

gnu tar
```
root@OpenWrt:/opt/lxc# ls -l /bin/tar
lrwxrwxrwx    1 root     root            20 Jul  6 00:01 /bin/tar -> /usr/libexec/tar-gnu
```

Goodwu, 2024-07-07:
Additional info about mount.
```
root@OpenWrt:/opt/lxc# mount
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/loop0 on /overlay type f2fs (rw,lazytime,noatime,background_gc=on,nodiscard,no_heap,user_xattr,inline_xattr,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,alloc_mode=default,checkpoint_merge,fsync_mode=posix,discard_unit=block)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work,xino=off)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
root@OpenWrt:/opt/lxc#
```

brada4, 2024-07-07:
Question is whether problem is present with busybox tar, or with gnu tar only?

Goodwu, 2024-07-08:
> tar --absolute-names --numeric-owner -xpJf

busybox tar does not support the "--numeric-owner" argument.
```
root@OpenWrt:~# busybox tar --numeric-owner -xpJf /var/cache/lxc//download/debian/bookworm/arm64//default/rootfs.tar.xz -C /opt/lxc/debian/rootfs1
tar: unrecognized option: numeric-owner
BusyBox v1.36.1 (2024-03-22 22:09:42 UTC) multi-call binary.

Usage: tar c|x|t [-zahvokO] [-f TARFILE] [-C DIR] [-T FILE] [-X FILE] [FILE]...

Create, extract, or list files from a tar file

	c	Create
	x	Extract
	t	List
	-f FILE	Name of TARFILE ('-' for stdin/out)
	-C DIR	Change to DIR before operation
	-v	Verbose
	-O	Extract to stdout
	-o	Don't restore user:group
	-k	Don't replace existing files
	-z	(De)compress using gzip
	-a	(De)compress based on extension
	-h	Follow symlinks
	-T FILE	File with names to include
	-X FILE	File with glob patterns to exclude
root@OpenWrt:~#
```

Goodwu, 2024-07-08:
> Question is whether problem is present with busybox tar, or with gnu tar only?

Does not support "-J" argument either. So lxc requires gnu tar.
```
root@OpenWrt:~# busybox tar -xpJf /var/cache/lxc//download/debian/bookworm/arm64//default/rootfs.tar.xz -C /opt/lxc/debian/rootfs1
tar: unrecognized option: J
```

brada4, 2024-07-08:
You can create proper report with package versions if it is still broken
https://github.com/openwrt/packages/commit/3e331ddcc08a6f5031427e846215628082cd73f5
Your assumption that your custom setup should be fixed breaking default is selfish and wrong.

Goodwu, 2024-07-10:
> You can create proper report with package versions if it is still broken [openwrt/packages@3e331dd](https://github.com/openwrt/packages/commit/3e331ddcc08a6f5031427e846215628082cd73f5) Your assumption that your custom setup should be fixed breaking default is selfish and wrong.

Sorry, I don't understand what do you mean totally.
1. Which part of my setup is "custom"? I downloaded the firmware from official site, installed the lxc related packages, enter the lxc container name and choose the debian bookwork template, then click add button.
2. You mean this is not a bug or should not be fixed?
3. Which part of my words is selfish and wrong? I encountered this bug, found some information from net, add the info as reference. Which part is selfish and wrong?

Goodwu, 2024-07-10:
These are lxc related package versions, and havn't found upgrades to lxc package yet.
```
root@OpenWrt:~# opkg list-installed|grep lxc
liblxc - 5.0.3-1
luci-app-lxc - git-21.102.30274-c6d196c
luci-i18n-lxc-zh-cn - git-24.180.72161-85dec60
lxc - 5.0.3-1
lxc-attach - 5.0.3-1
lxc-common - 5.0.3-1
lxc-configs - 5.0.3-1
lxc-console - 5.0.3-1
lxc-create - 5.0.3-1
lxc-hooks - 5.0.3-1
lxc-info - 5.0.3-1
lxc-ls - 5.0.3-1
lxc-start - 5.0.3-1
lxc-stop - 5.0.3-1
lxc-templates - 5.0.3-1
rpcd-mod-lxc - 20201208
```
I also tried the tar command line from lxc package 6.0.1, here's the result.
gnu-tar:
```
root@OpenWrt:~# tar --numeric-owner --xattrs-include='*' -xpJf /var/cache/lxc//download/debian/bookworm/arm64//default/rootfs.tar.xz -C /opt/lxc/debian/rootfs1
tar: XATTR support is not available
tar: ./usr/share/zoneinfo/right/US: Directory renamed before its status could be extracted
tar: ./usr/share/zoneinfo/right/Mexico: Directory renamed before its status could be extracted
tar: ./usr/share/zoneinfo/right/Europe: Directory renamed before its status could be extracted
tar: ./usr/share/zoneinfo/right/Chile: Directory renamed before its status could be extracted
tar: ./usr/share/zoneinfo/right/Canada: Directory renamed before its status could be extracted
tar: ./usr/share/zoneinfo/right/Brazil: Directory renamed before its status could be extracted
tar: ./usr/share/zoneinfo/right/Atlantic: Directory renamed before its status could be extracted
---Following with more "Directory renamed before its status could be extracted" error messages.
```
bsdtar:
```
root@OpenWrt:~# bsdtar --numeric-owner -xpJf /var/cache/lxc//download/debian/bookworm/arm64//default/rootfs.tar.xz -C /opt/lxc/debian/rootfs1
./usr/bin/ping: Cannot restore extended attributes on this system: No such file or directory
bsdtar: Error exit delayed from previous errors.
root@OpenWrt:~#
```
It seems that some one didn't do enough test.

brada4, 2024-07-10:
default tar is busybox, bsdtar and gnu tar is your xchoice.

Goodwu, 2024-07-10:
> default tar is busybox, bsdtar and gnu tar is your xchoice.

No. If you have read my post, you should have know that busybox tar behaves even worse.
busybox tar **DOES NOT** work at all. It even does not support extracting .xz file.
And gnu tar is required by luci-app-lxc:
```
root@OpenWrt:~# opkg remove tar
No packages removed.
Collected errors:
 * print_dependents_warning: Package tar is depended upon by packages:
 * print_dependents_warning: 	luci-app-lxc
 * print_dependents_warning: These might cease to work if package tar is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.
root@OpenWrt:~#
```

https://github.com/openwrt/openwrt/issues/15888#issuecomment-2213641346
https://github.com/openwrt/openwrt/issues/15888#issuecomment-2213650754

Goodwu, 2024-07-10:
> You can create proper report with package versions if it is still broken [openwrt/packages@3e331dd](https://github.com/openwrt/packages/commit/3e331ddcc08a6f5031427e846215628082cd73f5) Your assumption that your custom setup should be fixed breaking default is selfish and wrong.

I am very disappointed that you behaved in an unprofessional, unfriendly and unfriendly manner.

Goodwu, 2024-07-10:
@graysky2 Could you please take a look at this bug?

Goodwu, 2024-07-10:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1728489

brada4, 2024-07-10:
Fill the bug in right place with package name and versions in first post. 
https://openwrt.org/bugs

graysky2, 2024-07-10:
The issue is with my rebase of the patches for the bump: https://github.com/openwrt/packages/commit/3e331ddcc08a6f5031427e846215628082cd73f5#diff-02fb82400a6ae4aff7d4daca6e2b100f25d318c3aedf0f050a3d9e8c82b59399R25

> Add "--absolute-names" to tar command in /usr/share/lxc/templates/lxc-download can solve this problem.

@Goodwu - can you verify that adding that argument to the tar command on lines 384 and 386 of `/usr/share/lxc/templates/lxc-download` fixes the issue?

graysky2, 2024-07-10:
There is a lot going on in this report.  To keep things on topic the concern is that the patch I rebased is causing breakage and that there are multiple potential sources of `tar`.  Am I correct?

Test using `/bin/tar` supplied by busybox (the default as @brada4 mentioned:
```
# opkg search /bin/tar  
busybox - 1.36.1-r1

# lxc-create -t download -n test -- --dist archlinux --release current --arch amd64
Downloading the image index
Downloading the rootfs
Downloading the metadata
The image cache is now ready
Unpacking the rootfs
tar: XATTR support is not available

---
You just created an Archlinux  x86_64 (20240710_04:18) container.
```

I am thinking that the warning about extended attributes is related to busybox's tar not to my filesystem because the data partition where the lxc was decompress is ext4 and:
```
# cat /proc/fs/ext4/nvme0n1p4/options |grep xat
user_xattr
```
Adding `--absolute-names` to the command does not solve that.

@Goodwu - what is the filesystem type on which your lxc?  You can see the path within `/etc/lxc/lxc.conf` for example on my system:
```
# cat /etc/lxc/lxc.conf    
lxc.lxcpath = /mnt/data/lxc
```

graysky2, 2024-07-10:
There is a lot going on in this report.  To keep things on topic the concern is that the patch I rebased is causing breakage and that there are multiple potential sources of `tar`.  Am I correct?

Test using `/bin/tar` supplied by busybox (the default as @brada4 mentioned) can be a little deceptive:
```
# opkg search /bin/tar  
busybox - 1.36.1-r1
```
...yet that is just the symlink as @Goodwu showed.  The target is actually supplied by tar:
```
# opkg search /usr/libexec/tar-gnu
tar - 1.35-r1
```
In any case, that is the default and I am able to successfully decompress without the errors you are showing to your overlay mount which I highly discourage you from doing.  Instead, mount a filesystem via USB or other means for the lxc and define the mount point you choose in `/etc/lxc/lxc.conf` like this:
```
lxc.lxcpath = /mnt/data/lxc
```

Here is what it looks like for me:
```
# lxc-create -t download -n test -- --dist archlinux --release current --arch amd64
Downloading the image index
Downloading the rootfs
Downloading the metadata
The image cache is now ready
Unpacking the rootfs
tar: XATTR support is not available

---
You just created an Archlinux  x86_64 (20240710_04:18) container.
```

Digging into the defaults for tar, there is neither ACL nor xattr enabled by default, so the warning above makes sense.

@Goodwu - can you please verify that if you edit `/usr/share/lxc/templates/lxc-download` and add that argument to line 386, that you get the expected behavior?

graysky2, 2024-07-10:
I just build tar with both ACL and xattr and when I repeated the lxc-create to my ext4 filesystem:
```
# lxc-create -t download -n test -- --dist archlinux --release current --arch amd64
Using image from local cache
Unpacking the rootfs
tar: setxattrat: Cannot set 'security.capability' extended attribute for file './usr/bin/newgidmap': Not supported
tar: setxattrat: Cannot set 'security.capability' extended attribute for file './usr/bin/newuidmap': Not supported
tar: setxattrat: Cannot set 'security.capability' extended attribute for file './usr/bin/ping': Not supported
tar: setxattrat: Cannot set 'security.capability' extended attribute for file './usr/bin/rcp': Not supported
tar: setxattrat: Cannot set 'security.capability' extended attribute for file './usr/bin/rlogin': Not supported
tar: setxattrat: Cannot set 'security.capability' extended attribute for file './usr/bin/rsh': Not supported
tar: setxattrat: Cannot set 'system.posix_acl_default' extended attribute for file './var/lib/tpm2-tss/system/keystore': Not supported
tar: setxattrat: Cannot set 'system.posix_acl_access' extended attribute for file './var/log/journal': Not supported
tar: setxattrat: Cannot set 'system.posix_acl_default' extended attribute for file './var/log/journal': Not supported

---
You just created an Archlinux  x86_64 (20240710_04:18) container.
```

I get the same with or without the suggested `--absolute-names`

Goodwu, 2024-07-12:
> There is a lot going on in this report. To keep things on topic the concern is that the patch I rebased is causing breakage and that there are multiple potential sources of tar. Am I correct?
No, it's not about your patch. I mentioned you only because I thought you might be one of the maintainers of the lxc package.

As you mentioned that I'm using on overlayfs.
I tested on a loop mounted f2fs and without problem, so it very likely is related to overlayfs like [this](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1728489)

>  can you please verify that if you edit /usr/share/lxc/templates/lxc-download and add that argument to line 386, that you get the expected behavior?
Yes, adding the argument can solve the problem.

> I get the same with or without the suggested --absolute-names
The `--absolute-names` argument only solves the `Directory renamed before its status could be extracted` problem.