openwrt/openwrt #15895: After flasing Archer C6U (EU) ver.1 - Bootloop

robinson-28, 2024-07-07:
### Describe the bug

U-Boot 1.1.3 (Dec 13 2021 - 15:33:52)

Board: Ralink APSoC DRAM:  128 MB
relocate_code Pointer at: 87f58000

Config XHCI 40M PLL
******************************
Software System Reset Occurred
******************************
flash manufacture id: 20, device id 40 18
Warning: un-recognized chip ID, please update bootloader!
*** Warning - bad CRC, using default environment

============================================
Ralink UBoot Version: 5.0.0.0
--------------------------------------------
ASIC MT7621A DualCore (MAC to MT7530 Mode)
DRAM_CONF_FROM: Auto-Detection
DRAM_TYPE: DDR3
DRAM bus: 16 bit
Xtal Mode=3 OCP Ratio=1/3
Flash component: SPI Flash
Date:Dec 13 2021  Time:15:33:52
============================================
THIS IS uboot
icache: sets:256, ways:4, linesz:32 ,total:32768
dcache: sets:256, ways:4, linesz:32 ,total:32768

##### The CPU freq = 880 MHZ ####
estimate memory size =128 Mbytes

Press '4' or 't' to break the booting process

Press 'x' to enter recovery web server                                                                                                                       0
nm_init:791
nm_initFwupPtnStruct:276
nm_lib_readPtnTable:738
[NM_Debug](nm_lib_readPtnTable) 00743: NM_PTN_TABLE_BASE = 0xfe0000
[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 920, requried len = 8192
[NM_Debug](nm_lib_readPtnTable) 00751: Reading Partition Table from NVRAM ... OK

[NM_Debug](nm_lib_readPtnTable) 00759: Parsing Partition Table ... OK

[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 2, requried len = 3
factory boot check integer ok.

3: System Boot system code via Flash.
## Booting image at bc040000 ...
   Image Name:   MIPS OpenWrt Linux-6.6.36
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    3307811 Bytes =  3.2 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 2, requried len = 3
[NM_Error](nm_api_checkTestMode) 00417: factory boot check testmode flag is not 1.

### test_mode=disable,boot_args:console=ttyS1,115200 root=/dev/mtdblock2 rootfstype=squashfs init=/etc/preinit mtdparts=raspi:256k(uboot),2048k(uImage),14016k@0x240000(rootfs),64k@0xff0000(ART) mem=128M test_mode=disable
## Transferring control to Linux (at address 80001000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 6.6.36 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r26870-4a2f712f85) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sat Jul  6 03:22:37 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[... 196 lines cut ...]
[    1.944751] Rebooting in 1 seconds..

### OpenWrt version

SNAPSHOT

### OpenWrt release

r26870-4a2f712f85

### OpenWrt target/subtarget

ramips/mt7621

### Device

Archer C6U

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robinson-28, 2024-07-07:
Yes, from clean repository.
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=tplink_archer-c6u-v1

brada4, 2024-07-07:
Does release work?

robinson-28, 2024-07-08:
No, this release does not work. Also earlier.

closed 2024-07-10 by openwrt-bot