openwrt/openwrt #15924: Device tree of Ruijie RG-EW3200GX PRO is out of date

asterwyx, 2024-07-11:
### Describe the bug

Firmware compiled from the master branch for Ruijie RG-EW3200GX PRO is not bootable cause the device tree is out of date. I've dumped my device tree and acquired the updated device tree. I'm not sure if I could contribute to this repo cause this is a version-specific problem. How do you manage multiple versions of device tree?

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Ruijie RG-EW3200GX PRO

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

RolandoMagico, 2024-07-14:
Hi @asterwyx ,
can you provide more information? What are the differences in the device tree? Are there different hardware revisions of this router which require different device trees?

asterwyx, 2024-08-05:
Yes, there are different hardware revisions of this router I think. In my fork of [openwrt](https://github.com/asterwyx/openwrt) branch [rg](https://github.com/asterwyx/openwrt/tree/rg) you can find my patch. It works for me now.

RolandoMagico, 2024-08-05:
Do you know which hardware version you're using?
There are already two variants supported in OpenWrt:
- The initial one using the memory layout here: https://github.com/asterwyx/openwrt/blob/rg/target/linux/mediatek/dts/mt7622-ruijie-rg-ew3200gx-pro.dts
- A model variant "E5" which uses a diferent memory layout: https://github.com/asterwyx/openwrt/blob/rg/target/linux/mediatek/dts/mt7622-reyee-ax3200-e5.dts

So if you know your hardware revision, you can create a pull request similar to the one here:
https://github.com/openwrt/openwrt/commit/7dbac3433fef1af0f76a515e962db450c3407c19