openwrt/openwrt #15925: OPKG uninstall Luci-pppoe package deletes  whole Luci Interface

palbiez, 2024-07-11:
### Describe the bug

Hi all,
I wanted to uninstall onto my D-Link 1325 Wifi Dump AP the unused PPPOE packages. 
I uninstalled via luci.

After uninstalling luci-proto-ppp incl. ununsed packages the whole Luci will be removed. 

Before (from my second AP) this packages are installed
`root@dapx1860:~# opkg list-installed
base-files - 1554-r23809-234f1a2efa
busybox - 1.36.1-1
ca-bundle - 20230311-1
cgi-io - 2022-08-10-901b0f04-21
dnsmasq - 2.90-2
dropbear - 2022.82-6
firewall4 - 2023-09-01-598d9fbb-1
fstools - 2023-02-28-bfe882d5-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-2
hostapd-common - 2023-09-08-e5ccbfc6-6
iw - 5.19-1
iwinfo - 2023-07-01-ca79f641-1
jansson4 - 2.14-3
jshn - 2023-05-23-75a3b870-1
jsonfilter - 2024-01-23-594cfa86-1
kernel - 5.15.150-1-bee655e7f41679981b85a223f3aba35b
kmod-cfg80211 - 5.15.150+6.1.24-3
kmod-crypto-acompress - 5.15.150-1
kmod-crypto-aead - 5.15.150-1
kmod-crypto-ccm - 5.15.150-1
kmod-crypto-cmac - 5.15.150-1
kmod-crypto-crc32c - 5.15.150-1
kmod-crypto-ctr - 5.15.150-1
kmod-crypto-gcm - 5.15.150-1
kmod-crypto-gf128 - 5.15.150-1
kmod-crypto-ghash - 5.15.150-1
kmod-crypto-hash - 5.15.150-1
kmod-crypto-hmac - 5.15.150-1
kmod-crypto-manager - 5.15.150-1
kmod-crypto-null - 5.15.150-1
kmod-crypto-rng - 5.15.150-1
kmod-crypto-seqiv - 5.15.150-1
kmod-crypto-sha512 - 5.15.150-1
kmod-gpio-button-hotplug - 5.15.150-3
kmod-hwmon-core - 5.15.150-1
kmod-leds-gpio - 5.15.150-1
kmod-lib-crc-ccitt - 5.15.150-1
kmod-lib-crc32c - 5.15.150-1
kmod-lib-lzo - 5.15.150-1
kmod-mac80211 - 5.15.150+6.1.24-3
kmod-mt76-connac - 5.15.150+2023-09-11-f1e1e67d-1
kmod-mt76-core - 5.15.150+2023-09-11-f1e1e67d-1
kmod-mt7915-firmware - 5.15.150+2023-09-11-f1e1e67d-1
kmod-mt7915e - 5.15.150+2023-09-11-f1e1e67d-1
kmod-nf-conntrack - 5.15.150-1
kmod-nf-conntrack6 - 5.15.150-1
kmod-nf-flow - 5.15.150-1
kmod-nf-log - 5.15.150-1
kmod-nf-log6 - 5.15.150-1
kmod-nf-nat - 5.15.150-1
kmod-nf-reject - 5.15.150-1
kmod-nf-reject6 - 5.15.150-1
kmod-nfnetlink - 5.15.150-1
kmod-nft-core - 5.15.150-1
kmod-nft-fib - 5.15.150-1
kmod-nft-nat - 5.15.150-1
kmod-nft-offload - 5.15.150-1
kmod-ppp - 5.15.150-1
kmod-pppoe - 5.15.150-1
[... 237 lines cut ...]
ramips/mt76x8

### Device

D-Link DAP-1325 A1

### Image kind

Official downloaded image

### Steps to reproduce

Uninstall  luci-proto-ppp ppp pppoe via luci

### Actual behaviour

Luci is uninstalled

### Expected behaviour

Luci won't uninstalled only PPP package for luci uninstalled

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pkgadd, 2024-07-11:
That is not a bug, but expected and intended behaviour. luci/ luci-ssl is a metapackage depending on luci-proto-ppp, you can't have the former without the later (but as being an empty metapackage, removing the former won't actually remove the webinterface either).

jow-, 2024-07-11:
It will just remove empty meta collection packages.

closed 2024-07-11 by jow-