openwrt/openwrt #15929: Xiaomi AX3600 - ath11k crash and "Out of memory"

MisiekDP, 2024-07-12:
### Describe the bug

When using the router (when browsing the web), crashes occur in the form of: switching off the WiFi network, disconnection of the connection. If you go through the router's logbook, you will notice all sorts of errors, including those with the message "Out of memory" and the killing of given processes.
Using a package that allows the logging of component usage, it is possible to see increased RAM usage up to the time of failure. There was also a situation where, due to lack of memory, it was also impossible to log into the LUCI GUI panel (the only solution to enable WiFi was to restart the device).

Memory usage during the error:
[image]

CPU usage during the error:
[image]

Signal quality:
[image]

Kernel logs:
```
[    9.190032] ath10k_pci 0000:01:00.0: qca9887 hw1.0 target 0x4100016d chip_id 0x004000ff sub 0000:0000
[    9.190088] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[    9.198407] ath10k_pci 0000:01:00.0: firmware ver 10.1-ct-87-__fW-022-ecad3248 api 2 features wmi-10.x,has-wmi-mgmt-tx,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,txrate-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 61a92df5
[    9.233886] ath10k_pci 0000:01:00.0: Loading BDF type 0
[    9.234107] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id N/A crc32 b4ce36fa
[   10.104510] ath10k_pci 0000:01:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[   10.111937] ath10k_pci 0000:01:00.0: wmi print 'P 128 V 8 T 410'
[   10.111983] ath10k_pci 0000:01:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[   10.117287] ath10k_pci 0000:01:00.0: wmi print 'alloc rem: 25560 iram: 24780'
[   10.168326] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 2 htt-op 2 cal nvmem max-sta 128 raw 0 hwcrypto 1
[   10.168688] ath10k_pci 0000:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   10.307586] ath: EEPROM regdomain: 0x0
[   10.307594] ath: EEPROM indicates default country code should be used
[   10.307598] ath: doing EEPROM country->regdmn map search
[   10.307603] ath: country maps to regdmn code: 0x3a
[   10.307607] ath: Country alpha2 being used: US
[   10.307611] ath: Regpair used: 0x3a
[   10.318788] ath11k c000000.wifi: ipq8074 hw2.0
[   10.318824] ath11k c000000.wifi: FW memory mode: 1
[   10.322440] remoteproc remoteproc0: powering up cd00000.q6v5_wcss
[   10.327112] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
[   11.558989] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[   11.564891] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.603292] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[   11.610769] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   11.610824] ath11k c000000.wifi: fw_version 0x290c84a5 fw_build_timestamp 2023-03-25 07:34 fw_build_id WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1
[   11.824869] ath11k c000000.wifi: htt event 48 not handled
[   14.206812] br-lan: port 1(lan1) entered blocking state
[   14.206861] br-lan: port 1(lan1) entered disabled state
[   14.211110] device lan1 entered promiscuous mode
[   14.224016] br-lan: port 2(lan2) entered blocking state
[   14.224070] br-lan: port 2(lan2) entered disabled state
[   14.228417] device lan2 entered promiscuous mode
[   14.237737] br-lan: port 3(lan3) entered blocking state
[   14.238138] br-lan: port 3(lan3) entered disabled state
[   14.243347] device lan3 entered promiscuous mode
[   14.367869] ath: EEPROM regdomain: 0x8268
[   14.367920] ath: EEPROM indicates we should expect a country code
[   14.370882] ath: doing EEPROM country->regdmn map search
[   14.377030] ath: country maps to regdmn code: 0x37
[   14.382336] ath: Country alpha2 being used: PL
[   14.386976] ath: Regpair used: 0x37
[   14.391364] ath: regdomain 0x8268 dynamically updated by user
[   14.796871] br-lan: port 4(phy1-ap0) entered blocking state
[   14.796921] br-lan: port 4(phy1-ap0) entered disabled state
[   14.801492] device phy1-ap0 entered promiscuous mode
[   14.806935] br-lan: port 4(phy1-ap0) entered blocking state
[   14.812029] br-lan: port 4(phy1-ap0) entered forwarding state
[   14.819772] device phy1-ap0 left promiscuous mode
[   14.823331] br-lan: port 4(phy1-ap0) entered disabled state
[   14.889297] br-lan: port 4(phy1-ap0) entered blocking state
[   14.889353] br-lan: port 4(phy1-ap0) entered disabled state
[   14.893982] device phy1-ap0 entered promiscuous mode
[   14.899494] br-lan: port 4(phy1-ap0) entered blocking state
[... 467 lines cut ...]
### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

1. Start up the router.
2. Use the internet.
3. After some random time (especially when using a high 250Mpbs connection), expect a rash of errors and "Out of memory" messages and some process to be killed.

### Actual behaviour

When using the router (when browsing the web), crashes occur in the form of: switching off the WiFi network, disconnection of the connection. If you go through the router's logbook, you will notice all sorts of errors, including those with the message "Out of memory" and the killing of given processes.

### Expected behaviour

I expect stable router operation without a rash of errors.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lorand-horvath, 2024-07-12:
You might want to try the latest snapshot version 26937 or newer, there are some fixes for ath11k.

edrikk, 2024-07-12:
Yes, snapshot will be significantly more stable based on my experience for the AX3600 vs. the 23.X branch.
The OOM issue for my has been significantly reduced due to the addition of the SMP Affinity script in snapshot.  You can also try running the script yourself on each reboot on 23.05.

https://github.com/openwrt/openwrt/issues/13562#issuecomment-1817573570

https://forum.openwrt.org/t/re-xiaomi-ax3600-ath11k-firmware-crash-qcom-q6v5-wcss-pil-cd00000-q6v5-wcss-fatal-error-received/180276/11

MisiekDP, 2024-07-13:
Wow, thanks for ideas guys @edrikk @lorand-horvath but could you tell me how to build this SNAPSHOT version with my own packages? All the time I used firmware selector with Attendedsysupgrade package but how to build SNAPSHOT?

edrikk, 2024-07-13:
I’m fairly sure ASU gives that option.  At least in advanced mode maybe?

https://openwrt.org/docs/guide-user/installation/attended.sysupgrade

But maybe first just try running the script on your current version after a reboot.

MisiekDP, 2024-07-13:
@edrikk I can send you later screenshots from Advanced mode later. I tried yesterday using ASU but I couldn't find any options to choose SNAPSHOT.

And yea, it is good idea to try script first - could you mąybe tell me how to run this script and where? Or maybe you have any instruction for this?
And how to check if this script is running? Should I get any logs of this script in Systm Log section?

danpawlik, 2024-07-13:
@MisiekDP in the snapshot release, auc has been replaced with owut, but if I remember correctly, command should be more less: `auc -c -b 23.05 -B snapshot`.
Another way is just to add desire packages on https://firmware-selector.openwrt.org/

AgustinLorenzo, 2024-07-14:
Hi boys,

For this failure:

```
[175506.181831] ath11k c000000.wifi: Spurious quick kickout for STA e8:db:84:e5:3c:56
[187480.659123] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 2
[189620.400508] ath11k c000000.wifi: Spurious quick kickout for STA e8:db:84:e5:3c:56

[228661.743942] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
[232841.819263] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
[234274.697449] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
```

@qosmio in the NSS repository added this patch and the bug disappeared, the patch comes from ath-next: https://github.com/qosmio/openwrt-ipq/blob/qualcommax-6.x-nss-wifi/package/kernel/mac80211/patches/nss/ath11k/453-ath11k-flush-management-frames-to-firmware-before-wa.patch

Maybe when @robimarko or @ansuel synchronize again with ath-next this patch will be already incorporated and the bug will disappear.

Regards, Agustin

MisiekDP, 2024-09-09:
@edrikk Could you tell me if placing the script in rc.local (local startup) is still required? I recently upgraded my AX3600 to OpenWrt 23.05.4 (r24012-d8dd03c46f) (stable release) and I'm wondering if this script is already permanently ‘sewn’ into the OpenWRT system or if it needs to be added manually in rc.local (I wouldn't want to duplicate it).