openwrt/openwrt #15942: util-linux build fails on riscv64

hauke, 2024-07-13:
### Describe the bug

The build of the util-linux packages fails for RiscV on Arch Linux.

I build OpenWrt master with this configuration:
```
CONFIG_TARGET_starfive=y
CONFIG_TARGET_starfive_generic=y
CONFIG_TARGET_starfive_generic_DEVICE_visionfive-v1=y
```

The build fails like this:
```
[313/686] Linking target more
FAILED: more 
riscv64-openwrt-linux-musl-gcc  -o more more.p/text-utils_more.c.o -L/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/usr/lib -L/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/lib -Wl,--as-needed -Wl,--no-undefined -fuse-ld=bfd -fuse-ld=bfd -znow -zrelro -Wl,--start-group lib/libcommon.a -fuse-ld=bfd -znow -zrelro /lib/libncursesw.so -Wl,--end-group
/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/lib/gcc/riscv64-openwrt-linux-musl/13.3.0/../../../../riscv64-openwrt-linux-musl/bin/ld.bfd: /lib/libncursesw.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:909: /home/hauke/openwrt/openwrt/build_dir/target-riscv64_riscv64_musl/util-linux-2.40.1/.built] Error 1
make[3]: Leaving directory '/home/hauke/openwrt/openwrt/package/utils/util-linux'
time: package/utils/util-linux/compile#20.58#4.85#24.24
    ERROR: package/utils/util-linux failed to build.
```

It wants to link against the host file and that fails. The file exists:
```
[hauke@hauke-p14s openwrt]$ ls -al /lib/libncursesw.so
lrwxrwxrwx 1 root root 16 May 25 08:43 /lib/libncursesw.so -> libncursesw.so.6
```

The log file in `build_dir/target-riscv64_riscv64_musl/util-linux-2.40.1/openwrt-build/meson-logs/meson-log.txt` shows that it used pkg-config to get the library and gets the correct ldflags:
```
Called: `/home/hauke/openwrt/openwrt/staging_dir/host/bin/pkg-config --libs ncursesw` -> 0
stdout:
-L/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/usr/lib -L/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -lncursesw
```

Later on it wants to use it from the host system:
```
/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/lib/gcc/riscv64-openwrt-linux-musl/13.3.0/../../../../riscv64-openwrt-linux-musl/bin/ld.bfd: /lib/libncursesw.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
```

The log file also contains this strange library part:
```
libraries: =/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/lib/gcc/riscv64-openwrt-linux-musl/13.3.0/:/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/lib/gcc/riscv64-openwrt-linux-musl/13.3.0/../../../../riscv64-openwrt-linux-musl/lib/riscv64-openwrt-linux-musl/13.3.0/:/home/hauke/openwrt/openwrt/staging_dir/toolchain-riscv64_riscv64_gcc-13.3.0_musl/lib/gcc/riscv64-openwrt-linux-musl/13.3.0/../../../../riscv64-openwrt-linux-musl/lib/:/lib64/lp64d/riscv64-openwrt-linux-musl/13.3.0/:/lib64/lp64d/:/usr/lib64/lp64d/riscv64-openwrt-linux-musl/13.3.0/:/usr/lib64/lp64d/:/lib/riscv64-openwrt-linux-musl/13.3.0/:/lib/:/usr/lib/riscv64-openwrt-linux-musl/13.3.0/:/usr/lib/
``` 
I compared this to an aarch64 build and an aarch64 build does not contain the host paths like `/usr/lib/`. I do not know what adds these host paths. 

On a Debian system it builds because it does not find a `libncursesw.so` on the host system. On Debian it is located in `/usr/lib/aarch64-linux-gnu/libncursesw.so` and that is not in the search path. The search path also looks exactly the same on Debian. 

### OpenWrt version

r26943-7f44f8d8d6

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

starfive/generic

### Device

visionfive-v1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

t-8ch, 2024-07-21:
I think it comes from `STARTFILE_PREFIX_SPEC` in `gcc/config/riscv/linux.h`
which ends up in `startfile_prefix_spec` in `gcc/specs`.

t-8ch, 2024-07-22:
It looks to me like the paths in `ncursesw.pc` are wrong.
They point to `$(TOOLCHAIN_DIR)`, but the actual libraries are in `$(STAGING_DIR)`.

The following change makes the build work for me:
```diff
diff --git a/package/libs/ncurses/Makefile b/package/libs/ncurses/Makefile
index 095805153327..0fcc38884014 100644
--- a/package/libs/ncurses/Makefile
+++ b/package/libs/ncurses/Makefile
@@ -181,6 +181,8 @@ define Build/InstallDev
        $(SED) 's,^\(prefix\|exec_prefix\)=.*,\1=$(STAGING_DIR)/usr,g' -e 's/$$$$INCS //g' \
                $(2)/bin/ncursesw6-config
        ln -sf $(STAGING_DIR)/host/bin/ncursesw6-config $(1)/usr/bin/ncursesw6-config
+       $(SED) 's,$(TOOLCHAIN_DIR),$(STAGING_DIR),g' \
+               $(1)/usr/lib/pkgconfig/ncursesw.pc
 endef
 
 define Host/Compile
```

Does this make sense?

hauke, 2024-07-27:
Thanks, fixing the `ncursesw.pc` file fixed the problem.

commit 91573ac1 (2024-07-28): ncurses: Fix path in ncursesw.pc

The file contains the the /usr/lib path from the toolchain directory and
not from the target directory. The /usr/lib directory for the toolchain
is empty and the shared library is not in the specified paths. On RISCV
the linker of util-linux was finding the libncursesw.so in my host
system, tried to link against it and failed. Fix the .pc file.

Fixes: #15942

Link: https://github.com/openwrt/openwrt/pull/16018
PR #16018 (2024-07-28): ncurses: Fix path in ncursesw.pc
The file contains the the /usr/lib path from the toolchain directory and not from the target directory. The /usr/lib directory for the toolchain is empty and the shared library is not in the specified paths. On RISCV the linker of util-linux was finding the libncursesw.so in my host system, tried to link against it and failed. Fix the .pc file.

Fixes: #15942
commit 461102d9 (2024-09-15): ncurses: Fix path in ncursesw.pc

The file contains the the /usr/lib path from the toolchain directory and
not from the target directory. The /usr/lib directory for the toolchain
is empty and the shared library is not in the specified paths. On RISCV
the linker of util-linux was finding the libncursesw.so in my host
system, tried to link against it and failed. Fix the .pc file.

Fixes: #15942

Link: https://github.com/openwrt/openwrt/pull/16018

(cherry picked from commit 91573ac145aa70a12b0984ec75507ac648569240)
Link: https://github.com/openwrt/openwrt/pull/16390
PR #16390 (2024-09-15): [23.05]: Misc backports for 23.03
 * ncurses: Fix path in ncursesw.pc
    
    The file contains the the /usr/lib path from the toolchain directory and
    not from the target directory. The /usr/lib directory for the toolchain
    is empty and the shared library is not in the specified paths. On RISCV
    the linker of util-linux was finding the libncursesw.so in my host
    system, tried to link against it and failed. Fix the .pc file.
    
    Fixes: #15942
    Co-authored-by: Thomas Weißschuh <thomas@t-8ch.de>
    Signed-off-by: Hauke Mehrtens <hauke@hauke-m.de>
    Link: https://github.co [... cut]
closed 2024-07-28 by openwrt-bot