openwrt/openwrt #15943: libgpiod broken

Josua-SR, 2024-07-14:
### Describe the bug

Using libgpiod example utilities, such as `gpioinfo` throws an unexpected error message:

```
root@OpenWrt:/# gpioinfo 
gpioinfo: error creating line iterator: Invalid argument
root@OpenWrt:/# gpioinfo /dev/gpiochip0
gpioinfo: error creating line iterator: Invalid argument
root@OpenWrt:/# gpiofind 1
gpiofind: error performing the line lookup: Invalid argument
root@OpenWrt:/# gpioget 0 2
gpioget: error reading GPIO values: Invalid argument
root@OpenWrt:/# ls -lh /dev/gpiochip*
crw-------    1 root     root      254,   0 Jul 14 14:58 /dev/gpiochip0
crw-------    1 root     root      254,   1 Jul 14 14:58 /dev/gpiochip1
crw-------    1 root     root      254,   2 Jul 14 14:58 /dev/gpiochip2
```

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

SolidRun Clearfog Pro A1

### Image kind

Official downloaded image

### Steps to reproduce

```
opkg update
opkg install gpiod-tools
gpioinfo
```

### Actual behaviour

```
root@OpenWrt:/# gpioinfo 
gpioinfo: error creating line iterator: Invalid argument
```

### Expected behaviour

All GPIOs available in the system, by `/dev/gpiochip*` are listed

### Additional info

See also https://zameermanji.com/blog/2022/1/18/ubuntu-21-10-makes-libgpiod-unusable/

Apparently there is a mismatch between kernel gpio character device api version, and libgpiod version.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

bropines, 2024-10-06:
Still not fixed

brilliantcontract, 2025-02-13:
Hardware: Raspberry Pi 3 Model B Rev 1.2
OpenWrt: OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-23.236.53405-fc638c8

```
gpioinfo gpiochip0
gpioinfo: error creating line iterator: Invalid argument
```

brilliantcontract, 2025-02-13:
But commands like `echo 1 > /sys/class/gpio/gpio26/value` works fire.