openwrt/openwrt #15947: wifi does not come up after DFS event

hauke, 2024-07-14:
### Describe the bug

I got a DFS event, the AP changes channels for the wifi network. Wifi came up fine, but the IP configuration on the 2. network is gone. 

Network configuration looks like this:
```
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd05:a874:10c1::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'wan.7'
	option proto 'pppoe'
	option username 'user'
	option password 'password'

config device 'lantv_dev'
	option type 'bridge'
	option name 'br-lantv'

config interface 'lantv'
	option proto 'static'
	option device 'br-lantv'
	option ipaddr '192.168.2.1'
	option netmask '255.255.255.0'
	option ip6assign '60

root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '120'
	option channels '94-144'
	option band '5g'
	option htmode 'HE80'
	option disabled '0'
	option country 'DE'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'haukeswlan'
	option sae_pwe 2
	option encryption 'sae-mixed'
	option key 'password1'

config wifi-iface 'tv_radio1'
	option device 'radio1'
	option network 'lantv'
	option mode 'ap'
	option ssid 'haukeswlan-tv'
	option sae_pwe 2
	option encryption 'sae-mixed'
	option key 'pssword2'
```

The bridges are looking like this:
```
root@OpenWrt:~# brctl show
bridge name	bridge id		STP enabled	interfaces
br-lan		7fff.c4411ef89b9f	no		lan4
							lan2
							wl0-ap0
							lan3
							lan1
							wl1-ap0
br-lantv		8000.c6411ef89ba1	no		wl1-ap1
root@OpenWrt:~# 
```

The `lantv` is only exposed as a wifi network to connect some IoT devices to. 

I got a DFS event today and changes channels. The main wifi is up again, but the lantv is not fully up.
```
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: wl1-ap0: DFS-RADAR-DETECTED freq=5600 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5610 cf2=0
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: wl1-ap0: DFS-NEW-CHANNEL freq=5500 chan=100 sec_chan=1
Sun Jul 14 19:47:41 2024 daemon.warn hostapd: DFS failed to schedule CSA (-22) - trying fallback
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: wl1-ap0: AP-DISABLED
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED aa:bb:cc:dd:ee:ff
Sun Jul 14 19:47:41 2024 kern.info kernel: [337397.832690] br-lantv: port 1(wl1-ap1) entered disabled state
Sun Jul 14 19:47:41 2024 kern.info kernel: [337397.880423] device wl1-ap1 left promiscuous mode
Sun Jul 14 19:47:41 2024 kern.info kernel: [337397.885166] br-lantv: port 1(wl1-ap1) entered disabled state
Sun Jul 14 19:47:41 2024 daemon.notice netifd: Interface 'lantv' is now down
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: nl80211: Failed to remove interface wl1-ap1 from bridge br-lantv: No such device
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: nl80211: deinit ifname=wl1-ap0 disabled_11b_rates=0
Sun Jul 14 19:47:41 2024 daemon.notice netifd: Interface 'lantv' is disabled
Sun Jul 14 19:47:41 2024 daemon.notice netifd: Network device 'wl1-ap1' link is down
Sun Jul 14 19:47:41 2024 daemon.notice netifd: bridge 'br-lantv' link is down
Sun Jul 14 19:47:41 2024 daemon.notice netifd: Interface 'lantv' has link connectivity loss
```

The SSID for lantv is broadcasted, but the interface has no ip address:
```
root@OpenWrt:~# ubus call network.interface.lantv status
{
	"up": false,
	"pending": false,
	"available": false,
	"autostart": true,
	"dynamic": false,
	"proto": "static",
	"device": "br-lantv",
	"data": {
		
	},
	"errors": [
		{
			"subsystem": "interface",
			"code": "DEVICE_CLAIM_FAILED"
		}
	]
}
root@OpenWrt:~# ip addr show dev br-lantv
22: br-lantv: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether c6:41:1e:f8:9b:a1 brd ff:ff:ff:ff:ff:ff
root@OpenWrt:~# 
```

### OpenWrt version

r24002+1-78858e5d6c

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Self-built image

### Steps to reproduce

Setup a guest network with only wifi networks like it is described here: https://openwrt.org/docs/guide-user/network/wifi/guestwifi/guest-wlan
Trigger a event which will reconfigure the network. 
The guest network is loose its network configuration. 

### Actual behaviour

No IP configuration on 2. network

### Expected behaviour

Normal network configuration on 2. network

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-14:
Any indications +/-1h indicating other radar event banning fallback?

hauke, 2024-07-14:
No other DFS events in the days before or the 2 hours after.
```
root@OpenWrt:~# logread |grep DFS
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: wl1-ap0: DFS-RADAR-DETECTED freq=5600 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5610 cf2=0
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: wl1-ap0: DFS-NEW-CHANNEL freq=5500 chan=100 sec_chan=1
Sun Jul 14 19:47:41 2024 daemon.warn hostapd: DFS failed to schedule CSA (-22) - trying fallback
Sun Jul 14 19:47:41 2024 daemon.notice hostapd: wl1-ap0: DFS-RADAR-DETECTED freq=5600 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5610 cf2=0
Sun Jul 14 19:47:42 2024 daemon.notice hostapd: wl1-ap0: interface state HT_SCAN->DFS
Sun Jul 14 19:47:42 2024 daemon.notice hostapd: wl1-ap0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
Sun Jul 14 19:48:47 2024 daemon.notice hostapd: wl1-ap0: DFS-CAC-COMPLETED success=1 freq=5500 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5530 cf2=0
Sun Jul 14 19:48:48 2024 daemon.notice hostapd: wl1-ap0: interface state DFS->ENABLED
Sun Jul 14 20:17:42 2024 daemon.notice hostapd: wl1-ap0: DFS-NOP-FINISHED freq=5580 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5580 cf2=0
Sun Jul 14 20:17:42 2024 daemon.notice hostapd: wl1-ap0: DFS-NOP-FINISHED freq=5600 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5600 cf2=0
Sun Jul 14 20:17:42 2024 daemon.notice hostapd: wl1-ap0: DFS-NOP-FINISHED freq=5620 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5620 cf2=0
Sun Jul 14 20:17:42 2024 daemon.notice hostapd: wl1-ap0: DFS-NOP-FINISHED freq=5640 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5640 cf2=0
root@OpenWrt:~# 
```

brada4, 2024-07-14:
I dont get this nor 2x radar event.
```
DFS failed to schedule CSA
```
And if all DFS channels ran out it goes DFS->DISABLED instantly after RADAR event, then redoes DFS after next NOP event no idea what those mean.

httpstorm, 2024-07-15:
@hauke 
A few years back, when I got a WRT3200ACM, the 5GHz radio did not start at all or took at least 5 minutes. So I had no choice but to do something about it. I disabled region reporting in the radio drivers, so I can configure the correct region in settings. I also created a patch for `wireless-regdb` which removes the DFS flag for my country. Since then the radio always starts instantly, the speed is about 768 Mbit/s. No more DFS issues. I live in a location which is far away from any radars. In case you like to give it a test, the branch name on my repository is `gvalkov`.

httpstorm, 2025-12-26:
I posted a firmware for [WRT3200ACM/2025-12-25.r32350](https://httpstorm.com/openwrt/WRT3200ACM/2025-12-25.r32350/) on my website. Use at your own risk.
- wireless-regdb: DFS is removed for some countries: 00 BG DE EE FR US
- the radios do not report region to avoid conflicts, the user can set their own
- reverted to `swconfig` interfaces `lan` `wan`, and removed DSA `lan1-4`
- **do not** import network configuration that expects `lan1-4`, the LAN interface is `lan`
- add my own software and tools: *httpstorm* web server is on port 83 webroot `/mnt`
- 5G modem support: QMI cellular `wwan0` `/dev/cdc-wdm0`
- iPhone tethering support: `eth0`
- https://github.com/httpstorm/openwrt/commits/gvalkov/
 
Wi-Fi benchmark on channel 100 using [gns](https://httpstorm.com/download/g-net-speed/) to MBP2019 16"
- RX 720 Mbit/s (90 MB/s)
- TX 680 Mbit/s (85 MB/s)

httpstorm, 2025-12-27:
@brada4 
I remember when I first got the WRT3200ACM it didn't want to use any of the DFS channels. Because radio0 and radio1 reported region 98 (invalid but translated to FR), radio2 US and I had a third value as my region. Eventually I got radio0 to start on channel 100 after 5 minutes of scanning, but this wasn't practical for testing, so I created the patches used in this firmware.

Now all radios come online instantly on any channel. I shared them here, because I thought they might be helpful for people who live far away from any radars, but struggle because of how things work. If my publication is not in place, please let me know, and I will remove it.