openwrt/openwrt #15959: ipq40xx: CPU hostapd Tainted

siwind, 2024-07-16:
### Describe the bug

After  update to the latest commit of master branch, my router report following log:

```
[   58.818619] qca8k-ipq4019 c000000.switch lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[   58.818773] br-lan: port 3(lan3) entered blocking state
[   58.825949] br-lan: port 3(lan3) entered forwarding state
[   61.104423] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   61.104499] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   61.139693] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   61.140669] ath10k_ahb a000000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
[   61.532719] ath10k_ahb a000000.wifi: rts threshold -1
[   61.533008] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   61.536833] ------------[ cut here ]------------
[   61.547833] WARNING: CPU: 3 PID: 1695 at backports-6.9.9/net/mac80211/main.c:270 ieee80211_do_open+0x4e8/0x5e0 [mac80211]
[   61.552544] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) wireguard pppox ppp_mppe ppp_generic nft_tproxy nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mac80211(O) libchacha20poly1305 curve25519_neon cfg80211(O) usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc poly1305_arm nfnetlink nf_tproxy_ipv6 nf_tproxy_ipv4 nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4
[   61.553585]  macvlan libcurve25519_generic hwmon crc_ccitt compat(O) chacha_neon asn1_decoder ntfs3 msdos ip6_udp_tunnel udp_tunnel oid_registry tun autofs4 nls_utf8 nls_iso8859_1 nls_cp437 xxhash_generic crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_arm sha1_arm_neon sha1_arm ghash_arm_ce cmac blake2b_generic arc4 usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 exfat btrfs xor xor_neon raid6_pq libcrc32c dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax tpm crc32c_generic encrypted_keys trusted
[   61.692792] CPU: 3 PID: 1695 Comm: hostapd Tainted: G           O       6.6.40 #0
[   61.714929] Hardware name: Generic DT based system
[   61.722484]  unwind_backtrace from show_stack+0x10/0x14
[   61.727164]  show_stack from dump_stack_lvl+0x40/0x4c
[   61.732296]  dump_stack_lvl from __warn+0x94/0xbc
[   61.737497]  __warn from warn_slowpath_fmt+0x158/0x15c
[   61.742188]  warn_slowpath_fmt from ieee80211_do_open+0x4e8/0x5e0 [mac80211]
[   61.747253]  ieee80211_do_open [mac80211] from ieee80211_open+0x58/0x7c [mac80211]
[   61.754427]  ieee80211_open [mac80211] from __dev_open+0xc4/0x170
[   61.761803]  __dev_open from __dev_change_flags+0x168/0x1dc
[   61.767963]  __dev_change_flags from dev_change_flags+0x18/0x54
[   61.773349]  dev_change_flags from devinet_ioctl+0x5b4/0x7ec
[   61.779256]  devinet_ioctl from inet_ioctl+0x16c/0x268
[   61.785152]  inet_ioctl from sock_ioctl+0x240/0x60c
[   61.790100]  sock_ioctl from sys_ioctl+0x25c/0xc0c
[   61.794886]  sys_ioctl from ret_fast_syscall+0x0/0x54
[   61.799730] Exception stack(0xcf7c3fa8 to 0xcf7c3ff0)
[   61.804874] 3fa0:                   00000010 bef54858 00000010 00008914 bef54858 00000000
[   61.809913] 3fc0: 00000010 bef54858 00000010 00000036 00000001 00000001 00000000 00000020
[   61.818061] 3fe0: 0059bea4 bef54718 0051d8d0 b6f6766c
[   61.826338] ---[ end trace 0000000000000000 ]---
[   61.835883] br-lan: port 5(phy0-ap0) entered blocking state
[   61.835962] br-lan: port 5(phy0-ap0) entered disabled state
[   61.841292] ath10k_ahb a000000.wifi phy0-ap0: entered allmulticast mode
[   61.847264] ath10k_ahb a000000.wifi phy0-ap0: entered promiscuous mode
[   61.853931] br-lan: port 5(phy0-ap0) entered blocking state
[   61.859984] br-lan: port 5(phy0-ap0) entered forwarding state
[   61.866063] br-lan: port 5(phy0-ap0) entered disabled state
[   62.034655] ath10k_ahb a000000.wifi: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   63.560225] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   63.560330] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   63.596005] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   63.596989] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
[   63.881873] ath10k_ahb a800000.wifi: rts threshold -1
[   63.882119] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   63.886063] ------------[ cut here ]------------
[   63.896970] WARNING: CPU: 3 PID: 1695 at backports-6.9.9/net/mac80211/main.c:270 ieee80211_do_open+0x4e8/0x5e0 [mac80211]
[   63.901691] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) wireguard pppox ppp_mppe ppp_generic nft_tproxy nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mac80211(O) libchacha20poly1305 curve25519_neon cfg80211(O) usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc poly1305_arm nfnetlink nf_tproxy_ipv6 nf_tproxy_ipv4 nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4
[   63.902186]  macvlan libcurve25519_generic hwmon crc_ccitt compat(O) chacha_neon asn1_decoder ntfs3 msdos ip6_udp_tunnel udp_tunnel oid_registry tun autofs4 nls_utf8 nls_iso8859_1 nls_cp437 xxhash_generic crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_arm sha1_arm_neon sha1_arm ghash_arm_ce cmac blake2b_generic arc4 usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 exfat btrfs xor xor_neon raid6_pq libcrc32c dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax tpm crc32c_generic encrypted_keys trusted
[   64.041910] CPU: 3 PID: 1695 Comm: hostapd Tainted: G        W  O       6.6.40 #0
[   64.064079] Hardware name: Generic DT based system
[   64.071633]  unwind_backtrace from show_stack+0x10/0x14
[   64.076319]  show_stack from dump_stack_lvl+0x40/0x4c
[   64.081443]  dump_stack_lvl from __warn+0x94/0xbc
[   64.086647]  __warn from warn_slowpath_fmt+0x158/0x15c
[   64.091324]  warn_slowpath_fmt from ieee80211_do_open+0x4e8/0x5e0 [mac80211]
[   64.096362]  ieee80211_do_open [mac80211] from ieee80211_open+0x58/0x7c [mac80211]
[   64.103587]  ieee80211_open [mac80211] from __dev_open+0xc4/0x170
[   64.110962]  __dev_open from __dev_change_flags+0x168/0x1dc
[   64.117115]  __dev_change_flags from dev_change_flags+0x18/0x54
[   64.122504]  dev_change_flags from devinet_ioctl+0x5b4/0x7ec
[   64.128403]  devinet_ioctl from inet_ioctl+0x16c/0x268
[... 258 lines cut ...]
CONFIG_PACKAGE_transmission-web=y
CONFIG_PACKAGE_trojan=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_vlmcsd=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_zlib=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

siwind, 2024-07-16:
```
root@OpenWrt:~# uname -a
Linux OpenWrt 6.6.40 #0 SMP Tue Jul 16 03:39:32 2024 armv7l GNU/Linux
root@OpenWrt:~# opkg list-installed | grep ath
ath10k-board-qca4019 - 20240513-r1
ath10k-firmware-qca4019-ct - 2020.11.08-r1
kmod-ath - 6.6.40.6.9.9-r1
kmod-ath10k-ct - 6.6.40.2024.03.03~eb3f488a-r1
kmod-nf-nathelper - 6.6.40-r1
kmod-nf-nathelper-extra - 6.6.40-r1
ucode-mod-math - 2024.07.11~1a8a0bcf-r1
```

lorand-horvath, 2024-07-16:
What target and build version is this?

siwind, 2024-07-16:
The targe is:

define Device/p2w_r619ac-128m
    $(call Device/p2w_r619ac)
    DEVICE_VARIANT := 128M NAND
endef
TARGET_DEVICES += p2w_r619ac-128m

lorand-horvath, 2024-07-16:
The latest SNAPSHOT version is not 27647, far from it. It's 26948 https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic/

siwind, 2024-07-16:
> The latest SNAPSHOT version is not 27647, far from it. It's 26948 https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic/

I build it from the latest source code.     :--(

lorand-horvath, 2024-07-16:
It cannot be 27647, that's all I'm saying. That's about 700 commits ahead of the main branch. How is that possible?

mrkiko, 2024-07-16:
I have been hit by exactly the same warning and it's caused by ath10k-ct.
Replace it with the ath10k module and be sure to delect -ct firmwareand select standard one.
the warning gone away in mycase.

mrkiko, 2024-07-16:
Not saying it's a definitive solution, just speaking what I've seen.

dannil, 2024-07-16:
Also seeing this on `r26951-37f5a3bb01`, commit 37f5a3bb01305ca9e853f512fc3e4b929be83f0c.

Manawyrm, 2024-07-17:
Yup, seeing the same on ipq40xx/IPQ4019 (FritzBox 7530), with `r26954-b4f1deab90`. 

This bug caused a rather annoying issue, which might frustrate newcomers a lot:
WiFi tx power was set to 0dBm after the crash, but WiFi still worked (a little). 
RSSIs were awful of course, as was the user experience.

I only noticed the missing tx power in `iwinfo phy1-ap0 txpowerlist` / after seeing the way too low RSSI.

dannil, 2024-07-17:
I've flashed `r26933-d40f9ad48f` (commit d40f9ad48f49b445acd3af3b967a4d5802b48cea) and the error doesn't show up anymore, so it's in [the diff between d40f9ad and 37f5a3b](https://github.com/openwrt/openwrt/compare/d40f9ad48f49b445acd3af3b967a4d5802b48cea...37f5a3bb01305ca9e853f512fc3e4b929be83f0c?diff=unified&w=), I have a (big hunch) that it's either 1bfcc1e or 7ca4190 (or both) since they touched ath10k-related code paths.

rkboni, 2024-07-23:
> Also seeing this on `r26951-37f5a3bb01`, commit [37f5a3b](https://github.com/openwrt/openwrt/commit/37f5a3bb01305ca9e853f512fc3e4b929be83f0c). Device is Netgear R7800 (target ipq806x).

Also noted on Meraki MR-52 as linked above.

markmentovai, 2024-07-31:
Before https://github.com/torvalds/linux/commit/0a44dfc070749514b804ccac0b1fd38718f7daa1 (2024-02-08, in 6.9), [`ieee80211_do_open`](https://github.com/torvalds/linux/blob/af4acac7caa17cf10742c80e2dd859e21d06abb9/net/mac80211/iface.c#L1436) would call [`ieee80211_hw_config`](https://github.com/torvalds/linux/blob/f04d2c247e0407ff8219395bef11ddc0837b7397/net/mac80211/main.c#L170). The meat of that happens inside the driver-specific `drv_config`, but the key here is that `ieee80211_hw_config` was (and still is) error-tolerant:

```c
		ret = drv_config(local, changed);
		/*
		 * Goal:
		 * HW reconfiguration should never fail, the driver has told
		 * us what it can support so it should live up to that promise.
		 *
		 * Current status:
		 * rfkill is not integrated with mac80211 and a
		 * configuration command can thus fail if hardware rfkill
		 * is enabled
		 *
		 * FIXME: integrate rfkill with mac80211 and then add this
		 * WARN_ON() back
		 *
		 */
		/* WARN_ON(ret); */
```

But since https://github.com/torvalds/linux/commit/0a44dfc070749514b804ccac0b1fd38718f7daa1, brought into OpenWrt via mac80211-backports 6.9.9 at https://github.com/openwrt/openwrt/commit/1bfcc1ea8a78ae834055bf554aa45720c09ea919, [`ieee80211_do_open`](https://github.com/torvalds/linux/blob/0874bcd0e1c97db0bada32df0934d5cf2507bedd/net/mac80211/iface.c#L1423) will be diverted to call the new [`ieee80211_hw_conf_init`](https://github.com/torvalds/linux/blob/a01b1e9f9955679fa04adead06c100022e313e55/net/mac80211/main.c#L250) on interface bring-up, in place of the old (and still error-tolerant) [`ieee80211_hw_config`](https://github.com/torvalds/linux/blob/a01b1e9f9955679fa04adead06c100022e313e55/net/mac80211/main.c#L191). Notably, the newer `ieee80211_hw_conf_init` does not share the error tolerance:

```c
	WARN_ON(drv_config(local, changed));
```

The warning is likely being raised for a condition that was present all along, but that `ieee80211_hw_config` was silent about. If the condition occurs during bring-up, `ieee80211_hw_conf_init` will now issue a warning.

[[`drv_config`](https://github.com/torvalds/linux/blob/1decf05d0f4de78ef67dc3f794709258c689e09e/net/mac80211/driver-ops.h#L146) → [`ath10k_config`](https://github.com/greearb/ath10k-ct/blob/eb3f488a200fafc46140fd51b5e21f737ee50f24/ath10k-6.7/mac.c#L10554)]

The driver in this case is ath10k-ct, and it’s been reported that ath10k does not suffer from the same problem. OpenWrt is currently using https://github.com/greearb/ath10k-ct/commit/eb3f488a200fafc46140fd51b5e21f737ee50f24, 2024-03-02, (although https://github.com/openwrt/openwrt/pull/16036/ would bring that to https://github.com/greearb/ath10k-ct/commit/ac71b14dc93aef0af6f0f24808b0afb673eaa5f5, 2024-07-30). Both the older [ath10k-ct 6.7 `ath10k_config`](https://github.com/greearb/ath10k-ct/blob/eb3f488a200fafc46140fd51b5e21f737ee50f24/ath10k-6.7/mac.c#L6293) and the newer [ath10k-ct 6.9 `ath10_config`](https://github.com/greearb/ath10k-ct/blob/ac71b14dc93aef0af6f0f24808b0afb673eaa5f5/ath10k-6.9/mac.c#L6294) behave identically. They are also a bit additive on top of [mainline ath10k `ath10k_config`](https://github.com/torvalds/linux/blob/1decf05d0f4de78ef67dc3f794709258c689e09e/drivers/net/wireless/ath/ath10k/mac.c#L5419). This is specific to ath10k-ct:

```c
	if (changed & IEEE80211_CONF_CHANGE_RETRY_LIMITS)
		ret = ath10k_config_retry_limit(ar, conf->long_frame_max_tx_count);
```

Nothing wrong with that, but the ath10k-ct-specific [`ath10k_config_retry_limit`](https://github.com/greearb/ath10k-ct/blob/eb3f488a200fafc46140fd51b5e21f737ee50f24/ath10k-6.7/mac.c#L5380) is the function responsible for our old familiar friend:

```c
		/* Stock wave-1 firmware (at least) will crash if there are > 2
		 * retries made, due to coding issues related to rate-ctrl
		 * logic in the firmware.  So, we can only enable this feature
		 * if firmware specifically tells us the feature is supported.
		 */
		ath10k_warn(ar, "Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: %d\n",
			    limit);
		return -EINVAL;
```

I see this warning on my ipq806x and ipq40xx devices with ath10k radios. So, follow this `-EINVAL` back out, and you’ll see that this is just that same old error case, “upgraded” to a scarier-looking `WARN_ON` by the new `ieee80211_hw_conf_init` in https://github.com/torvalds/linux/commit/0a44dfc070749514b804ccac0b1fd38718f7daa1. It’s as harmless (or not) as it’s ever been.

That said, the `WARN_ON` does demand more attention, and this is an opportune moment.

@greearb @hnyman
#15997

ptpt52, 2024-08-26:
same issue on my rt-acrh17

remittor, 2024-09-23:
same issue on my Xiaomi R3D (ath10k)
```
[   53.337660] ------------[ cut here ]------------
[   53.348676] WARNING: CPU: 0 PID: 1537 at backports-6.9.9/net/mac80211/main.c:270 ieee80211_do_open+0x4f0/0x644 [mac80211]

[   53.549978] CPU: 0 PID: 1537 Comm: hostapd Tainted: G        W  O       6.6.50 #0
[   53.572111] Hardware name: Generic DT based system
[   53.579753]  unwind_backtrace from show_stack+0x10/0x14
[   53.584436]  show_stack from dump_stack_lvl+0x40/0x4c
[   53.589559]  dump_stack_lvl from __warn+0x94/0xbc
[   53.594767]  __warn from warn_slowpath_fmt+0x158/0x15c
[   53.599454]  warn_slowpath_fmt from ieee80211_do_open+0x4f0/0x644 [mac80211]
[   53.604494]  ieee80211_do_open [mac80211] from ieee80211_open+0x58/0x7c [mac80211]
[   53.611698]  ieee80211_open [mac80211] from __dev_open+0xd4/0x194
[   53.619074]  __dev_open from __dev_change_flags+0x184/0x204
[   53.625235]  __dev_change_flags from dev_change_flags+0x18/0x54
[   53.630616]  dev_change_flags from devinet_ioctl+0x5b4/0x7ec
[   53.636520]  devinet_ioctl from inet_ioctl+0x16c/0x268
[   53.642422]  inet_ioctl from sock_ioctl+0x240/0x60c
[   53.647371]  sock_ioctl from sys_ioctl+0x25c/0xc0c
[   53.652146]  sys_ioctl from ret_fast_syscall+0x0/0x54
[   53.657004] Exception stack(0xc4ffffa8 to 0xc4fffff0)
[   53.662130] ffa0:                   00000010 be8947d8 00000010 00008914 be8947d8 be8947c8
[   53.667172] ffc0: 00000010 be8947d8 00000010 00000036 00000001 b6aaee90 be894830 ba408960
[   53.675328] ffe0: b6f24f74 be894698 00498bf8 b6edd704
```

lhotari, 2025-02-14:
Similar issue with TP-Link Deco M4R v1 with 24.10.0:

```
[   87.412288] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   87.420053] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[   87.469971] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   87.481302] ath10k_pci 0000:00:00.0: wmi print 'free: 114572 iram: 12644 sram: 29508'
[   87.856819] ath10k_pci 0000:00:00.0: rts threshold -1
[   87.862509] ath10k_pci 0000:00:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   87.873980] ------------[ cut here ]------------
[   87.878752] WARNING: CPU: 0 PID: 1735 at backports-6.12.6/net/mac80211/main.c:272 0x8292089c [mac80211@9bc65065+0x8a000]
[   87.890091] Modules linked in: ath9k(O) ath9k_common(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet ath9k_hw(O) ath10k_pci(O) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) lzo cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c hwmon crc_ccitt compat(O) sha512_generic seqiv sha3_generic jitterentropy_rng drbg hmac geniv rng cmac crypto_acompress gpio_button_hotplug(O) crc32c_generic
[   87.957219] CPU: 0 PID: 1735 Comm: hostapd Tainted: G           O       6.6.73 #0
[   87.964958] Stack : 00000000 80758350 00000000 830faea8 00000000 00000000 00000000 00000000
[   87.973611]         00000000 00000000 00000000 00000000 00000000 00000001 830d9b10 3ff49e35
[   87.982265]         830d9ba8 00000000 00000000 830d9a58 00000038 807014a4 00000000 ffffffea
[   87.990922]         000000db 830d9a64 000000db 807f9720 8071a9b8 830d9af0 00000000 8292089c
[   87.999577]         00000009 830faea8 00000000 82f2d600 00000018 803d09c8 00000000 809b0000
[   88.008231]         ...
[   88.010758] Call Trace:
[   88.010890] [<807014a4>] 0x807014a4
[   88.017088] [<8292089c>] 0x8292089c [mac80211@9bc65065+0x8a000]
[   88.023295] [<803d09c8>] 0x803d09c8
[   88.026982] [<800661e8>] 0x800661e8
[   88.030588] [<800661f0>] 0x800661f0
[   88.034190] [<806d7e24>] 0x806d7e24
[   88.037821] [<800849b0>] 0x800849b0
[   88.041480] [<8292089c>] 0x8292089c [mac80211@9bc65065+0x8a000]
[   88.047644] [<80084a40>] 0x80084a40
[   88.051293] [<82902394>] 0x82902394 [mac80211@9bc65065+0x8a000]
[   88.057569] [<8292089c>] 0x8292089c [mac80211@9bc65065+0x8a000]
[   88.063887] [<82920a60>] 0x82920a60 [mac80211@9bc65065+0x8a000]
[   88.070143] [<804bc9f0>] 0x804bc9f0
[   88.073836] [<804bcf08>] 0x804bcf08
[   88.077556] [<804bcfb4>] 0x804bcfb4
[   88.081186] [<800be27c>] 0x800be27c
[   88.084815] [<804b0dac>] 0x804b0dac
[   88.088444] [<805a8570>] 0x805a8570
[   88.092073] [<8048a6a4>] 0x8048a6a4
[   88.095734] [<805ab2ac>] 0x805ab2ac
[   88.099454] [<804ee8d0>] 0x804ee8d0
[   88.103113] [<804ef6b8>] 0x804ef6b8
[   88.106771] [<8048ae48>] 0x8048ae48
[   88.110550] [<8048c418>] 0x8048c418
[   88.114208] [<80221310>] 0x80221310
[   88.117899] [<80488e54>] 0x80488e54
[   88.121563] [<8048cbd0>] 0x8048cbd0
[   88.125169] [<80489024>] 0x80489024
[   88.128834] [<8006d420>] 0x8006d420
[   88.132475] 
[   88.134087] ---[ end trace 0000000000000000 ]---
```

dchard, 2025-02-21:
Same on MikroTik Hap Ac3 LTE6 kit and Chateau12-LTE.

greearb, 2025-03-05:
I think you can just comment out the error return in ath10k-ct code that relates to retry-count?  We are only using 6.11 and higher currently, but I can make similar changes there for going forward.

develBuzz, 2025-03-19:
Same on FB7530. Had to switch back to openwrt 23.05.5 which works fine.

dchard, 2025-03-27:
@DragonBluep 
@hauke 

Ath10k-ct 6-14 is out, plus there is a missing patch that might solve these kernel warnings:
[PATCH 1/2] wifi: mac80211: Add non-atomic station iterator.

Some fixes are also pushed for the current 6.10, which I already pulled but without the above mentioned patch, the kernel warnings are still there. Would you be so kind to at least pull the missing patch and update the PKG_HASH so we can start testing this? 

Much appreciate your help!

closed 2025-03-31 by openwrt-bot