openwrt/openwrt #15964: OpenWrt creates a device for unchecked local VLAN

cmonty14, 2024-07-17:
### Describe the bug

Following the [Security Considerations with VLANs](https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial#security_considerations_with_vlans) I defined this _Bridge VLAN filtering_:
[image]

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ramips/mt7621

### Device

MikroTik RouterBOARD 760iGS

### Image kind

Official downloaded image

### Steps to reproduce

1. LuCI > Network > Interfaces > Devices
2. Configure LAN bridge: br-lan
3. Bridge VLAN filtering
4. Add new VLAN ID 999, uncheck local, set Untagged + Is Primary VLAN

This creates following device in _/etc/config/network_:
```
config bridge-vlan
	option device 'br-lan'
	option vlan '999'
	option local '0'
	list ports 'lan4:u*'
	list ports 'lan5:u*'
	list ports 'sfp:u*'
```

### Actual behaviour

According to Wiki, "VLAN 999 is not used anywhere else and is only there for added security", however I can see br-lan.999 in the device list and I can assign an interface to it.
[image]

### Expected behaviour

There must be **no** device `br-lan.999`.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-17:
Do you see it in brctl or it is some usermode logic only?

cmonty14, 2024-07-18:
Just to be clear about the information you request, can you please specify the command syntax?
And what do you mean with "usermode logic"?

brada4, 2024-07-18:
I mean uci luci has setting in all corners but what is in `brctl show` configured in kernel.

cmonty14, 2024-07-19:
> I mean uci luci has setting in all corners but what is in `brctl show` configured in kernel.

Actually I don't say there's an issue with the bridge.
Anyway, here's the requested output:
[image]

What I say is that there's an device `br-lan.999` that should not exist.

brada4, 2024-07-19:
Post text output.
And from
```
bridge vlan show
```

cmonty14, 2024-07-19:
> Post text output. And from
> 
> ```
> bridge vlan show
> ```

Command `bridge` does not exist.

brada4, 2024-07-19:
It is how it works, the interface is disconnected from host.
`bridge` command is in package `ip-bridge`

cmonty14, 2024-07-20:
> It is how it works, the interface is disconnected from host.

How do you identify this?

Here's the output of `bridge vlan show`:
```
root@rb760igs:~# bridge vlan show
port              vlan-id  
sfp               10
                  20
                  30
                  40
                  100
                  999 PVID Egress Untagged
lan3              10
                  20
                  30
                  40
                  100 PVID Egress Untagged
lan4              10
                  20
                  30
                  40
                  100
                  999 PVID Egress Untagged
lan5              10
                  20
                  30
                  40
                  100
                  999 PVID Egress Untagged
br-lan            10
                  20
                  30
                  40
                  100
```

brada4, 2024-07-20:
But wire is down, openwrt is not connected.

cmonty14, 2024-07-21:
> But wire is down, openwrt is not connected.

This is a screenshot of the last 2 interfaces on my device:
[image]
Besides the missing IPv6 adress, I cannot detect any other difference.
Therefore I don't see the expected "openwrt is not connected".
Please advise.

brada4, 2024-07-21:
Please share /etc/config/network. You knowingly assigned IP address to interface, does not look like unchecked.

cmonty14, 2024-07-21:
> You knowingly assigned IP address to interface, does not look like unchecked.

This is correct.
Means, I can create an interface for device br-lan.999.

But this contradicts the statement in [Wiki](https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial#security_considerations_with_vlans) saying:
"Note: Because local is not checked for VLAN 90, OpenWrt won't even create a device for it and there should be no interface for it, unlike the other VLANs."

So let me make clear:

- I add a VLAN ID with local unchecked
- OpenWrt creates a device
- I can create an interface for this device

The last 2 steps **must not be possible**. 

Do we have a common understanding of this issue?

brada4, 2024-07-22:
You have to post in forums/documentation regarding false statement.

cmonty14, 2024-07-22:
> You have to post in forums/documentation regarding false statement.

Why do you think the statement is false?
If I add a VLAN ID **with local unchecked**, and OpenWrt creates a device, what's the difference to  adding a VLAN ID **with local checked**?

Nobody has proved yet that this is an issue with LuCI.

cmonty14, 2024-07-28:
This issue is not reproducable with OpenWrt 23.05.4

closed 2024-07-28 by cmonty14

cmonty14, 2024-08-14:
The issue is not reproducible with latest OpenWrt release.

peterwillcn, 2024-08-22:
ok