openwrt/openwrt #15970: Why doesn't x86 image include kernel flags for docker support?

odkrys, 2024-07-18:
### Describe the bug

https://forum.openwrt.org/t/docker-pull-fails-failed-to-register-layer-operation-not-supported/138253

I did a clean update from 21.06.6 to 23.05.4. But docker doesn't work. The problem appears to have existed since version 22.x. I don't know why the issue wasn't resolved.

https://github.com/MichaIng/DietPi/issues/7102
https://cloud.google.com/container-optimized-os/docs/concepts/supported-filesystems?hl=en

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

mini pc

### Image kind

Official downloaded image

### Steps to reproduce

docker pull blah/blah

### Actual behaviour

docker: failed to register layer: lsetxattr security.capability /usr/sbin/nginx: operation not supported.

### Additional info

# CONFIG_EXT4_FS_POSIX_ACL is not set
# CONFIG_EXT4_FS_SECURITY is not set

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-20:
Your test case does not exist.
As for why - because you did not submit a PR?

odkrys, 2024-07-20:
I also had the problem. I just downgraded docker and dockerd to version 20.10.17 in the 21.02 repository.

odkrys, 2024-08-16:
I verified this myself today by building an image with the above two kernel flags and it fixed the docker pull issue. A PR is needed.

joez, 2024-09-06:
The `dockerd` package has an option for this (`feeds/packages/utils/dockerd/Config.in`), just enable it and build the firmware by yourself

```
	config DOCKER_STO_EXT4
		bool "Enables support for ext3 or ext4 as the backing filesystem"
		default n
		select KERNEL_EXT4_FS_POSIX_ACL
		select KERNEL_EXT4_FS_SECURITY
```

ynezz, 2024-09-06:
FYI https://github.com/openwrt/openwrt/pull/16181 once merged should fix this in snapshots and in upcoming next release, but unfortunately the changes are a bit invasive, so unlikely to be backported into 23.05 release, so this will remain broken. Anyway, help with testing welcome!

closed 2024-09-07 by odkrys