openwrt/openwrt #15971: opkg: kernel info in /usr/lib/opkg/status not updated after updating from 23.05.3 to 23.05.4

Goodwu, 2024-07-18:
### Describe the bug

system info after update:
```
root@OpenWrt:/usr/lib/opkg# ubus call system board
{
	"kernel": "5.15.162",
	"hostname": "OpenWrt",
	"system": "ARMv8 Processor rev 4",
	"model": "FriendlyARM NanoPi NEO 2",
	"board_name": "friendlyarm,nanopi-neo2",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "23.05.4",
		"revision": "r24012-d8dd03c46f",
		"target": "sunxi/cortexa53",
		"description": "OpenWrt 23.05.4 r24012-d8dd03c46f"
	}
}
```

kernel package information from kernel.control file(It's right)
```
root@OpenWrt:~# cat /usr/lib/opkg/info/kernel.control
Package: kernel
Version: 5.15.162-1-1624aae9d5c213813ab71ed41e73c980
Depends: libc
Source: package/kernel/linux
SourceName: kernel
License: GPL-2.0-only
LicenseFiles: COPYING
Section: sys
SourceDateEpoch: 1720645656
Architecture: aarch64_cortex-a53
Installed-Size: 95
Description:  Virtual kernel package
root@OpenWrt:~#
```

kernel info from opkg:(It's wrong)
```
root@OpenWrt:~# opkg list kernel
kernel - 5.15.150-1-0603ef37b358fabb1f132a4507e3b82b
root@OpenWrt:~#
```

kernel info in /usr/lib/opkg/status:(It's wrong)
```
Package: kernel
Version: 5.15.150-1-0603ef37b358fabb1f132a4507e3b82b
Depends: libc
Status: install user installed
Architecture: aarch64_cortex-a53
Installed-Time: 1711145382
```

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

sunxi/cortexa53

### Device

FriendlyARM NanoPi NEO 2

### Image kind

Official downloaded image

### Steps to reproduce

1. Update the system using "openwrt-23.05.4-sunxi-cortexa53-friendlyarm_nanopi-neo2-squashfs-sdcard.img.gz"
2. With all three options checked.(1. keep current config; 2. don't backup same file under /rom; 3. backup package list at /etc/backup/installed_packages.txt)
3. Continue and wait for reboot
4. Check the version of package kernel.

### Actual behaviour

Kernel package with old version number.

### Expected behaviour

Kernel package with new version number.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-18:
Check webpage www.openwrt.org and never ever upgrade to unreleased versions.

dannil, 2024-07-18:
[A note about using new stable releases before they are announced](https://forum.openwrt.org/t/a-note-about-using-new-stable-releases-before-they-are-announced/125589), you're most likely hitting the second bullet point.

abysso2, 2024-07-19:
I also upgraded my nanopi r4s but all the wrong information you mentioned are right here ...

brada4, 2024-07-19:
Please upgrade in few weeks time when 23.05.4 is released, it is a bit confusing as it is shown in firmware-selector. For now 23.05.3 is stale release.

Goodwu, 2024-07-20:
> [A note about using new stable releases before they are announced](https://forum.openwrt.org/t/a-note-about-using-new-stable-releases-before-they-are-announced/125589), you're most likely hitting the second bullet point.

OK, now I know I should update openwrt after it has been officially released.
But all the points from the note you mentioned can't explain what I have met.

I want to emphasize that the problem is that the **opkg database doesn't match the actual information of firmware embeded packages**.

I do the update without internet and did not run opkg update after update. 
So there's nothing here about build systems or ongoing builds or anything like that, just the existing device and the new firmware.
And I have verified that the img I downloaded last time is exactly match the one on the web now.
```
MacBook-Air:openwrt xxx$ shasum -a 256 openwrt-23.05.4-sunxi-cortexa53-friendlyarm_nanopi-neo2-squashfs-sdcard.img.gz
8a4ed5cf2ad2bf854fb2fa66364bc85392db6a24b1745c856b2abdfb7e656db0  openwrt-23.05.4-sunxi-cortexa53-friendlyarm_nanopi-neo2-squashfs-sdcard.img.gz
MacBook-Air:openwrt xxx$
```
The sha256 sum is exactly match that on [the web](https://mirror-03.infra.openwrt.org/releases/23.05.4/targets/sunxi/cortexa53/)

The information you all ignored is that the new information is already present in /usr/lib/dpkg/info/kernel.control, while the information in /usr/lib/dpkg/status is still old.

This is not the first time I met this problem. When I upgrade from 23.05.2 to 23.05.3 two weeks ago, the same thing have occurred once and I reinstall openwrt from beginning.

Goodwu, 2024-07-20:
> I also upgraded my nanopi r4s but all the wrong information you mentioned are right here ...

You have the same problem or not?