openwrt/openwrt #15975: mvebu/wrt1900acv2 mwlwifi failed as to latest mac80211 v6.9.9-0-g28fdf4518483 backport

njhsi, 2024-07-21:
### Describe the bug

on the master/SNAPSHOT, mwlwifi not loaded and wifi no working
DMESG
=======
12.910052] Loading modules backported from Linux version **v6.9.9-0-g28fdf4518483**
[   12.917484] Backport generated by backports.git v6.1.97-1-29-gf1d24a3683b2
[   13.021847] <<Marvell Mac80211 Wireless PCIE Network Driver version 10.4.10.0>>
[   13.029203] pci 0000:00:01.0: enabling device (0140 -> 0142)
[   13.034912] ------------[ cut here ]------------
[   13.039557] WARNING: CPU: 1 PID: 1135 at /b/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.9.9/net/mac80211/main.c:794 0xbf35bfe4 [mac80211@89ff5acd+0x5
d000]
[   13.057988] Modules linked in: mwlwifi(O+) mac80211(O) cfg80211(O) compat(O) cryptodev(O) ip6_udp_tunnel udp_tunnel veth tun dns_resolver zram zsmalloc crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 seqiv sha3_generic jitterentropy_rng drbg md5 kpp hmac ghash_arm_ce echainiv geniv rng cmac chacha20poly1305 blake2b_generic xhci_pci xhci_pci_renesas ohci_pci uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_fsl gpio_button_hotplug(O) btrfs xor xor_neon raid6_pq libcrc32c dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax
[   13.109561] CPU: 1 PID: 1135 Comm: kmodloader Tainted: G           O       6.6.41 #0
[   13.117340] Hardware name: Marvell Armada 380/385 (Device Tree)
[   13.123288] Function entered at [<c010ea70>] from [<c0109e6c>]
[   13.129145] Function entered at [<c0109e6c>] from [<c08fba34>]
[   13.135002] Function entered at [<c08fba34>] from [<c012833c>]
[   13.140858] Function entered at [<c012833c>] from [<c01284bc>]
[   13.146713] Function entered at [<c01284bc>] from [<bf35bfe4>]
[   13.152579] Function entered at [<bf35bfe4>] from [<bf3c78a8>]
[   13.158436] Function entered at [<bf3c78a8>] from [<bf3d346c>]
[   13.164292] Function entered at [<bf3d346c>] from [<c055625c>]
[   13.170148] Function entered at [<c055625c>] from [<c05ae3ac>]
[   13.176005] Function entered at [<c05ae3ac>] from [<c05ae748>]
[   13.181861] Function entered at [<c05ae748>] from [<c05ae9d0>]
[   13.187716] Function entered at [<c05ae9d0>] from [<c05ac8f0>]
[   13.193572] Function entered at [<c05ac8f0>] from [<c05ad898>]
[   13.199428] Function entered at [<c05ad898>] from [<c05af4bc>]
[   13.205283] Function entered at [<c05af4bc>] from [<c010151c>]
[   13.211139] Function entered at [<c010151c>] from [<c019126c>]
[   13.216995] Function entered at [<c019126c>] from [<c0193168>]
[   13.222851] Function entered at [<c0193168>] from [<c0100060>]
[   13.228706] Exception stack(0xc3c9bfa8 to 0xc3c9bff0)
[   13.233781] bfa0:                   00000000 00000000 b6ec8620 000277d4 004ef23d 00000015
[   13.241994] bfc0: 00000000 00000000 b6ec8620 00000080 004ef23d 000277d4 004f52c8 00000000
[   13.250205] bfe0: beda2c5c beda2c40 004f11c4 b6f4ce6c
[   13.255286] ---[ end trace 0000000000000000 ]---
[   13.259925] ieee80211 alloc hw failed
**[   13.263600] mwlwifi: mwlwifi hw alloc failed
[   13.267888] mwlwifi: probe of 0000:01:00.0 failed with error -12**
[   13.273984] pci 0000:00:02.0: enabling device (0140 -> 0142)
[   13.279682] ------------[ cut here ]------------

### OpenWrt version

r0-39c824f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900AC v2

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

borkra, 2024-07-22:
The new chanctx needs to be added to the driver

borkra, 2024-07-23:
This patch fixes it
[006-wifi-mac80211-simplify-non-chanctx-drivers.txt](https://github.com/user-attachments/files/16343146/006-wifi-mac80211-simplify-non-chanctx-drivers.txt)

krant, 2024-07-25:
@borkra thanks for the patch! Confirming it fixes WRT3200ACM WiFi crashes.

mr-tomtom, 2024-07-25:
Thanks for the patch! Any timeline on when this could go into the snapshot builds?

borkra, 2024-07-27:
I do not have time at the moment dealing with creating pull requests

mbrinkmann, 2024-07-27:
Thanks a lot!

mr-tomtom, 2024-07-28:
Thanks @krant

PR #16016 (2024-07-28): mwlwifi: fix mac80211 broken after update to 6.9.9
Fixes #15975.
Thanks to @borkra for the actual fix!
Verified on WRT3200ACM.
closed 2024-07-28 by openwrt-bot