openwrt/openwrt #15976: after updating to 23.05.04 ram usage increased

olumolu, 2024-07-21:
### Describe the bug

After updating to 03 to 04 on xiaomi mi router 4a gigabit v1 my ram usages increase with 13 mb more 
https://forum.openwrt.org/uploads/default/original/3X/5/2/52885926c16e3ca60542fe22fea3fbb852f1d826.jpeg

### OpenWrt version

23.05.04

### OpenWrt release

OpenWrt 23.05.4 (r24012-d8dd03c46f)

### OpenWrt target/subtarget

openwrt-23.05 branch (git-24.086.45142-09d5a38)

### Device

xiaomi mi router 4a v1 gigabit

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

degen91, 2024-07-21:
I believe it is related to this commit: [generic: 6.6: backport upstream commits for mtk_eth_soc](https://github.com/openwrt/openwrt/commit/93ce299d67231c6e744cdb18f2fe7407bd00242e)

You can read more about its impact on ram usage on the forum: [Mtk_eth_soc will use even more ram with upstream patch](https://forum.openwrt.org/t/mtk-eth-soc-will-use-even-more-ram-with-upstream-patch/200524).

wlshdjj, 2024-07-25:
The same problem to x86_64.

Powerbless, 2024-07-25:
AX3600 and i think i have the same problem

olumolu, 2024-07-25:
This seems like only happening to the mi and x69 routers only

5k1n, 2024-08-18:
Same on 23.05.3/4 Xiaomi ax3600

AltanOzkan, 2024-12-03:
**Noticeable RAM Usage Increase in Xiaomi Mi Router 4A Gigabit Edition**  

**Observation:**  
I observed a significant RAM usage increase with newer OpenWrt versions on my Xiaomi Mi Router 4A Gigabit Edition. All tests were conducted with default installations (no additional packages).  

| Version      | Average RAM Usage |  
|--------------|-------------------|  
| 21.02.7      | ~35-40 MB         |  
| 22.03.7      | ~35-40 MB         |  
| 23.05.5      | ~75-80 MB         |  
| 24.10.0-rc1  | ~75-80 MB         |  

The jump in RAM usage starting from version 23.05.5 is concerning for devices with limited memory. Can this be looked into? Thank you!

lukasz1992, 2025-01-27:
Please check 24.10.0-rc6, should be better

olumolu, 2025-01-27:
i have updated but it was better than before but not what used to see less than that but it is ok. newer kernel will use higher amount of ram this is normal.