openwrt/openwrt #15981: Netgear WNDR3800 eth1 not functional since switch to kernel v6.6.x

RussellSenior, 2024-07-21:
### Describe the bug

With a default configuration, the normal eth1 interface as WAN does not function normally, it does not get a dhcp lease and does not appear (via tcpdump) receive any traffic, despite the interface showing UP.

### OpenWrt version

r26971-66177c081f

### OpenWrt release

SNAPSHOT (built locally)

### OpenWrt target/subtarget

ath79/generic

### Device

Netgear WNDR3800

### Image kind

Self-built image

### Steps to reproduce

wait for boot to complete, with WAN port connected to a DHCP server, check interface status, run tcpdump -i eth1

### Actual behaviour

eth1 does not get configured or appear to receive any traffic

### Expected behaviour

eth1 should pass traffic

### Additional info

I spent some time bisecting this. The brokenness appeared to start with the switch to kernel v6.6, however the eth1 behavior was masked by various boot hangs, which made the bisection not very clear: r26848-731318667d hangs during boot; r26851-d836a688d1 does not hang and gives the clear eth1-not-working behavior. Everything was last working normally on or about r26629-7a9b1f7f25.

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_netgear_wndr3800=y
CONFIG_DEVEL=y
CONFIG_BUILD_LOG=y
CONFIG_TARGET_INITRAMFS_COMPRESSION_LZMA=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

RussellSenior, 2024-07-22:
Yes, it's from a clean respository. SNAPSHOT is what is returned from the command suggested.

RussellSenior, 2024-07-22:
Try this: http://downloads.openwrt.org/snapshots/targets/ath79/generic/openwrt-ath79-generic-netgear_wndr3800-initramfs-kernel.bin (680f77e5e9709eaf5cc5fe9e2a0044e68bced633059a908fae104180748d29ad, from Sun Jul 21 23:01:37 2024). Same behavior. Github-actions is stupid.

RussellSenior, 2024-07-22:
Here is what my interfaces look like (macaddrs obfuscated) on failure:
```
root@OpenWrt:/# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP qlen 1000
    link/ether 02:8e:f2:xx:yy:mm brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether 00:8e:f2:xx:yy:nn brd ff:ff:ff:ff:ff:ff
    inet6 fe80::28e:f2ff:fexx:yynn/64 scope link 
       valid_lft forever preferred_lft forever
4: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 02:8e:f2:xx:yy:mm brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fdd6:fa1a:7446::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::8e:f2ff:fexx:yymm/64 scope link 
       valid_lft forever preferred_lft forever

```

lorand-horvath, 2024-07-22:
@RussellSenior I confirm the WAN port is not functional on my WNDRMAC v2 (same hardware as WNDR3800) on any 6.6 snapshots. This may be related to the GPIO issues that were fixed for 6.6 and documented here: https://github.com/openwrt/openwrt/issues/15739

markmentovai, 2024-07-23:
I can confirm this bug on an ath79/netgear,wndr3800.

In `dmesg`, I get one of these when I expect the link to go up:

```
[  101.809574] eth1: link up (1000Mbps/Full duplex)
```

but it’s a lie. Although the Tx counter shows moderate activity, the Rx shows nothing. There’s no matching `link down` message when I unplug the cable, it’s just a single `link up` the first time after boot, and that’s the end of it.

I don’t see a link at the other end of the cable at all, except during boot while the bootloader has control. Once OpenWrt takes over, there’s nothing.

lorand-horvath, 2024-07-23:
@markmentovai Sorry if there's some confusion, but I just posted more info on the initial GPIO-related issue https://github.com/openwrt/openwrt/issues/15739#issuecomment-2244461396 since I believe there were some changes made on June 20 which may have caused the broken WAN (eth1).

FYI, the GPIO endianness bug fix was first included in the code later, on June 24 with test build 26761 that I received from @DragonBluep and was obviously containing the June 20 changes.

lorand-horvath, 2024-07-23:
https://github.com/openwrt/openwrt/issues/15739#issuecomment-2244944057
> I don't have much knowledge about this part. But I am pretty sure that the WAN port is connected to the RTL8366 switch, not the AR7100 CPU (eth1). Perhaps the switch config is wrong ref: https://fcc.report/FCC-ID/PY308300092/1408032

@DragonBluep WAN port is connected to AR7161 CPU (eth1), not to RTL8366S switch, according to diagram:
https://openwrt.org/toh/netgear/wndr3700#switch_ports_for_vlans

markmentovai, 2024-07-23:
https://github.com/openwrt/openwrt/issues/15981#issuecomment-2244963655
> [#15739 (comment)](https://github.com/openwrt/openwrt/issues/15739#issuecomment-2244928491)
> 
> > I don't have much knowledge about this part. But I am pretty sure that the WAN port is connected to the RTL8366 switch, not the AR7100 CPU (eth1). Perhaps the switch config is wrong ref: https://fcc.report/FCC-ID/PY308300092/1408032
> 
> @DragonBluep WAN port is connected to AR7161 CPU (eth1), not to RTL8366S switch, according to diagram: https://openwrt.org/toh/netgear/wndr3700#switch_ports_for_vlans

That’s only partially correct. It’s true that the ar7161 SoC has two ag71xx Ethernet interfaces, and OpenWrt bridges eth0 into the ports marked LAN connected to the rtl8366s switch, while eth1 is independent. But the FCC photos that @DragonBluep shared also clearly show that the port marked WAN is also connected to the rtl8366s.

In reality, the WAN port uses the independent eth1 interface, but is also routed through the rtl8366s which provides the PHY for this port. This is clear if you look at the [rtl8366s datasheet](https://semiconductors.es/datasheet/RTL8366S.html). The block diagram is in §3.1 on page 16, and the wndr3x00 family uses the configuration in §4.3.1 on page 19.

This explains why the switch is used to control the WAN port LED, while both ag71xx interfaces on the CPU are used for different purposes, LAN and WAN.

RussellSenior, 2024-07-23:
Ah, yeah. See page 19 of http://realtek.info/pdf/rtl8366s_8366sr_datasheet_vpre-1.4_20071022.pdf ... that said, the openwrt /etc/config/network has never included eth1 in any port configuration.

lorand-horvath, 2024-07-23:
Interesting, so what changed between kernel 6.1 and 6.6 that could break the WAN port functionality? I assume more GPIO driver-related changes...

markmentovai, 2024-07-23:
My guess is that the problem will still be on the rtl8366s side (and yes, perhaps i2c control via gpio) rather than the ag71xx side.

KanjiMonster, 2024-07-23:
Both things can be true at the same time. Some switch chips support a wan port mode where one of the PHYs can be directly wired to a second (R)GMII port; the RTL8366SR is one of these, and the setup is configured via pin strapping AFAICT.

So while technically the wan port is connected to the switch, it's not part of the regular switching (AFAICT).

DragonBluep, 2024-07-23:
Can you find this message in the kernel log? If so, I guess the issue is caused by some general modules, not the driver.
`kern.info kernel: [    7.152451] ag71xx 1a000000.eth: connected to PHY at rtl8366s:04 [uid=001cc960, driver=Generic PHY]`

markmentovai, 2024-07-23:
Those are the correct PHYs.

There’s nothing discernible in the `dmesg` diff between working 6.1 and current 6.6 that explains the problem. It will take more debugging.

lorand-horvath, 2024-07-23:
Can this commit have anything to do with the issue? https://github.com/openwrt/openwrt/commit/150d0ceab7380538798df76a5d7d3c12a3632050

markmentovai, 2024-07-23:
https://github.com/openwrt/openwrt/issues/15981#issuecomment-2245262798:
> Can this commit have anything to do with the issue? [150d0ce](https://github.com/openwrt/openwrt/commit/150d0ceab7380538798df76a5d7d3c12a3632050)

Not in theory, and not in my experience. I did a build with that reverted and it made no difference.

The MAC addresses for both eth0 and eth1 are set in the same way, so it couldn’t really be the case that eth0 would work before eth1, but then it wouldn’t work for eth1.

namiltd, 2024-07-23:
Please attach full kernel log.

lorand-horvath, 2024-07-23:
@namiltd Please see attached builds 26620 (kernel 6.1) and 26990 (kernel 6.6) for WNDRMAC v2, both having eth1 interface as DHCP client (default).
[26620-6.1.txt](https://github.com/user-attachments/files/16353526/26620-6.1.txt)
[26990-6.6.txt](https://github.com/user-attachments/files/16353527/26990-6.6.txt)

RussellSenior, 2024-07-23:
> I debugged this and came up with a [temporary hack-patch](https://github.com/markmentovai/openwrt/commit/68a9a4401b91158bdae8f1ce9c903b591cf3fddd) to help others determine whether it resolves the bug for them. It’s not checkin-quality for reasons described in the patch. More work is needed.
> 
> Please test the patch and let me know if it works for you.

This makes eth1 work again for me, when applied on top of 66177c081f.

markmentovai, 2024-07-23:
I suspect that this was caused by https://github.com/torvalds/linux/commit/9b01c885be364526d8c05794f8358b3e563b7ff8 (2023-02-13, in kernel 6.3). That commit also offers a suggested fix, which I'll try as soon as I'm at a computer again.

The upstream change was not without fallout either ([1](https://lore.kernel.org/lkml/20230224035553.GA1089605@roeck-us.net/), [2](https://lore.kernel.org/all/167715661799.11159.2057121677394149658.git-patchwork-notify@kernel.org/)).

lorand-horvath, 2024-07-24:
@markmentovai That's some great detective work right there, thanks for taking the time and effort!

markmentovai, 2024-07-24:
I’ve published a [proper patch](https://github.com/markmentovai/openwrt/commit/10eeb4371bf67af18659cece1d3feb1e13fe8054) (non-hack), and I’d appreciate a test run or two before promoting it to a pull request.

lorand-horvath, 2024-07-24:
> I’ve published a [proper patch](https://github.com/markmentovai/openwrt/commit/10eeb4371bf67af18659cece1d3feb1e13fe8054) (non-hack), and I’d appreciate a test run or two before promoting it to a pull request.

@markmentovai Thank you for this, I would appreciate if you could provide a WNDRMAC v2 build with LuCI so I can test.

markmentovai, 2024-07-24:
https://github.com/openwrt/openwrt/issues/15981#issuecomment-2248814705
> @markmentovai Thank you for this,

You’re welcome!

> I would appreciate if you could provide a WNDRMAC v2 build with LuCI, so I can test.

Sorry, I don’t use or build with LuCI and wouldn’t know if I was giving you something working or broken.

KanjiMonster, 2024-07-25:
> I’ve published a [proper patch](https://github.com/markmentovai/openwrt/commit/10eeb4371bf67af18659cece1d3feb1e13fe8054) (non-hack), and I’d appreciate a test run or two before promoting it to a pull request.

After looking a bit, I think a better place would be to add it to the long list of realtek phy drivers in https://github.com/torvalds/linux/blob/master/drivers/net/phy/realtek.c#L1215 (note that it already contains an entry for rtl8366rb).

lorand-horvath, 2024-07-25:
> > I would appreciate if you could provide a WNDRMAC v2 build with LuCI, so I can test.
> 
> Sorry, I don’t use or build with LuCI and wouldn’t know if I was giving you something working or broken.

@markmentovai These devices are very easily unbrickable with TFTP, done it many times, so I don't care if it's a broken build. I only wanted to help with testing. I was just wondering and hoping to test whether your fix has any performance impact on NAT speeds.
I could test with a 1 Gbps connection; kernel 6.1-based builds provide around 600-700 Mbps NAT speeds using software flow offloading. But if you don't want me to test, that's also fine by me.

RussellSenior, 2024-07-25:
> I’ve published a [proper patch](https://github.com/markmentovai/openwrt/commit/10eeb4371bf67af18659cece1d3feb1e13fe8054) (non-hack), and I’d appreciate a test run or two before promoting it to a pull request.

Works here.

namiltd, 2024-07-25:
As [KanjiMonster](https://github.com/KanjiMonster)  mentioned https://github.com/markmentovai/openwrt/commit/10eeb4371bf67af18659cece1d3feb1e13fe8054#r144618826

As KanjiMonster](https://github.com/KanjiMonster) mentioned in:https://github.com/markmentovai/openwrt/commit/10eeb4371bf67af18659cece1d3feb1e13fe8054#r144618826, the code should be changed to:
```
    err = platform_driver_register(&rtl8366s_driver);
    if (err) {
        phy_driver_unregister(&rtl8366s_phy_driver);
        return err;
    }

KanjiMonster, 2024-07-25:
> As [KanjiMonster](https://github.com/KanjiMonster) mentioned in: [markmentovai@10eeb43#r144618826](https://github.com/markmentovai/openwrt/commit/10eeb4371bf67af18659cece1d3feb1e13fe8054#r144618826), the code should be changed to:
> 
> ```
> +    err = platform_driver_register(&rtl8366s_driver);
> +    if (err) {
> +        phy_driver_unregister(&rtl8366s_phy_driver);
> +        return err;
> +    }
> ```

Actually I revised my comment to move/add the phy driver to drivers/net/phy/realtek.c - which then may even be upstreamable, regardless of the rtl8366s driver being upstream (yet).

namiltd, 2024-07-25:
@KanjiMonster Please remove https://github.com/openwrt/openwrt/issues/15981#issuecomment-2249644813

markmentovai, 2024-07-25:
https://github.com/openwrt/openwrt/issues/15981#issuecomment-2249593922:
> After looking a bit, I think a better place would be to add it to the long list of realtek phy drivers in https://github.com/torvalds/linux/blob/master/drivers/net/phy/realtek.c#L1215 (note that it already contains an entry for rtl8366rb).

I’ve proposed a patch on [netdev](https://lore.kernel.org/netdev/20240725170519.43401-1-mark@mentovai.com/T/) with a Cc to [LKML](https://lore.kernel.org/linux-kernel/20240725170519.43401-1-mark@mentovai.com/T/) and all of you.

I’ve also updated [my OpenWrt rtl8366s_ignore_mmd patch branch](https://github.com/markmentovai/openwrt/tree/rtl8366s_ignore_mmd) to https://github.com/markmentovai/openwrt/commit/d5f40d599991e97da2ff1e843489931283b390af, which applies the fix as a pending-6.6 patch to the kernel’s drivers/net/phy/realtek.c rather than doing it in the OpenWrt-local target/linux/generic/files/drivers/net/phy/rtl8366s.c.

https://github.com/openwrt/openwrt/issues/15981#issuecomment-2249594681:
> @markmentovai These devices are very easily unbrickable with TFTP, done it many times, so I don't care if it's a broken build. I only wanted to help with testing. I was just wondering and hoping to [...]

Lóránd, if you’re OK with “no warranty” as always, you can try [openwrt-snapshot-r26912+1-3c95641366f7-d5f40d599991-ath79-generic-netgear_wndrmac-v2-squashfs-sysupgrade.bin](https://www.mentovai.com/tmp/openwrt-snapshot-r26912+1-3c95641366f7-d5f40d599991-ath79-generic-netgear_wndrmac-v2-squashfs-sysupgrade.bin) built from the above branch with [this defconfig](https://www.mentovai.com/tmp/openwrt-snapshot-r26912+1-3c95641366f7-d5f40d599991-ath79-generic-netgear_wndrmac-v2.defconfig).

lorand-horvath, 2024-07-26:
> Lóránd, if you’re OK with “no warranty” as always, you can try [...]

@markmentovai Just tested the snapshot build 26912 (kernel 6.6.36) you provided and it works great on my WNDRMAC v2, mostly the same behavior as 26620 (kernel 6.1.92), so there's no performance degradation or anything else out of the ordinary, AFAICS.
Thanks again, looking forward to seeing the patch merged into kernel 6.6 soon!

KanjiMonster, 2024-07-26:
> I’ve proposed a patch on [netdev](https://lore.kernel.org/netdev/20240725170519.43401-1-mark@mentovai.com/T/) with a Cc to [LKML](https://lore.kernel.org/linux-kernel/20240725170519.43401-1-mark@mentovai.com/T/) and all of you.

And it's already accepted: https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=225990c487c1

markmentovai, 2024-07-26:
https://github.com/openwrt/openwrt/issues/15981#issuecomment-2252899639:
> And it's already accepted: https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=225990c487c1

Yes, that was fast!

I’ll wait until it hits torvalds/linux and then submit an OpenWrt pull request to add it as a backport-6.6 patch, unless there’s a pressing need to do it sooner as a pending-6.6.

RussellSenior, 2024-07-26:
Why wait?

markmentovai, 2024-07-26:
https://github.com/openwrt/openwrt/issues/15981#issuecomment-2253462049:
> Why wait?

Yeah, I guess you’re right. If it’s already in net today, it’s headed for 6.11 at the very least, which should be good enough to call it a backport from 6.11.

→ https://github.com/openwrt/openwrt/pull/16012

closed 2024-07-29 by openwrt-bot