openwrt/openwrt #15988: Unable to load wireless network card driver for MT7922 on rpi4 (cm4), snapshot openwrt

lt5d-yunzi, 2024-07-22:
### Describe the bug

Unable to load the mt7922 driver correctly on my CM4 development board, using the CM4-WiFi7 by mcuzone motherboard 
Below are the pertinent logs：
```
[    0.253979] pci 0000:01:00.0: [14c3:0616] type 00 class 0x028000
[    0.254030] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
[    0.254071] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x00007fff 64bit]
[    0.254225] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    0.254614] pci 0000:01:00.0: BAR 0: assigned [mem 0x600100000-0x6001fffff 64bit pref]
[    0.254650] pci 0000:01:00.0: BAR 2: assigned [mem 0x600000000-0x600007fff 64bit]
[    5.827393] mt7921e 0000:01:00.0: enabling device (0000 -> 0002)
[    5.848828] mt7921e 0000:01:00.0: ASIC revision: 79220010
[    5.952266] mt7921e: probe of 0000:01:00.0 failed with error -12
```

### OpenWrt version

r26990-06b37a5856

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi Compute Module 4 Rev 1.0

### Image kind

Official downloaded image

### Steps to reproduce

The image used is a snapshot version built by adding software packages from the OpenWRT official website.
#### Only added:
```
kmod-usb-net-cdc-ether
kmod-usb-net-rndis
```

### Additional info

Possible useful information
```
root@OpenWrt:~# dmesg | grep mt7921e
[    5.827393] mt7921e 0000:01:00.0: enabling device (0000 -> 0002)
[    5.848828] mt7921e 0000:01:00.0: ASIC revision: 79220010
[    5.952266] mt7921e: probe of 0000:01:00.0 failed with error -12
root@OpenWrt:~# dmesg | grep firmware
[    0.039983] raspberrypi-firmware soc:firmware: Attached to firmware from 2024-04-17T17:27:09, variant start
[    0.049960] raspberrypi-firmware soc:firmware: Firmware hash is 86ccc427f35fdc604edc511881cdf579df945fb4
root@OpenWrt:~# uname -r
6.6.41
root@OpenWrt:~# modinfo mt7921e
filename:       /lib/modules/6.6.41/mt7921e.ko
license:        Dual BSD/GPL
depends:        mt7921-common,mt792x-lib,mt76-connac-lib,mt76
name:           mt7921e
vermagic:       6.6.41 SMP mod_unload aarch64
parm:           disable_aspm (bool)
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lt5d-yunzi, 2024-07-22:
> 报告的版本无效。''这是来自干净的存储库吗？

Yes, this is from https://firmware-selector.openwrt.org/ Compiled on the website

lt5d-yunzi, 2024-07-22:
> Invalid Target/Subtarget reported. `` Is this from a supported device?

Raspberry Pi 4B/400/CM4 (64bit)

castillofrancodamian, 2024-07-22:
I can't see the wireless interfaces either. I don't know if it's the same error.

```
root@OpenWrt:~# wifi config
Reference error: left-hand side expression is null
In wiphy_detect(), file /usr/share/hostap/wifi-detect.uc, line 82, byte 18:
  called from anonymous function (/usr/share/hostap/wifi-detect.uc:177:14)

 `        let name = phy.wiphy_name;`
  Near here -------------^

```

lt5d-yunzi, 2024-07-23:
There is no any output after executing "wifi config"

castillofrancodamian, 2024-07-23:
> There is no any output after executing "wifi config"

So it worked for you. That command generates a wireless configuration file.

lt5d-yunzi, 2024-07-23:
> > There is no any output after executing "wifi config"
> 
> So it worked for you. That command generates a wireless configuration file.

But my MT7922 is not working

brada4, 2024-07-23:
> MT7922

care to show lspci?

castillofrancodamian, 2024-07-23:
> > > There is no any output after executing "wifi config"
> > 
> > 
> > So it worked for you. That command generates a wireless configuration file.
> 
> But my MT7922 is not working

I'm going to have to create a issue.

lt5d-yunzi, 2024-07-25:
> > MT7922
> 
> care to show lspci?

```
root@OpenWrt:~# lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 Network controller: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter
root@OpenWrt:~# client_loop: send disconnect: Connection reset
```

SmallC500, 2026-08-16:
Everything is the same on Raspberry Pi 5 that I met the same problem

brada4, 2026-08-16:
I mean `dtovelay -l` , not OpenWrt specific at all. Same overlay needed to add more iomem for 7922