openwrt/openwrt #15994: Sysupgrade can't create backup on Asus TUF-AX4200

epiniguin, 2024-07-23:
### Describe the bug

Sysupgrade -b can't create a backup - there are two sysupgrade processes, but they don't make anything.
Tried stop wpad - it does not help.

### OpenWrt version

r26802-f637cf5ef7

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX4200

### Image kind

Self-built image

### Steps to reproduce

sysupgrade -b or backup via LUCI or trying to upgrade.

### Actual behaviour

Backup process is frozen, so it is impossible nor to create a backup nor to make a firmware upgrade.

### Expected behaviour

Sysupgrade should create a backup.

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_abt_asr3000=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_abt_asr3000=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_acelink_ew-7886cax=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_acelink_ew-7886cax=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_acer_predator-w6=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_acer_predator-w6=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_smartrg_sdg-8612=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_smartrg_sdg-8612=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_smartrg_sdg-8614=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_smartrg_sdg-8614=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_smartrg_sdg-8622=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_smartrg_sdg-8622=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_smartrg_sdg-8632=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_smartrg_sdg-8632=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_smartrg_sdg-8733=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_smartrg_sdg-8733=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_smartrg_sdg-8734=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_smartrg_sdg-8734=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_asus_rt-ax59u=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_asus_rt-ax59u=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_asus_tuf-ax4200=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_asus_tuf-ax4200=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_asus_tuf-ax6000=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_asus_tuf-ax6000=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_bananapi_bpi-r3=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_bananapi_bpi-r3=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_bananapi_bpi-r3-mini=y
[... 93 lines cut ...]
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_xiaomi_redmi-router-ax6000-stock=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_xiaomi_redmi-router-ax6000-stock=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_yuncore_ax835=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_yuncore_ax835=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_zbtlink_zbt-z8102ax=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_zbtlink_zbt-z8102ax=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_zbtlink_zbt-z8103ax=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_zbtlink_zbt-z8103ax=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_zyxel_ex5601-t0-ubootmod=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_zyxel_ex5601-t0-ubootmod=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_zyxel_ex5601-t0-stock=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_zyxel_ex5601-t0-stock=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_zyxel_ex5700-telenor=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_zyxel_ex5700-telenor=""
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_zyxel_nwa50ax-pro=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_zyxel_nwa50ax-pro=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-24:
What is in `/etc/sysupgrade.conf` ?

epiniguin, 2024-07-24:
> What's the last version in which it worked?

It was some snapshot from the January 2024. Maybe January 20 or 25.

epiniguin, 2024-07-24:
> What is in `/etc/sysupgrade.conf` ?

```
## This file contains files and directories that should
## be preserved during an upgrade.

# /etc/example.conf
# /etc/openvpn/
/etc/init.d/zswanctl
/etc/*.sh
/etc/uhttpd.full.crt
/etc/uhttpd.key
/root/.ssh
/etc/*.user
/etc/dnscrypt-proxy2/

brada4, 2024-07-25:
Restart device and save backup via web. It is same file.
Try
```
du -sk `sysupgrade -l` | sort -n
```
Largest file is how big?

epiniguin, 2024-07-25:
> Restart device and save backup via web. It is same file.

I wrote that sysupgrade can't make a backup nor via command line, nor via LUCI, nor as a first stage of an upgrade.

epiniguin, 2024-07-25:
> Largest file is how big?

28k

epiniguin, 2024-07-26:
[sysupgrade.log](https://github.com/user-attachments/files/16385203/sysupgrade.log)

epiniguin, 2024-07-26:
> > [sysupgrade.log](https://github.com/user-attachments/files/16385203/sysupgrade.log)
> 
> the three dots in the command were meant as a placeholder for arguments you normally use, duh

Normally I use LUCI. But now it doesn't matter - sysupgrade does nothing.
So `sh -vx $(which sysupgrade) -b /tmp/backup &>sysupgrade.log` does nothing. There are two sysupgrade processes but they do nothing and don't finish and don't create sysupgrade.log.

brada4, 2024-07-26:
If you do not provide log from "do nothing" we cannot help.

epiniguin, 2024-07-26:
> If you do not provide log from "do nothing" we cannot help.

I wrote that sysupgrade processes don't finish, so the log isn't created. I can only press ctrl-c.

epiniguin, 2024-07-26:
Ok, got it. I removed `&>sysupgrade.log` from your command and got the log on the screen.
The problem is in /etc/init.d/mwan3.
Below is the log end. The script stumbles at `/etc/init.d/mwan3 enabled`.
If I start `/etc/init.d/mwan3 enabled` manually it also doesn't finish. This is a standard script from mwan3 package, I didn't change it never.

```
...............
+ '[' /tmp/backup '!=' - ]
+ v 'Saving config files...'
+ date
+ _v 'Sat Jul 27 00:03:43 EEST 2024 upgrade: Saving config files...'
+ '[' -n 1 ]
+ '[' 1 -ge 1 ]
+ echo 'Sat Jul 27 00:03:43 EEST 2024 upgrade: Saving config files...'
Sat Jul 27 00:03:43 EEST 2024 upgrade: Saving config files...
+ logger -p info -t upgrade 'Saving config files...'
+ '[' 1 -gt 1 ]
+ TAR_V=
+ sed -i -e s,^/,, /tmp/sysupgrade.conffiles
+ set -o pipefail
+ local 'ret=0'
+ '[' 0 -eq 0 ]
+ gzip
+ /etc/init.d/adblock enabled
+ /etc/init.d/boot enabled
+ /etc/init.d/bootcount enabled
+ /etc/init.d/cron enabled
+ /etc/init.d/dawn enabled
+ /etc/init.d/ddns enabled
+ /etc/init.d/dnscrypt-proxy enabled
+ /etc/init.d/dnsmasq enabled
+ /etc/init.d/done enabled
+ /etc/init.d/dropbear enabled
+ /etc/init.d/firewall enabled
+ /etc/init.d/gpio_switch enabled
+ /etc/init.d/led enabled
+ /etc/init.d/log enabled
+ /etc/init.d/mwan3 enabled
^C

epiniguin, 2024-07-27:
I use mwan3.

[openwrt.log](https://github.com/user-attachments/files/16400228/openwrt.log)

epiniguin, 2024-07-28:
> hate calling it, but have you tried turning it off and on again

I can't stop or disable or enable mwan3, because these commands also freeze without doing anything.

P.S. Router reboot does not help :-)

epiniguin, 2024-07-28:
```
root@ax4200:~# ps w | grep mwan
 2417 root      1680 S    {mwan3track} /bin/sh /usr/sbin/mwan3track wan
 2418 root      1680 S    {mwan3track} /bin/sh /usr/sbin/mwan3track wan2
 2419 root      1664 S    {mwan3track} /bin/sh /usr/sbin/mwan3track wan3
 2420 root      1852 S    {mwan3rtmon} /bin/sh /usr/sbin/mwan3rtmon ipv4
 2421 root      1852 S    {mwan3rtmon} /bin/sh /usr/sbin/mwan3rtmon ipv6
 2558 root      1872 S    {mwan3rtmon} /bin/sh /usr/sbin/mwan3rtmon ipv4
 2564 root      1864 S    {mwan3rtmon} /bin/sh /usr/sbin/mwan3rtmon ipv6
28026 root      1336 S    grep mwan

epiniguin, 2024-07-28:
The problem was in my `mwan3.user` script where I restart some services including DDNS. And DDNS service restart is the cause of mwan3 lock(see below).
If I comment the line with ddns restart everything works as it needs - mwan3 isn't locked.

I found these lines with `sleep` in `/usr/lib/ddns/dynamic_dns_updater.sh` script . In my case $CHECK_SECONDS = 300.
I can't understand why these sleeps lock mwan3 continuously if I restart DDNS from `mwan3.user`.

```
# now we wait for check interval before testing if update was recognized
[ $DRY_RUN -eq 0 ] && {
write_log 7 "Waiting $CHECK_SECONDS seconds (Check Interval)"
sleep $CHECK_SECONDS &
PID_SLEEP=$!
wait $PID_SLEEP	# enable trap-handler
PID_SLEEP=0
} || write_log 7 "Dry Run: NO Check Interval waiting"
``` 

**mwan3.user**
```
if [ "${ACTION}" = "ifdown" ] || [ "${ACTION}" = "ifup" ] || [ "${ACTION}" = "connected" ] || [ "${ACTION}" = "disconnected" ] ; then
	# Exclude events for interfaces loopback and self
	if [ "${INTERFACE}" != "loopback" ] && [ "${INTERFACE}" != "self" ] ; then
		if [ "${ACTION}" = "connected" ] || [ "${ACTION}" = "disconnected" ] ; then
			#Reload DDNS
			#{sleep 3; /etc/init.d/ddns restart}
			# Recreate XFRM interfaces
			sleep 5
			/etc/mwan3_xfrm.sh
			#Restart DNSCRYPT-PROXY
			sleep 3
			/etc/init.d/dnscrypt-proxy restart
			# Restart Strongswan
			sleep 10
			/etc/init.d/zswanctl restart
		fi
		# Send email notification
		(/usr/bin/mailsend -to xxx -from xxx -starttls -port 587 -auth \
		-smtp smtp.gmail.com -sub "$HOSTNAME $INTERFACE $ACTION" +cc +bc -user "xxxxx" \
		-pass "xxxx" -cs "us-ascii" -enc-type "7bit" -M "mwan3: $INTERFACE $ACTION" >/dev/null) &
	fi
fi
```

jow-, 2024-07-28:
The lock file descriptor is likely leaked into the backgrounded sleep command which as a consequence holds it open, so it never gets released further up the call stack, stalling the script.

feckert, 2024-07-29:
Thanks for the hint. I've also noticed that before.
I have therefore sent a patch for the [procd](https://patchwork.ozlabs.org/project/openwrt/patch/20221220092858.4456-1-fe@dev.tdt.de/) which adds a new function `procd_unlock`.
I have already fixed this in my patchset by using this new function.
```

From: Florian Eckert <fe@dev.tdt.de>
Date: Tue, 20 Dec 2022 10:50:52 +0100
Subject: [PATCH] mwan3: add procd_unlock on mwan3 user scripts

Redmine-patch-id: 7425

---
 net/mwan3/files/etc/hotplug.d/iface/16-mwan3-user | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/net/mwan3/files/etc/hotplug.d/iface/16-mwan3-user b/net/mwan3/files/etc/hotplug.d/iface/16-mwan3-user
index 0960a2628784b597a1e2eefc5157eb3d5fa44adb..3e65809eb5165ca9fbf7e50a9b99617c33c83d93 100644
--- a/net/mwan3/files/etc/hotplug.d/iface/16-mwan3-user
+++ b/net/mwan3/files/etc/hotplug.d/iface/16-mwan3-user
@@ -21,6 +21,8 @@
 
 	[ -x /etc/mwan3.user ] || chmod 755 /etc/mwan3.user
 	env -i ACTION="$ACTION" INTERFACE="$INTERFACE" DEVICE="$DEVICE" /etc/mwan3.user
+
+	[ "$MWAN3_SHUTDOWN" != 1 ] && procd_unlock
 }
 
 exit 0
-- 
```

jow-, 2024-07-29:
This will likely not fix the problem if mwan3.user launches a background process which inherits descriptor 1000 since the unlock logic happens after the user code was invoked.

feckert, 2024-07-29:
You could be right. I will add the change to the mwan3 as you suggested.
How does it look independently with the [procd_unlock](https://patchwork.ozlabs.org/project/openwrt/patch/20221220092858.4456-1-fe@dev.tdt.de/) function? Can you merge this into `package/system/procd/files/procd.sh`?
It could also be useful elsewhere and is missing in the prcod.sh API from my point of view.

jow-, 2024-07-29:
Not sure about `procd_unlock` tbh. If you want to lock critical sections it is better to use the `flock` subshell invocation pattern:

```bash
(
  flock 123 || exit 1;
  # do stuff
) 123>/var/lock/something
```

The `procd_lock` method hardcodes descriptor number and derives the lockfile path from init script context so it is not really a freestanding library function usable outside init scripts. It will also cause any user of it to block init script operations or be blocked itself if the init script happens to run simultaneously (e.g. as a consequence of a config reload operation). I'd argue that this is not desired behavior for the majority of use-cases.

Them wan3 implementation is somewhat special here since extensive parts of the logic happen in the init script while other services simply use the init script to initialize a separate service executable.

If you want to prevent any non-initscript code to run concurrently with the mwan3 init script, it is likely better to use this pattern:

```bash
(
    flock 123 || exit 1
    # critical section
) 123>/var/lock/procd_mwan3.lock
```

feckert, 2024-10-29:
Fixed with commit https://github.com/openwrt/packages/commit/619629ce85991fbb5aa8677ea076ab2de1043ba4
For the correct solution, I have to refactor the locking as mentioned in by @rindeal in https://github.com/openwrt/openwrt/issues/15994#issuecomment-2258884581. Unfortunately, I don't have the time right now. Therefore I have merged the workaround as suggested by @jow- in https://github.com/openwrt/openwrt/issues/15994#issuecomment-2254677758