openwrt/openwrt #15996: Netgear WNDAP360 no wifi links available

rodin-ia, 2024-07-23:
### Describe the bug

After installing from scratch (u-boot->initramfs-> sysupgrade) the latest version owrt 23.05.4 no wifi adapters available

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ath79/generic

### Device

Netgear WNDAP360

### Image kind

Official downloaded image

### Steps to reproduce

Installation from scratch according to this manual https://openwrt.org/toh/netgear/wndap360

### Actual behaviour

root@OpenWrt:/# ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP qlen 1000
    link/ether 00:03:7f:e0:00:96 brd ff:ff:ff:ff:ff:ff
3: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 00:03:7f:e0:00:96 brd ff:ff:ff:ff:ff:ff
root@OpenWrt:/# 

### Expected behaviour

6: phy0-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 6c:b0:ce:fc:90:e0 brd ff:ff:ff:ff:ff:ff
7: phy1-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 6c:b0:ce:fc:90:e1 brd ff:ff:ff:ff:ff:ff

### Additional info

[u-boot.log](https://github.com/user-attachments/files/16354149/u-boot.log) - installation log
[kernel.log](https://github.com/user-attachments/files/16354157/kernel.log) - first boot dmsg
[logread.log](https://github.com/user-attachments/files/16354173/logread.log) - first boot logread

lspci -vvvv
00:11.0 Ethernet controller: Qualcomm Atheros AR922x Wireless Network Adapter (rev 01)
        Subsystem: Qualcomm Atheros AR9220-AC1A [AVM Fritz!Box FON WLAN 7270 v3]
        Device tree node: /sys/firmware/devicetree/base/ahb/apb/pcie-controller@17010000/wifi@0,11
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz+ UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 0
        Region 0: Memory at 10000000 (32-bit, non-prefetchable) [disabled] [size=64K]
        Capabilities: [40] Capability ID 0x80 [0000]
        Capabilities: [80] Null
lspci: Unable to load libkmod resources: error -2

00:12.0 Ethernet controller: Qualcomm Atheros AR922x Wireless Network Adapter (rev 01)
        Subsystem: Qualcomm Atheros AR9220-AC1A [AVM Fritz!Box FON WLAN 7270 v3]
        Device tree node: /sys/firmware/devicetree/base/ahb/apb/pcie-controller@17010000/wifi@0,12
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz+ UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 0
        Region 0: Memory at 10010000 (32-bit, non-prefetchable) [disabled] [size=64K]
        Capabilities: [40] Capability ID 0x80 [0000]
        Capabilities: [80] Null

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rodin-ia, 2024-07-23:
firmaware OpenWrt 19.07.9 r11405-2a3558b0de / LuCI openwrt-19.07 branch git-22.045.73925-36e5c1c - wifi available
[dmesg 19.07.09.txt](https://github.com/user-attachments/files/16354309/dmesg.19.07.09.txt) - dmesg

rodin-ia, 2024-07-23:
The problem was solved by installing the `kmod-owl-loader` package

namiltd, 2024-07-24:
Shouldn't it be added permanently to the definition of this router in  target\linux\ath79\image\generic.mk?

rodin-ia, 2025-05-09:
The full story of the problem and solution is well described in the issue: https://github.com/openwrt/openwrt/issues/11025

closed 2025-05-09 by rodin-ia