openwrt/openwrt #15997: [SNAPSHOT r26990-06b37a5856] ipq806x/generic crash on Meraki MR-52 bringing up Wifi

rkboni, 2024-07-23:
### Describe the bug

After a `sysupgrade` to the latest snapshot, the MR-52 kernel-faults initializing Wifi on boot:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.41 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.3.0 r26990-06b37a5856) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sun Jul 21 22:32:38 2024
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Meraki MR52
[    0.000000] OF: fdt: Ignoring memory range 0x41500000 - 0x44000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] OF: reserved mem: 0x40000000..0x40ffffff (16384 KiB) nomap non-reusable nss@40000000
[    0.000000] OF: reserved mem: 0x41000000..0x411fffff (2048 KiB) nomap non-reusable smem@41000000
[    0.000000] OF: reserved mem: 0x41200000..0x414fffff (3072 KiB) nomap non-reusable rsvd@41200000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000] percpu: Embedded 13 pages/cpu s21396 r8192 d23660 u53248
[    0.000000] pcpu-alloc: s21396 r8192 d23660 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Kernel command line: loader=u-boot part=bootkernel2 console=ttyMSM0,115200n8 ubi.mtd=ubi ubi.mtd=art
[    0.000000] Bootloader command line not present
[    0.000000] Unknown kernel command line parameters "loader=u-boot part=bootkernel2", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113426
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 440880K/457728K available (7567K kernel code, 628K rwdata, 2056K rodata, 1024K init, 249K bss, 16848K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000002] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000022] Switching to timer-based delay loop, resolution 160ns
[    0.000328] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000356] CPU: Testing write buffer coherency: ok
[    0.000408] pid_max: default: 32768 minimum: 301
[    0.008528] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.008549] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.016211] qcom_scm: convention: smc legacy
[    0.017718] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1.
[    0.017948] Setting up static identity map for 0x44300000 - 0x44300060
[    0.018161] rcu: Hierarchical SRCU implementation.
[    0.018171] rcu: 	Max phase no-delay instances is 1000.
[    0.018812] smp: Bringing up secondary CPUs ...
[    0.020939] smp: Brought up 1 node, 2 CPUs
[    0.020959] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.020973] CPU: All CPU(s) started in SVC mode.
[    0.037041] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.037246] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.037280] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.041649] pinctrl core: initialized pinctrl subsystem
[    0.044312] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.044706] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.046507] thermal_sys: Registered thermal governor 'step_wise'
[    0.047268] cpuidle: using governor ladder
[    0.047343] cpuidle: using governor menu
[    0.082642] usbcore: registered new interface driver usbfs
[    0.082789] usbcore: registered new interface driver hub
[    0.082886] usbcore: registered new device driver usb
[    0.082953] pps_core: LinuxPPS API ver. 1 registered
[... 584 lines cut ...]

### Device

Meraki MR52

### Image kind

Official downloaded image

### Steps to reproduce

Install SNAPSHOT build `r26990-06b37a5856` (upgrading from `SNAPSHOT r26340-9bdaebaff3 (kernel 6.6.30)`) via `sysupgrade -c -v /tmp/firmware.bin`.  Reboot, note OOPS in kernel log.

### Actual behaviour

Kernel crash bringing up Wifi.  
Only 1 of 3 Wifi interfaces are enabled, and connecting via that interface still works.

### Expected behaviour

Kernel doesn't crash opening one of the Wifi devices.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rkboni, 2024-07-23:
Also, what's up with the bot complaining about version / release / target / subtarget?  Does it not like my `dmesg` dump in the description?

rkboni, 2024-07-23:
Looks related to https://github.com/openwrt/openwrt/issues/15959, though that's ipq40xx -- but both related to atk10k driver/firmware.  Maybe tellingly, I don't see the `txpower` being off (set to `* 23 dBm ( 199 mW)`).

dannil, 2024-07-24:
I can post here for completeness as well, I'm seeing this as well on Netgear R7800 (posted first at https://github.com/openwrt/openwrt/issues/15959#issuecomment-2233837860), it has definitely something to do with the mac80211 6.9.9 backports in 1bfcc1ea8a78ae834055bf554aa45720c09ea919 and/or 7ca4190c186b6018f66421ce04daed5deeefdabb as the errors don't show up before that commit.

markmentovai, 2024-07-31:
I’ve shared an analysis of this bug at https://github.com/openwrt/openwrt/issues/15959#issuecomment-2259603978 (that report beat this one, but they’re the same bug).

closed 2025-03-31 by openwrt-bot