openwrt/openwrt #15999: Drastic drop in 2.4GHz throughput is observed in OoenWrt 23.05.4 due to a mac80211 backport.

vochong, 2024-07-24:
### Describe the bug

Some drastic drop (about by half)  in 2.4GHz throughput is observed in OoenWrt 23.05.4 due to the recent mac80211 backports.

Reverting this change  https://github.com/torvalds/linux/commit/4e5bd2287021823c10862379312648dd55e17f49 would restore the throughput back to normal.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ipq806x/generic

### Device

NETGEAR Nighthawk X4S R7800

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

egc112, 2024-07-25:
I can confirm these findings and after building and removing the offending line my 2.4 GHz speed is doubled again because now 2 streams are used again

Question: Does this also apply to Main builds with kernel 6.6?

tmn505, 2024-07-25:
Follow-up on the issue: https://lore.kernel.org/all/29f7aa2c233277ab4bb56bcff8e228036ee96a2e.camel@sipsolutions.net
Please test.

vochong, 2024-07-26:
@nbd168

AgustinLorenzo, 2024-07-26:
> I can confirm these findings and after building and removing the offending line my 2.4 GHz speed is doubled again because now 2 streams are used. I compiled my own build using 23.05 Snapshot, you can find the patch I used at [my repo](https://github.com/egc112/OpenWRT-egc-add-on/blob/main/999-revert-set-station-rx-nss-on-reconfig-egc.patch)
> 
> Question: Does this also apply to Main builds with kernel 6.6?

Hi @egc112 

In effect, kernel 6.6 is also affected, see my comment on @hauke PR.

Regards, Agustin

brada4, 2024-07-27:
Cc TUF ax4200 , rt ax53u, c7v5

hauke, 2024-08-03:
Fixed in #16007

closed 2024-08-03 by hauke

f45tb00t, 2024-09-10:
I definitely can confirm. Updated my Ubiquiti LR 6 to 23.05.4 and the downstream decreased massively. Massively means from 1000++mbit down ~6mbit.  A rollback to .3 did the trick and fixed the situation.

Could you tell us please when a fixed version will be ready to install?

brada4, 2024-09-10:
You can try 23.05-SNAPSHOT (just install, document, get back to 23.05.3)

f45tb00t, 2024-09-10:
> You can try 23.05-SNAPSHOT (just install, document, get back to 23.05.3)

To be honest for this week I am done with TShooting.... Had also issues with my FW and other network equipment. Maybe next week.