openwrt/openwrt #16000: Zbtlink ZBT-WE3526 WiFi 5GHz signal strength at 3 dBm

technout, 2024-07-24:
### Describe the bug

I did read the solved and related issue [#13427](https://github.com/openwrt/openwrt/issues/13427)

Experiencing the same problem with my new ZBT-WE3526 router. 5G Wifi signal strength is max at 3 dBm. Like the other poster reported: the dropdown menu "Maximum transmit power" gives the options 0 to 3 dBm (1 mW) and currently at driver default.

So maybe my ZBT-WE3526 is actually ZBT-WE1326 and i did flash the 1326 OpenWrt 23.05.4 firmware on it. Sadly the problems with wifi where bigger with that firmware, both 2.4GHz and 5GHz could not be enabled. Even after a reboot or wifi settings change.
So i went back to 3526 firmware, the 2.4 GHz wifi is working normal on this firmware btw.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

Zbtlink ZBT-WE3526

### Image kind

Official downloaded image

### Steps to reproduce

Configure the wifi networks for 2.4 and 5G for normal use.
Enabling the 5G wifi.
Trying to find and connect the 5G wifi from a (mobile) client device.
Trying different language regio's and different wifi settings like channel and bandwith, same result.

### Actual behaviour

Wifi 5GHz signal very week, unusable for normal use.
Experiencing that 5G wifi is only working at very close distance to the router, max 3 meters and no object in between.
Experiencing that 5G wifi disconnects very often on the client or client could not find the 5G network at all.

### Expected behaviour

Wifi 5GHz signal strong and working correctly at larger distance from the router, like most 5G wifi will work.

### Additional info

```
root@openwrt_new:~# cat /etc/config/wireless

config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option cell_density '0'
	option country 'DE'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
	option channel '120'
	option band '5g'
	option htmode 'VHT80'
	option cell_density '0'
	option country DE'

config wifi-iface 'wifinet2'
[... 434 lines cut ...]
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "wifi5"
          Access Point: 48:BC:5B:xx:xx:xx
          Mode: Master  Channel: 120 (5.600 GHz)  HT Mode: VHT80
          Center Channel 1: 122 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/b/g/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
</p>
</details> 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2024-07-25:
Please try ZBT-WE1326 version and do not keep the configuration.

castillofrancodamian, 2024-07-25:
It has the same error as the mt7921 driver.

123serge123, 2024-07-25:
I'm not sure that is the main reason but there is a logical error in configuration. In dmesg logs:
`
[ 0.963526] mt7621-pci 1e140000.pcie: pcie0 no card, disable it (RST & CLK)
[ 0.970432] mt7621-pci 1e140000.pcie: PCIE1 enabled
[ 0.975312] mt7621-pci 1e140000.pcie: PCIE2 enabled
`
So only pcie1 and pcie2 slots are active where pcie1 is mt7612 and pcie2 is mt7603.
At the same time [url=https://github.com/openwrt/openwrt/blob/76a0c2932cf9b34af2543618ec90ea7afebbc703/target/linux/ramips/dts/mt7621_zbtlink_zbt-we3526.dts#L77]dts-file[/url] describe pcie0 and pcie1 only. 
As result mt7612 try to read eeprom from pcie1 node and refuse it:
`
[ 41.991647] mt76x2e 0000:01:00.0: EEPROM data check failed: 7603
`
Can you change pcie0 -> pcie1 and pcie1->pcie2 in dts-file and recompile

technout, 2024-07-25:
> Please try ZBT-WE1326 version and do not keep the configuration.

Thanks! I will try that first. Maybe that solves the problem.

technout, 2024-07-25:
> Please try ZBT-WE1326 version and do not keep the configuration.

Solved it! Simple as that. Both Wi-Fi signals are now strong and working as expected.

DragonBluep, 2024-07-25:
> > Please try ZBT-WE1326 version and do not keep the configuration.
> 
> Solved it! Simple as that. Both Wi-Fi signals are now strong and working as expected.

Does wifi LEDs work correctly? Can you provide the MAC address information (the last tree bytes of the LAN/WAN/WLAN/label MAC)?

technout, 2024-07-25:
> > > Please try ZBT-WE1326 version and do not keep the configuration.
> > 
> > 
> > Solved it! Simple as that. Both Wi-Fi signals are now strong and working as expected.
> 
> Does wifi LEDs work correctly? Can you provide the MAC address information (the last tree bytes of the LAN/WAN/WLAN/label MAC)?

LEDs are working correctly as far as i can see: for LAN ports, power and wifi.
Wifi2.4 blinks green and wifi5 blinks red.

The MAC address from iwinfo is the same as the sticker on the backside, so that seems alright.
```
root@OpenWrt:~# iwinfo
phy0-ap0  ESSID: "wifi1"
          Access Point: xx:xx:xx:6A:2C:D0
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "wifi5"
          Access Point: xx:xx:xx:6A:2C:D1
          Mode: Master  Channel: 52 (5.260 GHz)  HT Mode: VHT80
          Center Channel 1: 58 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

DragonBluep, 2024-07-25:
What is MAC address on the product label? It's missing from the we3526. I'm going to fix them all at once.

victron, 2024-07-26:
> Please try ZBT-WE1326 version and do not keep the configuration.

I'm owner same device and can confirm that it's working (even with my config.)
Thank you for work !!!

closed 2024-07-28 by openwrt-bot

dfateyev, 2024-08-17:
> There are multiple versions of the ZBT Router, I've the Wiflyer (white) with 256 MB memory.

It is probably the Wiflyer device that we investigated [earlier](https://github.com/openwrt/openwrt/pull/11220#issuecomment-1336263591).
Also note [this](https://github.com/openwrt/openwrt/issues/13427#issuecomment-1714615212), WF3526-P was already added to the model list as an alias for ZBT WE1326.