openwrt/openwrt #16002: Xiaomi AX3000T softbricks due to different Winbond flash chip

andi-at-1, 2024-07-25:
### Describe the bug

Xiaomi AX3000T surfaced with different, unsupportet flash chips, **leading to a softbrick when its flashed with openwrt regardless of what method**
Soc: Mediatek MT7981BA, Ram: Winbond W632GU6QB-11, **Flash: Winbond 25n01kvzeir**

Flash is not supported with current release and snapshot version. However immortalwrt has a patch so it works fine:
https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/785

ToDo:
* Rewrite the wiki with a big warning that it can softbrick with the current version!
* Include the patch in stable and snapshot asap.

There is no physical difference visible to the routers, however you can spot the chip if you shine a torch through the underside of the router. Chip is located on the left site, you can read either "Winbond" or " ESMT" quite good. Here is a picture of the location: https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/790?u=andi-at-1

If anyone is already working on this just ignore it :)

### OpenWrt version

OpenWrt 23.05.4  & snapshot

### OpenWrt release

OpenWrt 23.05.4  & snapshot

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Steps to reproduce

try to flash anything to a router with a Winbond chip, it will softbrick. (Xiaomi recovery tool works)
Immortalwrt works just fine

### Actual behaviour

it softbricks

### Expected behaviour

no softbrick

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-07-29:
Can you try this commit:
https://github.com/robimarko/openwrt/commit/686d8bd0b8df64de619e34a33fdd68176c8375d5

It should fix it for master, as for 23.05, that would require backporting the whole KV revision support first.

robimarko, 2024-07-29:
@rx78gp01 Thanks for catching that, it's been ages since I added an SPI-NAND device.
I am now comparing datasheets as both `.ecc` and` .free` OPS are incorrect for W25N01KV as it uses 8 byte parity area instead.

andi-at-1, 2024-07-29:
https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/819?u=andi-at-1
I just tested this build, initramfs and sysupgrade works fine. Didn't test uboot

robimarko, 2024-07-30:
@rx78gp01 Can you sanity check this:
```
static int w25n01kv_ooblayout_ecc(struct mtd_info *mtd, int section,
				  struct mtd_oob_region *region)
{
	if (section > 3)
		return -ERANGE;

	region->offset = 64 + (8 * section);
	region->length = 7;

	return 0;
}

static int w25n01kv_ooblayout_free(struct mtd_info *mtd, int section,
				   struct mtd_oob_region *region)
{
	if (section > 3)
		return -ERANGE;

	region->offset = (16 * section) + 2;
	region->length = 8;

	return 0;
}

static const struct mtd_ooblayout_ops w25n01kv_ooblayout = {
	.ecc = w25n01kv_ooblayout_ecc,
	.free = w25n01kv_ooblayout_free,
};
```

andi-at-1, 2024-07-30:
if i should test something please build it for me, i have two routers sitting around but definitely not enough skills to compile it by myself (assuming make menuconfig needs more than just selecting the chipset...)

robimarko, 2024-07-31:
@rx78gp01 This should be it:
https://github.com/robimarko/openwrt/commit/fac5b45b4e70bcb5043b238e0440745d8b745ab7

robimarko, 2024-08-04:
Indeed, 96 bytes (4*16 byte user data + 4*8 byte parity)

andi-at-1, 2024-08-04:
> > This should be it: [robimarko@fac5b45](https://github.com/robimarko/openwrt/commit/fac5b45b4e70bcb5043b238e0440745d8b745ab7)
> 
> That's right. It may need to set 96 oobsize.
> 
> ```
> NAND_MEMORG(1, 2048, 96, 64, 1024, 20, 1, 1, 1),
> ```
> 
> > if i should test something please build it for me, i have two routers sitting around but definitely not enough skills to compile it by myself (assuming make menuconfig needs more than just selecting the chipset...)
> 
> You can try [this](https://github.com/rx78gp01/openwrt/releases/download/v23.05.4/openwrt-23.05.4-mediatek-filogic-ax3000t-firmwares-tiny_2024-08-02_15-29-W25N01KV_oobsize96.zip) test build [23.05 patch file](https://github.com/user-attachments/files/16487587/0151-mtd-spinand-winbond-Support-for-W25MxxGV-W25NxxKV-W25NxxxW-series-oobsize96.patch)

Thank you! I've installed it and:
* Initramfs WORKS
* Sysupgrade WORKS

Then i tried to install uboot and something went wrong, step 1 complained about not finding mtd8 (mabe some copy&paste error) and i shouldn't have ignored that... i'll investigate tomorrow, but that shouldn't be related to the patch

andi-at-1, 2024-08-05:
Ok i think the bootloader has a problem, its lost at stage 1?

Commands i entered for bootloader flashing. First command failed: "ubidetach: error!: cannot detach "/dev/mtd8", but the rest was working. (yeah i should have stopped there...)

<details>
root@OpenWrt:/tmp# opkg install kmod-mtd-rw_5.15.164+git-20160214-2_aarch64_cort
ex-a53.ipk
Installing kmod-mtd-rw (5.15.164+git-20160214-2) to root...
Configuring kmod-mtd-rw.
root@OpenWrt:/tmp# insmod /lib/modules/$(uname -r)/mtd-rw.ko i_want_a_brick=1

root@OpenWrt:/tmp# ubidetach -p /dev/mtd8; ubiformat /dev/mtd8 -y; ubiattach -p /dev/mtd8
ubidetach: error!: cannot detach "/dev/mtd8"
           error 19 (No such device)
ubiformat: mtd8 (nand), size 117440512 bytes (112.0 MiB), 896 eraseblocks of 131 072 bytes (128.0 KiB), min. I/O size 2048 bytes
libscan: scanning eraseblock 895 -- 100 % complete
ubiformat: 896 eraseblocks have valid erase counter, mean value is 2
ubiformat: formatting eraseblock 895 -- 100 % complete
UBI device number 0, total 896 LEBs (113770496 bytes, 108.5 MiB), available 870                                                                                                                                                              LEBs (110469120 bytes, 105.3 MiB), LEB size 126976 bytes (124.0 KiB)

root@OpenWrt:/tmp# ubimkvol /dev/ubi0 -n 0 -N ubootenv -s 128KiB
Volume ID 0, size 2 LEBs (253952 bytes, 248.0 KiB), LEB size 126976 bytes (124.0                                                                                                                                                              KiB), dynamic, name "ubootenv", alignment 1

root@OpenWrt:/tmp# ubimkvol /dev/ubi0 -n 1 -N ubootenv2 -s 128KiB
Volume ID 1, size 2 LEBs (253952 bytes, 248.0 KiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "ubootenv2", alignment 1
root@OpenWrt:/tmp# ubimkvol /dev/ubi0 -n 2 -N recovery -s 10MiB
Volume ID 2, size 83 LEBs (10539008 bytes, 10.0 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "recovery", alignment 1

root@OpenWrt:/tmp# ubiupdatevol /dev/ubi0_2 /tmp/openwrt-23.05.4-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-initramfs-recovery.itb

root@OpenWrt:/tmp# mtd write /tmp/openwrt-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-preloader.bin BL2
Unlocking BL2 ...

Writing from /tmp/openwrt-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-preloader.bin to BL2 ...
root@OpenWrt:/tmp# mtd write /tmp/openwrt-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-bl31-uboot.fip FIP
Unlocking FIP ...
Writing from /tmp/openwrt-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-bl3 1-uboot.fip to FIP ...

root@OpenWrt:/tmp# sysupgrade -n /tmp/openwrt-23.05.4-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-squashfs-sysupgrade.itb
Sun Aug  4 18:11:46 UTC 2024 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed
</details>

And thats what i get now over serial with no chance of any interaction. Suspicious messages:
"Loading Environment from UBI... spi-nand: spi_nand spi_nand@0: unknown raw ID ef ae 21 00" 
"BL2: v2.9(release):OpenWrt v2023-07-24-00ac6db3-2 (mt7981-spim-nand-ddr3)"
the build is old, mabe this uboot don´t have the patch? 

full boot log:
<details>
F0: 102B 0000
FA: 1040 0000
FA: 1040 0000 [0200]
F9: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 2400 0041 [0000]
G0: 1190 0000
EC: 0000 0000 [1000]
T0: 0000 0236 [010F]
Jump to BL

NOTICE:  BL2: v2.9(release):OpenWrt v2023-07-24-00ac6db3-2 (mt7981-spim-nand-ddr3)
NOTICE:  BL2: Built : 22:14:18, Jul 15 2024
NOTICE:  WDT: Cold boot
NOTICE:  WDT: disabled
NOTICE:  EMI: Using DDR3 settings
NOTICE:  EMI: Detected DRAM size: 256MB
NOTICE:  EMI: complex R/W mem test passed
NOTICE:  CPU: MT7981 (1300MHz)
NOTICE:  SPI_NAND parses attributes from parameter page.
NOTICE:  SPI_NAND Detected ID 0xef
NOTICE:  Page size 2048, Block size 131072, size 134217728
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.9(release):OpenWrt v2023-07-24-00ac6db3-2 (mt7981-spim-nand-ddr3)
NOTICE:  BL31: Built : 22:14:18, Jul 15 2024

U-Boot 2023.07.02-OpenWrt-r24012-d8dd03c46f (Jul 15 2024 - 22:14:18 +0000)

CPU:   MediaTek MT7981
Model: Xiaomi Router AX3000T
DRAM:  256 MiB
Core:  32 devices, 14 uclasses, devicetree: embed
Loading Environment from UBI... spi-nand: spi_nand spi_nand@0: unknown raw ID ef ae 21 00
Partition ubi not found!

** Cannot find mtd partition "ubi"
In:    serial@11002000
Out:   serial@11002000
Err:   serial@11002000
reset button found
Loading Environment from UBI... spi-nand: spi_nand spi_nand@0: unknown raw ID ef ae 21 00
Partition ubi not found!

** Cannot find mtd partition "ubi"
Net:   
Warning: ethernet@15100000 (eth0) using random MAC address - 86:d3:fb:1f:d0:63
eth0: ethernet@15100000
Error, no UBI device selected!
Error, no UBI device selected!
spi-nand: spi_nand spi_nand@0: unknown raw ID ef ae 21 00
MTD device ubi not found, ret -19
resetting ...
</details>

robimarko, 2024-08-05:
U-Boot is missing support for the same chip.

I will send the patch upstream today for Linux, will see if U-Boot is easy to add as well.

andi-at-1, 2024-08-05:
https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/823

According to here its mabe already patched?

robimarko, 2024-08-05:
Yes, with this horrible MTK SDK patch:
https://github.com/openwrt/openwrt/blob/main/package/boot/uboot-mediatek/patches/100-22-mtd-spi-nand-backport-from-upstream-kernel.patch

andi-at-1, 2024-08-05:
Ah okay, just wanted to mention that 

I used the files from @rx78gp01 testbuild and thought uboot would be no problem czz of the patch mentioned haha,  mabe i need some time to get it back to life for the next test 😄 

But the patch worked for initramfs and sysupdate if that already helps to get it into the official builds 👍

andi-at-1, 2024-08-06:
Router is back to life, i posted the recovery procedure in the support thread.

Thank you very much for your work! If i should test anything feel free to message me

commit 81481c80 (2024-08-12): generic: 6.6: add Winbond W25N01KV support

It seems that some Xiaomi AX3000T boards changed to using Winbond W25N01KV
SPI-NAND which is not supported in OpenWrt nor upstream kernel.

So, add a pending patch to support it as upstream supports rest of the KV
revision models.

Fixes: #16002
Link: https://github.com/openwrt/openwrt/pull/16088
PR #16088 (2024-08-12): generic: 6.6: add Winbond W25N01KV support
It seems that some Xiaomi AX3000T boards changed to using Winbond W25N01KV SPI-NAND which is not supported in OpenWrt nor upstream kernel.

So, add a pending patch to support it as upstream supports rest of the KV revision models.

Sent upstream: https://patchwork.ozlabs.org/project/linux-mtd/patch/20240805175125.6658-1-robimarko@gmail.com/

Fixes: #16002
closed 2024-08-12 by openwrt-bot