openwrt/openwrt #16006: [SNAPSHOT r26993-74879140a1] NanoPi R5C with NFA765 wi-fi module - Broken DFS

FrozenGamesYT, 2024-07-25:
### Describe the bug

On the newest SNAPSHOT build it seems like the dfs on 5gHz is broken.

_Thu Jul 25 23:00:17 2024 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->DFS
Thu Jul 25 23:00:17 2024 daemon.notice hostapd: phy1-ap0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
Thu Jul 25 23:00:17 2024 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Thu Jul 25 23:00:17 2024 daemon.err hostapd: Interface initialization failed
Thu Jul 25 23:00:17 2024 daemon.notice hostapd: phy1-ap0: interface state DFS->DISABLED
Thu Jul 25 23:00:17 2024 daemon.notice hostapd: phy1-ap0:   AP-DISABLED_

### OpenWrt version

r26993-74879140a1

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5C

### Image kind

Official downloaded image

### Steps to reproduce

Set DFS channel on 5gHz with device NanoPi R5C with nfa765 Wi-Fi module on newest snapshot build

### Actual behaviour

DFS check fails = not working Wi-Fi

### Expected behaviour

Expected DFS to work with working Wi-Fi

### Additional info

I think it correlates to the kernel update to 6.6.41 but I am unsure

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tenten8401, 2024-07-25:
Seeing the same issue on my R5C / NFA765, no DFS channels work

wlshdjj, 2024-07-26:
The same problem to QCA2066 on x86_64:

Fri Jul 26 15:52:17 2024 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Fri Jul 26 15:52:17 2024 daemon.notice hostapd: wlan1: interface state HT_SCAN->DFS
Fri Jul 26 15:52:17 2024 daemon.notice hostapd: wlan1: DFS-CAC-START freq=5300 chan=60 sec_chan=1, width=2, seg0=50, seg1=0, cac_time=60s
Fri Jul 26 15:52:17 2024 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Fri Jul 26 15:52:17 2024 daemon.err hostapd: Interface initialization failed
Fri Jul 26 15:52:17 2024 daemon.notice hostapd: wlan1: interface state DFS->DISABLED
Fri Jul 26 15:52:17 2024 daemon.notice hostapd: wlan1: AP-DISABLED

1715173329, 2024-07-27:
did you set country code?

tenten8401, 2024-07-27:
Yes, country code is set to US and the channels are legal and there is no radar nearby. I originally had this happen after updating a fully working config but it persists even setting up from scratch

-Isaac

brada4, 2024-07-27:
Extract channel list from `iw list` and share lsusb and ethtool -i from adapter. Not a solution but to completely document issue.

FrozenGamesYT, 2024-07-27:
> Extract channel list from `iw list` and share lsusb and ethtool -i from adapter. Not a solution but to completely document issue.

iw list:

Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 16
        max scan IEs length: 138 bytes
        max # sched scan SSIDs: 16
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
        Band 1:
                Capabilities: 0x19e3
                        RX LDPC
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x000b82100040):
                                +HTC HE Supported
                                TWT Requester
                                Dynamic BA Fragementation Level: 1
                                Broadcast TWT
                                OM Control
                                RX Control Frame to MultiBSS
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x02304c890d018008020c00):
                                HE40/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                DCM Max Constellation: 1
                                DCM Max Constellation Rx: 1
                                SU Beamformer
                                SU Beamformee
                                Beamformee STS <= 80Mhz: 3
                                Sounding Dimensions <= 80Mhz: 1
                                PPE Threshold Present
                                Max NC: 1
                                20MHz in 40MHz HE PPDU 2.4GHz
                                TX 1024-QAM
                                RX 1024-QAM
                        HE RX MCS and NSS set <= 80 MHz
                                1 streams: MCS 0-11
                                2 streams: MCS 0-11
                                3 streams: not supported
                                4 streams: not supported
                                5 streams: not supported
                                6 streams: not supported
                                7 streams: not supported
                                8 streams: not supported
                        HE TX MCS and NSS set <= 80 MHz
[... 692 lines cut ...]
                * [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
                * [ BSS_COLOR ]: BSS coloring support
                * [ POWERED_ADDR_CHANGE ]: can change MAC address while up

lspci:

0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
**0000:01:00.0 Network controller: Qualcomm Technologies, Inc QCNFA765 Wireless Network Adapter (rev 01)**
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)

ethtool -i phy1-ap0:

driver: ath11k_pci
version: 6.6.41
firmware-version: N/A
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

brada4, 2024-07-28:
Please leave only 5ghz channel info, dfs channels in there, supported , AND FORMAT CODE BLOCK

FrozenGamesYT, 2024-07-29:
> Please leave only 5ghz channel info, dfs channels in there, supported , AND FORMAT CODE BLOCK

done, sorry, new to github forums/issues posting

brada4, 2024-07-29:
In the meantime set automatic 80mhz channel select, its enough choices for that even excluding DFS.

FrozenGamesYT, 2024-07-29:
> In the meantime set automatic 80mhz channel select, its enough choices for that even excluding DFS.

i have but i do need 160mHz for wireless VR streaming which needs DFS due to 160mHz width

wlshdjj, 2024-07-30:
Maby I got it.
Seeing in this patch: https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/ath.git/commit/?id=f019f4dff2e4cb8704dc608496e3f2829de3e919
`+	if (ab->hw_params.support_dual_stations) {
+		limits[0].max = 2;
+		limits[0].types |= BIT(NL80211_IFTYPE_STATION);
+
+		limits[1].max = 1;
+		limits[1].types |= BIT(NL80211_IFTYPE_AP);
+		if (IS_ENABLED(CONFIG_MAC80211_MESH) &&
+		    ab->hw_params.interface_modes & BIT(NL80211_IFTYPE_MESH_POINT))
+			limits[1].types |= BIT(NL80211_IFTYPE_MESH_POINT);
+
+		combinations[0].limits = limits;
+		combinations[0].n_limits = 2;
+		combinations[0].max_interfaces = ab->hw_params.num_vdevs;
+		combinations[0].num_different_channels = 2;
+		combinations[0].beacon_int_infra_match = true;
+		combinations[0].beacon_int_min_gcd = 100;
+	} else {
+		limits[0].max = 1;
+		limits[0].types |= BIT(NL80211_IFTYPE_STATION);
+
+		limits[1].max = 16;
+		limits[1].types |= BIT(NL80211_IFTYPE_AP);
+
+		if (IS_ENABLED(CONFIG_MAC80211_MESH) &&
+		    ab->hw_params.interface_modes & BIT(NL80211_IFTYPE_MESH_POINT))
+			limits[1].types |= BIT(NL80211_IFTYPE_MESH_POINT);
+
+		combinations[0].limits = limits;
+		combinations[0].n_limits = 2;
+		combinations[0].max_interfaces = 16;
+		combinations[0].num_different_channels = 1;
+		combinations[0].beacon_int_infra_match = true;
+		combinations[0].beacon_int_min_gcd = 100;
+		combinations[0].radar_detect_widths = BIT(NL80211_CHAN_WIDTH_20_NOHT) |
+							BIT(NL80211_CHAN_WIDTH_20) |
+							BIT(NL80211_CHAN_WIDTH_40) |
+							BIT(NL80211_CHAN_WIDTH_80) |
+							BIT(NL80211_CHAN_WIDTH_80P80) |
+							BIT(NL80211_CHAN_WIDTH_160);
+	}`

wlshdjj, 2024-07-30:
You can see all changes about ath11k at https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/ath.git/log/?qt=grep&q=ath11k

FrozenGamesYT, 2024-07-31:
> Maby I got it. We use it as access point, so we don't need 2 stations. Seeing in this patch: https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/ath.git/commit/?id=f019f4dff2e4cb8704dc608496e3f2829de3e919
> 
> ## For affected chips, remove radar_detect_widths because now
> num_different_channels is set to 2. radar_detect_widths can be set only when
> num_different_channels is 1, see mac80211 function wiphy_verify_combinations
> for details. This means that in affectected chips DFS cannot be enabled in AP
> mode.
> ![抓取1](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg) [ ![抓取1](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg) ](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg) [ ](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg)
> 
> Here'r two patches for fixing it up: [0990-ath11k-revert-to-support-1-station-and-2-AP-interfaces_mac.c.patch](https://github.com/user-attachments/files/16437892/0990-ath11k-revert-to-support-1-station-and-2-AP-interfaces_mac.c.patch) [0991-ath11k-fix-dfs-startup-to-fail-after-adding-dual-stations-support_mac.c.patch](https://github.com/user-attachments/files/16437894/0991-ath11k-fix-dfs-startup-to-fail-after-adding-dual-stations-support_mac.c.patch)

how can we apply the patches?

wlshdjj, 2024-08-01:
> > Maby I got it. We use it as access point, so we don't need 2 stations. Seeing in this patch: https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/ath.git/commit/?id=f019f4dff2e4cb8704dc608496e3f2829de3e919
> > ## For affected chips, remove radar_detect_widths because now
> > num_different_channels is set to 2. radar_detect_widths can be set only when
> > num_different_channels is 1, see mac80211 function wiphy_verify_combinations
> > for details. This means that in affectected chips DFS cannot be enabled in AP
> > mode.
> > ![抓取1](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg)
> >       
> >         [
> >           
> >             ![抓取1](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg)
> >           
> >         ](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg)
> >         
> >         
> >           
> >             
> >               
> >             
> >             
> >               
> >               
> >             
> >           
> >           [
> >             
> >               
> >             
> >           ](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg)
> >         
> >        [ ![抓取1](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDIwMzMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3MDVkNzE0MjM4YzkyYmViNzZkNWE5ODNjYzM4MGJkMTFiZDVkMTI3NDg2MWI1N2IwMDY2OGFhYWEzNDM2MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qS36IUj9xKvZDBlE8ZZmM1w6hTGLGvq3Rt9f9J15aFg) ](https://private-user-images.githubusercontent.com/57551746/353646473-cdde8978-d028-4bc3-a47c-d83661a70d85.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0NTgzMTMsIm5iZiI6MTcyMjQ1ODAxMywicGF0aCI6Ii81NzU1MTc0Ni8zNTM2NDY0NzMtY2RkZTg5NzgtZDAyOC00YmMzLWE0N2MtZDgzNjYxYTcwZDg1LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUF [... cut ...]
> > Here'r two patches for fixing it up: [0990-ath11k-revert-to-support-1-station-and-2-AP-interfaces_mac.c.patch](https://github.com/user-attachments/files/16437892/0990-ath11k-revert-to-support-1-station-and-2-AP-interfaces_mac.c.patch) [0991-ath11k-fix-dfs-startup-to-fail-after-adding-dual-stations-support_mac.c.patch](https://github.com/user-attachments/files/16437894/0991-ath11k-fix-dfs-startup-to-fail-after-adding-dual-stations-support_mac.c.patch)
> 
> how can we apply the patches?

Put them into /packages/mac80211/pathces/ath11k, then make.
Now it works well.

FrozenGamesYT, 2024-08-04:
still an issue in newest snapshot.