openwrt/openwrt #16011: [bcm27xx] Kernel 6.6.41 does not boot, maybe "F2FS-fs (loop0): Disable nat_bits due to incorrect cp_ver"?

schuettecarsten, 2024-07-26:
### Describe the bug

Raspberry Pi 4 (bcm27xx) does not boot when Kernel 6.6.41 is used.
Full boolog from serial console before it locks with kernel 6.6.41:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.41 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 14.1.0 r26990+7-06b37a5856a) 14.1.0, GNU ld (GNU Binutils) 2.42) #0 S
MP Fri Jul 26 07:53:10 2024
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ae00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000002ae00000..0x000000002edfffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef64600..0x000000003ef647ff (0 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef64840..0x000000003ef64878 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 19 pages/cpu s38312 r8192 d31320 u77824
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=D8:3A:DD:83:61:F9 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 console=ttyAMA0,115200 root=PARTUUID=60809ed3-02 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 3825764K/4050944K available (8576K kernel code, 942K rwdata, 2364K rodata, 448K init, 553K bss, 159644K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000288] Console: colour dummy device 80x25
[    0.000302] printk: console [tty1] enabled
[    0.000709] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000738] pid_max: default: 32768 minimum: 301
[    0.001139] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001216] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001829] cgroup: Disabling memory control group subsystem
[    0.003115] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.003343] rcu: Hierarchical SRCU implementation.
[    0.003360] rcu:     Max phase no-delay instances is 1000.
[    0.004088] smp: Bringing up secondary CPUs ...
[    0.004549] Detected PIPT I-cache on CPU1
[    0.004660] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.005176] Detected PIPT I-cache on CPU2
[    0.005258] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.005739] Detected PIPT I-cache on CPU3
[    0.005823] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[... 708 lines cut ...]
```

### OpenWrt version

r24944+2018-906595c2627

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-07-26:
Could it be a USB issue?

schuettecarsten, 2024-07-27:
Not sure. The same image boots from a different SD card. Maybe the area on the SD card *after* the partition is broken. Unfortunately, it's quite hard to delete that area.

closed 2024-07-27 by schuettecarsten

schuettecarsten, 2024-07-28:
Yes, it is something with USB:

```
[    2.661910] usb 1-1.2: new full-speed USB device number 3 using xhci_hcd
[    2.762118] usb 1-1.2: device descriptor read/64, error -32
[    2.938932] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    2.982105] usb 1-1.2: device descriptor read/64, error -32
```

schuettecarsten, 2024-07-30:
I was able to fix this by completely erasing the SD card and re-flash.

brocaar, 2025-09-08:
Hi @schuettecarsten, I'm not sure if you have further looked into this, but I'm running into the same problem using a (self built) v24.10.2 image. I can not reproduce this 100% of the time. My assumption this has to do with the `/dev/loop0` device which uses the remaining free space of the rootfs partition to store the overlayfs and data the existence of a previous `/dev/loop0` device starting at a different offset (which would explain why you were able to "fix" this by completely erasing the SD card + re-flash).

In my case I am able to (not consistently, but regularly) by upgrading from an older self-built version to a self-built version based on v24.10.2. My assumption is the offset of the `/dev/loop0` might not be correct after the upgrade and thus OpenWrt fails to mount it, which means the overlayfs can't be mounted and settings are configuration is not available.

When I access the terminal (usb keyboard + hdmi display) and I "reset" OpenWrt by executing `firstboot -y && reboot`, the issue is resolved and in my case, all configuration is restored (I assume this is because `/dev/loop0` was not mounted when `firstboot` was executed?).

My assumption is that the above causes OpenWrt to reset the `/dev/loop0` offset but since it is not mounted, it does not delete its content. Or maybe because the start of `/dev/loop0` has changed after the upgrade, it is not in a clean state when mounted initially?

I need to further investigate this and I'm learning step-by-step how `sysupgrade` handles the backup and restore of configuration. Any feedback would be helpful.