openwrt/openwrt #16019: How to get WaveShare PCIe TO Gigabit ETH USB3.2 HAT+ to work in RPi5 snapshot?

mjmeans, 2024-07-28:
### Describe the bug

How to get WaveShare PCIe TO Gigabit ETH USB3.2 HAT+ to work in RPi5 snapshot?
It appears the 'dtparam=pciex1' entry on 'config.txt' either does not enable the pci bus, or openwrt doesn't know what to do with it. 'lspci' doesn't work either.

### OpenWrt version

latestr26997-a67b20e3d5

### OpenWrt release

r26997-a67b20e3d5

### OpenWrt target/subtarget

RPi5

### Device

RPi5

### Image kind

Official downloaded image

### Steps to reproduce

rpi5 with the pci ethernet hat. default squashfs snapshot to sdcard. boot. watch console. only eth0 shown. verify with ifconfig. edit /boot/config.txt to add the pci enabling command 'dtparam=pciex1'. reboot. watch console. still only eth0 shown. vverify ith ifconfig.

the hat was tested with raspberry pi os and the second ethernet port worked fine. so it's not the hat.

(P.S. not willing to create an account on the community support site. i have too many accounts with too many sites already.)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-07-29:
What is added to lspci and boot messages OpenWRT vs Pi system?

mjmeans, 2024-08-01:
I redownloaded the img yesterday using the configurator site, and added modules for luci and an RTL wifi dongle. I was able to get OpenWRT to recognize the additional eth1 in the HAT, but I have no idea why it is recognized since neither Luci or the RTL wifi chip driver has anything to do with the HAT. I'm supposing that it was just a bad snap the day I downloaded it. The HAT is working now so I will go ahead and close this. The WiFi in the pi isn't working right (enables for a short time only than disables itself). But that another issue probably not related to the HAT. I'll go ahead and close this.

closed 2024-08-01 by mjmeans