openwrt/openwrt #16022: Modem 5G/4G PCIe modem speed limitations

KFOzzx, 2024-07-28:
### Describe the bug

### Speed limitation on PCIe 5G modems :

The speeds are low on Banana BPI - R4 ...... 5G/4G modems

The system OpenWrt by default is assigning the main important roots to only Core 0 (CPU 0) even if you try
`echo f > /proc/irq/XX/smp_affinity`
or
`echo 0f > /proc/irq/XX/smp_affinity`
**will not solve the problem ,still will be assigned to Core 0 (CPU 0 )** 

[image]

[image]

[image]

-----------------------------------------------------
### Solution :
you need to do it manually from htop package or Taskset package(by using PID) to let system assign 4 Cores on network root to increase the network performance
:
root : napi/mtk_eth-5

### We Need to assign all loads between 4 Cores (CPU 0 , CPU 1 , CPU 2 and CPU 3)
=======================================================

**Video YouTube** 
https://www.youtube.com/watch?v=3L2u1EqVuwI
 
=======================================================
**Topic on OpenWRT**
https://forum.openwrt.org/t/openwrt-5g-4g-modems-m-2-speed-issue/200941/42

### OpenWrt version

r26996-73832e1a48

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

SuperKali, 2024-12-15:
I have the same issue on BPI R3 Mini

nicefile, 2025-02-09:
look this MIGTH be addressed in below upstream patch :
https://lore.kernel.org/all/20250204220622.156061-3-ahmed.zaki@intel.com/

giorez, 2025-07-06:
"you need to do it manually from htop package or Taskset package(by using PID) to let system assign 4 Cores on network root to increase the network performance"

1) how do I find the PID
2) how do I set also the second core (since I have just 2 cores) to the modem connection?
thanks

SuperKali, 2025-07-07:
@giorez I lost hope in it, looked for other alternatives