openwrt/openwrt #16032: Several SIGSEGV in ld-musl-aarch64.so.1 on bcm27xx

schuettecarsten, 2024-07-30:
### Describe the bug

I get SIGSEGV crashes on several executables, for example docker-compose (docker works!) or dnscrypt-proxy. The exception is always the same. Unfortunately, I do not get any further information except that both tools worked some weeks ago. Any idea how to debug this would be welcome.

```
Program received signal SIGSEGV, Segmentation fault.
0x0000007ff7fccb0c in ?? () from /lib/ld-musl-aarch64.so.1
```

### OpenWrt version

r26997+50-a67b20e3d59

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

schuettecarsten, 2024-07-30:
The invalid version is caused by merge of two pending PRs for dnsmasq and have no impact on this issue (tested!). Everything else is unchanged.

Rondom, 2024-07-30:
Can you use git bisect to narrow down when the issue started occurring, i.e. which commit is the culprit?

Or maybe a backtrace from gdb helps. Compile with CONFIG_DEBUG=y and then use ./scripts/remote-gdb to run against a gdbserver on the target reproducing the issue and then using bt full to get some backtrace.

schuettecarsten, 2024-07-30:
Still investigating, but as it it works with images from March 2024, it might be related to https://github.com/openwrt/openwrt/commit/87d7fa8885e0bb09ee088bdcc1060ac99699d39e

jow-, 2024-07-30:
Could you do a `bt full` to include local variable info and also print the value of `reloc_addr` using `p reloc_addr` ?

schuettecarsten, 2024-07-30:
```
(gdb) bt full
#0  0x0000007ff7fccb0c in do_relocs (dso=dso@entry=0x7ff7fff918 <app>, rel=0x5555551c08, rel_size=<optimized out>, stride=stride@entry=3) at ldso/dynlink.c:479
        base = <optimized out>
        syms = <optimized out>
        strings = <optimized out>
        sym = <optimized out>
        name = <optimized out>
        ctx = <optimized out>
        type = 1027
        sym_index = <optimized out>
        def = <optimized out>
        reloc_addr = 0x5555672358
        sym_val = <optimized out>
        tls_val = 0
        addend = <optimized out>
        skip_relative = 0
        reuse_addends = 0
        save_slot = 0
        tmp = <optimized out>
#1  0x0000007ff7fcd5c8 in reloc_all (p=p@entry=0x7ff7fff918 <app>) at ldso/dynlink.c:1421
        dyn = {1583366138, 15, 0, 11668792, 792, 4920, 2136, 7176, 1178976, 24, 1988, 24, 6785160, 6785144, 0, 0, 0, 0, 0, 0, 0, 549621594888, 0, 0, 0, 11668816, 11668784, 8, 8, 0, 12, 0, 0, 0, 0, 0, 0}
#2  0x0000007ff7fcf638 in __dls3 (sp=<optimized out>, auxv=<optimized out>) at ldso/dynlink.c:2029
        app = {base = 0x5555550000 "\177ELF\002\001\001", name = 0x7fffffffdf "/usr/sbin/dnscrypt-proxy", dynv = 0x5556070b40, next = 0x5556080c20, prev = 0x0, phdr = 0x5555550040, phnum = 12,
          phentsize = 56, syms = 0x5555550858, hashtab = 0x5555550318, ghashtab = 0x55555506c0, versym = 0x5555551afc, strings = 0x5555551338 "", syms_next = 0x5556080c20, lazy_next = 0x0, lazy = 0x0,
          lazy_cnt = 0, map = 0x5555550000 "\177ELF\002\001\001", map_len = 12582912, dev = 0, ino = 0, relocated = 0 '\000', constructed = 0 '\000', kernel_mapped = 1 '\001', mark = 0 '\000',
          bfs_built = 0 '\000', runtime_loaded = 0 '\000', deps = 0x7ff7ffb820, needed_by = 0x0, ndeps_direct = 2, next_dep = 2, ctor_visitor = 0x0, rpath_orig = 0x0, rpath = 0x0, tls = {
            next = 0x5556080d08, image = 0x5556070b40, len = 0, size = 16, align = 8, offset = 16}, tls_id = 1, relro_start = 11665408, relro_end = 11730944, new_dtv = 0x0, new_tls = 0x0, td_index = 0x0,
          fini_next = 0x0, shortname = 0x0, loadmap = 0x0, funcdescs = 0x0, got = 0x5556070d38, buf = 0x7ff7fffa88 <main_ctor_queue> "`\372\024VU"}
        vdso = {base = 0x7ff7ffa000 "\177ELF\002\001\001", name = 0x7ff7fe4730 "", dynv = 0x7ff7ffa560, next = 0x0, prev = 0x7ff7fffbb0 <ldso>, phdr = 0x7ff7ffa040, phnum = 3, phentsize = 56,
          syms = 0x7ff7ffa118, hashtab = 0x7ff7ffa0e8, ghashtab = 0x0, versym = 0x7ff7ffa220, strings = 0x7ff7ffa1a8 "", syms_next = 0x0, lazy_next = 0x0, lazy = 0x0, lazy_cnt = 0, map = 0x0, map_len = 0,
          dev = 0, ino = 0, relocated = 1 '\001', constructed = 0 '\000', kernel_mapped = 0 '\000', mark = 0 '\000', bfs_built = 0 '\000', runtime_loaded = 0 '\000', deps = 0x7ff7fe63d8 <no_deps>,
          needed_by = 0x0, ndeps_direct = 0, next_dep = 0, ctor_visitor = 0x0, rpath_orig = 0x0, rpath = 0x0, tls = {next = 0x0, image = 0x0, len = 0, size = 0, align = 0, offset = 0}, tls_id = 0,
          relro_start = 0, relro_end = 0, new_dtv = 0x0, new_tls = 0x0, td_index = 0x0, fini_next = 0x0, shortname = 0x7ff7fe4e78 "linux-gate.so.1", loadmap = 0x0, funcdescs = 0x0, got = 0x0,
          buf = 0x7ff7fff918 <app> ""}
        aux = {2256796664, 0, 0, 366503854144, 56, 12, 4096, 549620928512, 0, 366510639264, 0, 0, 0, 0, 0, 549755813512, 2183, 100, 0, 0, 0, 0, 0, 0, 0, 549755813496, 0, 0, 0, 0, 0, 549755813855}
        i = <optimized out>
        env_preload = <optimized out>
        replace_argv0 = <optimized out>
        vdso_base = <optimized out>
        argc = 1
        argv = 0x7ffffffc98
        argv_orig = <optimized out>
        envp = <optimized out>
        initial_tls = 0x7ff7fffd30 <builtin_tls>
        ptr = <optimized out>
#3  0x0000007ff7fceaac in __dls2 (base=<optimized out>, sp=0x7ffffffc90) at ldso/dynlink.c:1767
        auxv = 0x7ffffffd28
        ehdr = <optimized out>
        dyn = {16793584, 0, 0, 0, 400, 63448, 23896, 78584, 15576, 24, 15131, 24, 130104, 133336, 0 <repeats 23 times>}
        rel = <optimized out>
        rel_size = <optimized out>
        symbolic_rel_cnt = <optimized out>
        addends = <error reading variable addends (value requires 9431816 bytes, which is more than max-value-size)>
        dls2b_def = <optimized out>
        p1 = <optimized out>
        p2 = <optimized out>
        aux = <optimized out>
#4  0x0000000000000000 in ?? ()
```

```
(gdb) p reloc_addr
$1 = (size_t *) 0x5555672358
```

jow-, 2024-07-30:
Could you also provide your current `./scripts/diffconfig.sh` output? Did you tune any toolchain flags?

richfelker, 2024-07-30:
Segfault in `do_relocs` pretty much always means the file being processed (here the main `dnscrypt-proxy` binary) was compiled or linked wrong. It happens from trying to apply a relocation to an invalid or read-only address. If the program was linked a PIE, it's likely one of the object files linked (possibly from a static library) was compiled without `-fPIE`/`-fPIC`, or else there's a PIE-incompatible asm source file somewhere in the project.

schuettecarsten, 2024-07-31:
> Could you also provide your current `./scripts/diffconfig.sh` output? Did you tune any toolchain flags?

```
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BTRFS_PROGS_ZSTD=y
CONFIG_BUILDBOT=y
CONFIG_BUILD_PATENTED=y
CONFIG_BUSYBOX_CONFIG_BUNZIP2=y
CONFIG_BUSYBOX_CONFIG_BZIP2=y
CONFIG_BUSYBOX_CONFIG_BZIP2_SMALL=9
CONFIG_BUSYBOX_CONFIG_FEATURE_BZIP2_DECOMPRESS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_GZIP_LEVELS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SEAMLESS_BZ2=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SEAMLESS_LZMA=y
CONFIG_BUSYBOX_CONFIG_LZMA=y
CONFIG_BUSYBOX_CONFIG_UNLZMA=y
CONFIG_CCACHE=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_DEBUG=y
CONFIG_DOCKER_CGROUP_OPTIONS=y
CONFIG_DOCKER_NET_MACVLAN=y
CONFIG_DOCKER_NET_TFTP=y
CONFIG_DOCKER_OPTIONAL_FEATURES=y
CONFIG_DOCKER_STO_BTRFS=y
CONFIG_DOCKER_STO_EXT4=y
CONFIG_DROPBEAR_ECC=y
CONFIG_DROPBEAR_ECC_FULL=y
CONFIG_DROPBEAR_ECDSA_SK=y
CONFIG_DROPBEAR_ZLIB=y
CONFIG_GCC_DEFAULT_PIE=y
CONFIG_GCC_DEFAULT_SSP=y
CONFIG_GCC_USE_GRAPHITE=y
# CONFIG_GCC_USE_VERSION_13 is not set
CONFIG_GCC_USE_VERSION_14=y
CONFIG_GCC_VERSION="14.1.0"
CONFIG_GCC_VERSION_14=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_IMAGEOPT=y
CONFIG_INCLUDE_CONFIG=y
CONFIG_IPTABLES_CONNLABEL=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_ARM_PMU=y
CONFIG_KERNEL_ARM_PMUV3=y
CONFIG_KERNEL_BTRFS_FS_POSIX_ACL=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_KERNEL_CFQ_GROUP_IOSCHED=y
CONFIG_KERNEL_CGROUP_DEVICE=y
CONFIG_KERNEL_CGROUP_FREEZER=y
CONFIG_KERNEL_CGROUP_HUGETLB=y
CONFIG_KERNEL_CGROUP_NET_PRIO=y
CONFIG_KERNEL_CGROUP_PERF=y
CONFIG_KERNEL_EXT4_FS_POSIX_ACL=y
CONFIG_KERNEL_EXT4_FS_SECURITY=y
CONFIG_KERNEL_FS_POSIX_ACL=y
[... 1246 lines cut ...]
# CONFIG_PACKAGE_rtl8192cu-firmware is not set
# CONFIG_PACKAGE_rtl8192de-firmware is not set
# CONFIG_PACKAGE_rtl8192se-firmware is not set
# CONFIG_PACKAGE_rtl8723be-firmware is not set
# CONFIG_PACKAGE_rtl8723de-firmware is not set
# CONFIG_PACKAGE_rtl8821ae-firmware is not set
# CONFIG_PACKAGE_rtl8821ce-firmware is not set
# CONFIG_PACKAGE_rtl8822be-firmware is not set
# CONFIG_PACKAGE_rtl8822ce-firmware is not set
# CONFIG_PACKAGE_snapshot-tool is not set
# CONFIG_PACKAGE_uboot-envtools is not set
# CONFIG_PACKAGE_umbim is not set
# CONFIG_PACKAGE_uqmi is not set
# CONFIG_PACKAGE_wil6210-firmware is not set
# CONFIG_PACKAGE_wl12xx-firmware is not set
# CONFIG_PACKAGE_wl18xx-firmware is not set
# CONFIG_PACKAGE_wwan is not set
# CONFIG_PACKAGE_xdp-filter is not set
# CONFIG_PACKAGE_xdp-loader is not set
# CONFIG_PACKAGE_xdpdump is not set
# CONFIG_SDK is not set
# CONFIG_SDK_LLVM_BPF is not set
# CONFIG_VERSION_CODE_FILENAMES is not set
# CONFIG_VERSION_FILENAMES is not set
```

jow-, 2024-07-31:
Please retry with a completely clean build after disabling `CONFIG_GCC_DEFAULT_PIE`

schuettecarsten, 2024-07-31:
Building, will report when it's ready, might take some hours

-CONFIG_GCC_DEFAULT_PIE=y
+# CONFIG_GCC_DEFAULT_PIE is not set
```

schuettecarsten, 2024-07-31:
> Please retry with a completely clean build after disabling `CONFIG_GCC_DEFAULT_PIE`

Fixed. Thank you.
Is that a new option? I cannot remember to activate it... 🤔

richfelker, 2024-07-31:
That should NOT be taken as the fix, just a (rather time consuming) way to confirm the existing diagnosis. It may be a viable workaround, but it reduces the overall hardening of your system.

Someone just needs to look at the `readelf -a` for the broken binary and track down where the invalid relocation(s) came from, and fix that. Posting that output as an attachment would help.

schuettecarsten, 2024-08-01:
> Someone just needs to look at the `readelf -a` for the broken binary and track down where the invalid relocation(s) came from, and fix that. Posting that output as an attachment would help.

Please find the results (and the two binaries) in the attached files.

[broken.txt](https://github.com/user-attachments/files/16455616/broken.txt)
[working.txt](https://github.com/user-attachments/files/16455617/working.txt)
[dnscrypt-proxy.zip](https://github.com/user-attachments/files/16455620/dnscrypt-proxy.zip)

richfelker, 2024-08-01:
These binaries are stripped so it's virtually impossible to learn anything from them.

richfelker, 2024-08-01:
What I did notice is that there are a HUGE number of invalid relocations, all over rodata and some special sections I'm not familiar with. So it's not just one rogue asm file, and probably not even one static library built wrong.

I did notice that this program is written in golang, so it's possible that the golang tooling is setup wrong and not aware that it's supposed to be producing PIE output by default. This would explain the breakage.

Digging further, it seems this came up back in 2018 for the same project:

https://github.com/DNSCrypt/dnscrypt-proxy/issues/404

It looks like they may have "fixed" it by hard-coding "Alpine needs PIE" rather than properly detecting it or passing `-no-pie` to the linker. :facepalm:

schuettecarsten, 2024-08-01:
I will build unstripped binaries on the weekend, docker-compose and ctop are also affected.

richfelker, 2024-08-01:
Also golang stuff, no? Sounds like the problem is that it's not configured right for working with a default-PIE toolchain. OpenWRT needs to either patch it so `go` passes `-no-pie` to the linker or to force `-buildmode=pie` in place of whatever the non-PIE cgo buildmode that programs are using is.

richfelker, 2024-08-02:
I asked Alpine folks how they handle this, and their global package build config exports `GOFLAGS="-buildmode=pie"` in the environment. I don't think this is the fully "right" fix, as it doesn't make it safe to manually invoke go, but it will fix package builds. It should work for OpenWRT's purposes.

schuettecarsten, 2024-08-03:
> I asked Alpine folks how they handle this, and their global package build config exports `GOFLAGS="-buildmode=pie"` in the environment. I don't think this is the fully "right" fix, as it doesn't [...]

There are already options for Go PIE support but they look outdated:
https://github.com/search?q=repo%3Aopenwrt%2Fpackages+GO_PIE_SUPPORTED_OS_ARCH&type=code