openwrt/openwrt #16039: Apple devices can not connet via Wi-Fi.

Gibitwit, 2024-07-31:
### Describe the bug

Apple devices can not connect or they initially connect after the wireless has been configured but then loose connectivey soon there after and cannot connect again.

### OpenWrt version

All version 23 releases.

### OpenWrt release

All version 23.

### OpenWrt target/subtarget

ramips/mt7621

### Device

Zbtlink ZBT-WE3526

### Image kind

Official downloaded image

### Steps to reproduce

After flashing for the Luci interface.
Login.
Set administrator password.
Set SSH access to LAN.
Set the network name (SSID) for both 2.4 Ghz and 5Ghz to the same name.  Or set it for only the 2.4 Ghz band.
Enable both radios or only the 2.4 Ghz radio.
Attempt to connect with Apple devices.

### Actual behaviour

Apple devices either will not connect or will initially connect but soon loose connectivity and cannot reconnect.

### Expected behaviour

Apple devices connect and stay connected, and if need to reconnect can reconnect.

### Additional info

Apple devices that won't connect:
iMac Pro running MacOS 10.14 Mojave
iPhone 6
iPhone 6 Plus
iPad Air 2 I am not sure which model, but it is now broken.

The router is an IQ Router flashed to OpenWRT.

I tried various settings, wifi encrytion, no encryption.  Turned off Disassociate On Low Acknowledgement.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-07-31:
What is your wireless configuration?
Try setting the encryption to WPA2-PSK.

Gibitwit, 2024-08-01:
I have reverted to 22.03.7.  On 23.05.4 I tried WPA3 mixed mode, WPA2. and no encryption.  802.11w Management Frame Protection was always left on the default of optional.

Should I test something in 23.05.4?

brada4, 2024-08-01:
You have to add version tags to figure out which of 4 iqrouters you got.
Increase hostapgd log level, repeat failure and try to systematize characteristic log message(s)
https://openwrt.org/docs/guide-developer/debugging#logging_hostapd_behaviour
level 1, also setting log buffer larger like 1024

schuettecarsten, 2024-08-01:
Do you have "802.11r Fast Transition", "802.11k RRM" or "802.11v BSS transition" enabled?
You can find the options on the WLAN roaming tab in wireless settings.

Gibitwit, 2024-08-02:
It is an IQ Router V3.

I will need some time to switch to 23.05.4 to get the log if you need it.

> Do you have "802.11r Fast Transition", "802.11k RRM" or "802.11v BSS transition" enabled? You can find the options on the WLAN roaming tab in wireless settings.

I left that on default which I think is off.  Most settings were left untouched.

danpawlik, 2024-08-02:
@Gibitwit could you just share `cat /etc/config/wireless` please?

brada4, 2024-08-02:
Get serious.
Which image your router is installed with?
https://evenroute.com/eol

Gibitwit, 2024-08-03:
> @Gibitwit could you just share `cat /etc/config/wireless` please?

This is from 23.05.4.
`
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option cell_density '0'
	option country 'US'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'Skynet'
	option encryption 'sae-mixed'
	option disassoc_low_ack '0'
	option key ''
	option ieee80211w '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
	option channel '36'
	option band '5g'
	option htmode 'VHT80'
	option country 'US'
	option cell_density '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'Skynet'
	option encryption 'sae-mixed'
	option disassoc_low_ack '0'
	option key ''
	option ieee80211w '0'

`
Did you want it from 22.03.7?

This is the currently install image, which I just installed.
https://downloads.openwrt.org/releases/23.05.4/targets/ramips/mt7621/openwrt-23.05.4-ramips-mt7621-zbtlink_zbt-we3526-squashfs-sysupgrade.bin

This what we have been using.
https://downloads.openwrt.org/releases/22.03.7/targets/ramips/mt7621/openwrt-22.03.7-ramips-mt7621-zbtlink_zbt-we3526-squashfs-sysupgrade.bin

Gibitwit, 2024-08-03:
Here is the system log.
`
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using nameserver 
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sat Aug  3 08:24:14 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Aug  3 08:24:16 2024 daemon.warn odhcpd[1556]: No default route present, overriding ra_lifetime!
`
Kernel log.
`
[    0.000000] Linux version 5.15.162 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24012-d8dd03c46f) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Jul 15 22:14:18 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Zbtlink ZBT-WE3526
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 12 pages/cpu s17808 r8192 d23152 u49152
[    0.000000] pcpu-alloc: s17808 r8192 d23152 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113536
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Writing ErrCtl register=00040000
[    0.000000] Readback ErrCtl register=00040000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 443156K/458752K available (7317K kernel code, 628K rwdata, 884K rodata, 1272K init, 225K bss, 15596K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008059] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066228] pid_max: default: 32768 minimum: 301
[    0.071664] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... 253 lines cut ...]
[   21.372911] device lan4 entered promiscuous mode
[... cut ...]
[   25.238419] br-lan: port 5(phy1-ap0) entered blocking state
[   25.244087] br-lan: port 5(phy1-ap0) entered forwarding state
[   25.566029] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   25.574081] br-lan: port 1(lan1) entered blocking state
[   25.579299] br-lan: port 1(lan1) entered forwarding state
[   34.913728] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control off
[   34.922141] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
`

brada4, 2024-08-03:
SAE does not work with PMF explicitly disabled.

Gibitwit, 2024-08-03:
What is PMF?  Where is the checkbox?  What namiltd said?  I thought they said to disable it, I see they said enable.  I will try setting it to enabled.

Gibitwit, 2024-08-03:
When I inupt this ps | grep hostapd.  It returned.
1278 root      2912 S    {hostapd} /sbin/ujail -t 5 -n hostapd -U network -G network -C /etc/capabilities/w
 1281 network   4496 S    /usr/sbin/hostapd -s -g /var/run/hostapd/global
 4177 root      1376 S    grep hostapd

The rest of the commands worked as in the wiki.

brada4, 2024-08-03:
Do the very minimal setup required by "apple devices".
https://support.apple.com/en-us/102766

Gibitwit, 2024-08-03:
I only have 20, 40, or 80 hz for Channel width.  2.4 Ghz on 20 hz.  5 Ghz on 80 hz.

I am not sure what to do about NAT.  Should in enable maquerading on LAN -> WAN?

brada4, 2024-08-03:
You have wifi connection problems, first paragraph describes wifi apple is willing to connect to.

Gibitwit, 2024-08-03:
I do not know what I am supposed to do.  All the defaults align with Apples instructions, except channel width and DHCP lease time.  As I understand NAT/masquerading should not be used on LAN, but Apple's instructions do not specify.  We did not use masquerading on LAN with the version 22 releases.

danpawlik, 2024-08-03:
@Gibitwit try to change: 
```
option ieee80211w '1'
```

if you want to use WPA3, it is mandatory to use frame protection. If you use mixed, try to use 'optional' or change just to WPA2.

If that will not help, try:
```
config wifi-iface 'default_radio0'
       option device 'radio0'
       option network 'lan'
       option mode 'ap'
       option ssid 'Skynet'
       option encryption 'sae-mixed'
       option disassoc_low_ack '0'
       option key ''
       option ieee80211w '1'
        option wnm_sleep_mode '1'
        option bss_transition '1'

# (...)

config wifi-iface 'default_radio1'
       option device 'radio1'
       option network 'lan'
       option mode 'ap'
       option ssid 'Skynet'
       option encryption 'sae-mixed'
       option disassoc_low_ack '0'
       option key ''
       option ieee80211w '1'
        option wnm_sleep_mode '1'
        option bss_transition '1'
```

Gibitwit, 2024-08-05:
All three options for the frame protection have been tried.  Keep in mind the steps I listed to reproduce the bug, almost everything is on default.  Encryption mode does not matter.

This is the wireless config from version 22.03.7 for comparison.
config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option country 'US'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Skynet'
        option dtim_period '3'
        option disassoc_low_ack '0'
        option encryption 'sae-mixed'
        option key ''

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option country 'US'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Skynet'
        option dtim_period '3'
        option disassoc_low_ack '0'
        option encryption 'sae-mixed'
        option key ''

Gibitwit, 2024-08-05:
> @Gibitwit try to change:
> 
> ```
> option ieee80211w '1'
> ```
> 
> if you want to use WPA3, it is mandatory to use frame protection. If you use mixed, try to use 'optional' or change just to WPA2.
> 
> If that will not help, try:
> 
> ```
> # change mode to 'legacy', just for test, config should look like:
> 
> config wifi-device 'radio0'
>         option type 'mac80211'
>         option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
>         option channel '1'
>         option band '2g'
>         option cell_density '0'
>         option country 'US'
> 
> config wifi-iface 'default_radio0'
>         option device 'radio0'
>         option network 'lan'
>         option mode 'ap'
>         option ssid 'Skynet'
>         option encryption 'sae-mixed'
>         option disassoc_low_ack '0'
>         option key 'YOURPASS'
>         option ieee80211w '1'
>         option wnm_sleep_mode '1'
>         option bss_transition '1'
> 
> # (...)
> 
> config wifi-iface 'default_radio1'
>         option device 'radio1'
>         option network 'lan'
>         option mode 'ap'
>         option ssid 'Skynet'
>         option encryption 'sae-mixed'
>         option disassoc_low_ack '0'
>         option key 'YOURPASS'
>         option ieee80211w '1'
>         option wnm_sleep_mode '1'
>         option bss_transition '1'
> ```

This did not work.  Wireless is now disabled.  I clicked the buttons to disable the wifi, so that they would change to "Enable".  Then clicked the enable buttons.  On the left it still says disabled and the iMac Pro does not see the network.

I saw this in the system log.
Mon Aug  5 02:49:24 2024 daemon.notice netifd: wan (3184): udhcpc: started, v1.36.1
Mon Aug  5 02:49:24 2024 daemon.err odhcp6c[3185]: Failed to send RS (Address not available)
Mon Aug  5 02:49:24 2024 daemon.notice netifd: wan (3184): udhcpc: broadcasting discover
Mon Aug  5 02:49:24 2024 daemon.err odhcp6c[3185]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Mon Aug  5 02:49:25 2024 daemon.err odhcp6c[3185]: Failed to send SOLICIT message to ff02::1:2 (Address not available)

And this frequently appears in the system log.  What does it mean?
daemon.warn odhcpd[1557]: No default route present, overriding ra_lifetime!

danpawlik, 2024-08-05:
@Gibitwit errors that you paste are related to ipv6 and odhcp. 
There is no information related to the `hostapd`?
Please increase log output for the wireless: 
```
# wifi debug - https://openwrt.org/docs/guide-developer/debugging
uci set wireless.default_radio0.log_level=1
uci set wireless.default_radio1.log_level=1
uci commit wireless
```

Then reboot router (to be sure) and paste:
```
logread | grep -A5 -B5 hostapd
```

PS. Please use backtick "```"  when you provide a code (https://www.markdownguide.org/extended-syntax/#fenced-code-blocks)

Spudz76, 2024-08-05:
This was never specified but and might be assumed from context, but do you have any non-Apple devices and do any of those work?  If we don't know it works at all, the whole Apple thing might be pointless detail.

As far as I can tell from the Apple doc linked it should still connect and work but with a warning almost regardless of settings (aside from the WPA3 without MFP definitely can't work).  It doesn't say it would ignore the existence or not connect to stuff that's weak.

Gibitwit, 2024-08-07:
@danpawlik  Ok I wil try that next time.

@Spudz76 So far it has been only the Apple devices that have had trouble.  We first noticed it on an iMac Pro, I then checked several other systems including a Linux PC.  I will see what I can get together.

DragonBluep, 2024-08-07:
AFAIK, on version 23.x, this device doesn't work correctly at all. More details: https://github.com/openwrt/openwrt/issues/16000

Gibitwit, 2024-08-10:
> ```
> # wifi debug - https://openwrt.org/docs/guide-developer/debugging
> uci set wireless.default_radio0.log_level=1
> uci set wireless.default_radio1.log_level=1
> uci commit wireless
> ```
> 
> Then reboot router (to be sure) and paste:
> 
> ```
> logread | grep -A5 -B5 hostapd
> ```
Heres the log, it appear to still be level 2.
```
Sat Aug 10 21:58:46 2024 daemon.notice netifd: radio1 (1969): WARNING: Variable 'data' does not exist or is not an array/object
Sat Aug 10 21:58:46 2024 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sat Aug 10 21:58:47 2024 daemon.info procd: - init complete -
Sat Aug 10 21:58:47 2024 authpriv.warn dropbear[2221]: Failed listening on '22': Error listening: Address not available
Sat Aug 10 21:58:47 2024 authpriv.info dropbear[2221]: Not backgrounding
Sat Aug 10 21:58:47 2024 daemon.notice hostapd: Set new config for phy phy1:
Sat Aug 10 21:58:47 2024 daemon.notice wpa_supplicant[1284]: Set new config for phy phy1
Sat Aug 10 21:58:47 2024 daemon.notice hostapd: Set new config for phy phy0:
Sat Aug 10 21:58:47 2024 daemon.notice wpa_supplicant[1284]: Set new config for phy phy0
Sat Aug 10 21:58:48 2024 daemon.notice wpa_supplicant[1284]: Set new config for phy phy1
Sat Aug 10 21:58:48 2024 daemon.notice wpa_supplicant[1284]: Set new config for phy phy0
Sat Aug 10 21:58:48 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Sat Aug 10 21:58:48 2024 daemon.notice hostapd: Restart interface for phy phy0
Sat Aug 10 21:58:49 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 hw_mode=g supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 chanlist=1 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] channel=1  interface=phy0-ap0 bssid=de:3f:8e:b6:bc:68 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=0 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 sae_require_mfp=1 sae_pwe=2 wpa_passphrase=apple1976 wpa_psk_file=/var/run/hostapd-phy0-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=Skynet bridge=br-lan wds_bridge= snoop_iface=br-lan wpa_disable_eapol_key_retries=0 wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256 SAE okc=1 ieee80211w=1 group_mgmt_cipher=AES-128-CMAC dynamic_vlan=0 vlan_naming=1 vlan_
Sat Aug 10 21:58:49 2024 kern.info kernel: [   23.913153] br-lan: port 5(phy0-ap0) entered blocking state
Sat Aug 10 21:58:49 2024 kern.info kernel: [   23.918934] br-lan: port 5(phy0-ap0) entered disabled state
Sat Aug 10 21:58:49 2024 kern.info kernel: [   23.925596] device phy0-ap0 entered promiscuous mode
Sat Aug 10 21:58:49 2024 daemon.warn hostapd: nl80211: Failed to add interface phy0-ap0 into bridge br-lan: Resource busy
Sat Aug 10 21:58:49 2024 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Aug 10 21:58:49 2024 kern.info kernel: [   24.029300] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
Sat Aug 10 21:58:49 2024 kern.info kernel: [   24.036475] br-lan: port 5(phy0-ap0) entered blocking state
Sat Aug 10 21:58:49 2024 kern.info kernel: [   24.042093] br-lan: port 5(phy0-ap0) entered forwarding state
Sat Aug 10 21:58:49 2024 daemon.notice netifd: Network device 'phy0-ap0' link is up
Sat Aug 10 21:58:49 2024 kern.info kernel: [   24.049141] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Sat Aug 10 21:58:49 2024 daemon.notice netifd: bridge 'br-lan' link is up
Sat Aug 10 21:58:49 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Sat Aug 10 21:58:49 2024 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->ENABLED
Sat Aug 10 21:58:49 2024 daemon.notice hostapd: phy0-ap0: AP-ENABLED
Sat Aug 10 21:58:49 2024 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Sat Aug 10 21:58:49 2024 daemon.notice hostapd: Restart interface for phy phy1
Sat Aug 10 21:58:49 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-A-MPDU-LEN-EXP3] channel=36  interface=phy1-ap0 bssid=d8:99:d0:dc:ce:f5 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=0 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 sae_require_mfp=1 sae_pwe=2 wpa_passphrase=apple1976 wpa_psk_file=/var/run/hostapd-phy1-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=Skynet bridge=br-lan wds_bridge= snoop_if
Sat Aug 10 21:58:49 2024 kern.info kernel: [   24.570408] br-lan: port 6(phy1-ap0) entered blocking state
Sat Aug 10 21:58:49 2024 kern.info kernel: [   24.576050] br-lan: port 6(phy1-ap0) entered disabled state
Sat Aug 10 21:58:49 2024 kern.info kernel: [   24.582081] device phy1-ap0 entered promiscuous mode
[... cut ...]
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option country 'US'
        option cell_density '0'
        option log_level '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Skynet'
        option encryption 'sae-mixed'
        option disassoc_low_ack '0'
        option key ''
        option log_level '1'

```

danpawlik, 2024-08-11:
@Gibitwit so you have an error 
```
:33:1a:0f:54:35 IEEE 802.11: authenticated
Sat Aug 10 17:36:10 2024 daemon.debug hostapd: phy0-ap0: STA 40:33:1a:0f:54:35 IEEE 802.11: association SA Query attempt 1
Sat Aug 10 17:36:10 2024 daemon.debug hostapd: phy0-ap0: STA 40:33:1a:0f:54:35 IEEE 802.11: association SA Query attempt 2
Sat Aug 10 17:36:10 2024 daemon.debug hostapd: phy0-ap0: STA 40:33:1a:0f:54:35 IEEE 802.11: association SA Query attempt 3
Sat Aug 10 17:36:10 2024 daemon.debug hostapd: phy0-ap0: STA 40:33:1a:0f:54:35 IEEE 802.11: association SA Query attempt 4
Sat Aug 10 17:36:10 2024 daemon.debug hostapd: phy0-ap0: STA 40:33:1a:0f:54:35 IEEE 802.11: association SA Query attempt 5
Sat Aug 10 17:36:11 2024 daemon.debug hostapd: phy0-ap0: STA 40:33:1a:0f:54:35 IEEE 802.11: association SA Query timed out
```
And I already mentioned you, so I will repeat: you need to set 802.11w as optional, and your configuration does not have it.

Gibitwit, 2024-08-12:
All 802.11w settings were at the default values.  Managment frame protection default is optional.  Those options disappear when encrytion is not used.

danpawlik, 2024-08-12:
@Gibitwit worth to read: https://github.com/openwrt/openwrt/issues/7858 and https://forum.openwrt.org/t/documentation-on-802-11r/176972/43