openwrt/openwrt #16054: Two Belkin f9k1109 radio issues

ascendbeing, 2024-08-02:
### Describe the bug

1. On Snapshot (but not 23.05) the first radio, which is 2.4, is nonfunctional. See attached pictures. It "makes" the AP but nothing can see it and it can't be seen by other devices. On 23.05 no such issue occurs.
2. On both 23.05 as well as snapshot, the second radio reports as 2.4 or 5. But if you try to use it as 2.4, it won't be seen and channel analysis won't see any 2.4 APs either. Please see attached pictures.
3. I think the second radio is supposed to be 5ghz only. The first is 2.4 for sure because it works on 23.05. On neither have I got the second radio to do anything with 2.4.

I am using the stock settings out of the box. I clicked enable on the first radio to generate some of the details.

### OpenWrt version

r27061-ee8c05f9b6

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/rt3883

### Device

Belkin F9K1109 Version 1.0

### Image kind

Official downloaded image

### Additional info

root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:01.0/0000:01:00.0'
        option band '2g'
        option channel '1'
        option htmode 'HT20'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/10180000.wmac'
        option band '5g'
        option channel '36'
        option htmode 'HT40'
        option disabled '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

[image]
[image]
[image]
[image]
[image]
[image]

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ascendbeing, 2024-08-02:
I wish I knew how to restore this to stock or to padavan because I have two of these (1 is a f9k1103, same thing basically). In my limited experience padavan was running purringly, the CPU on these is probably choking because I only get like 10-15Mb/s wifi with software offloading enabled. If anyone can help me figure out how to do that, I would really appreciated that as a sidenote. I have a UBoot flasher recovery I can use but I tried flashing padavan via it and it kept openwrt.

Heres dmesg

[Tue Jul 30 19:45:27 2024] Linux version 6.6.43 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r27052-4e6212e62f) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Tue Jul 30 19:42:33 2024
[Tue Jul 30 19:45:27 2024] SoC Type: Ralink RT3883 ver:1 eco:5
[Tue Jul 30 19:45:27 2024] printk: bootconsole [early0] enabled
[Tue Jul 30 19:45:27 2024] CPU0 revision is: 0001974c (MIPS 74Kc)
[Tue Jul 30 19:45:27 2024] MIPS: machine is Belkin F9K1109 Version 1.0
[Tue Jul 30 19:45:27 2024] Initrd not found or empty - disabling initrd
[Tue Jul 30 19:45:27 2024] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[Tue Jul 30 19:45:27 2024] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[Tue Jul 30 19:45:27 2024] Zone ranges:
[Tue Jul 30 19:45:27 2024]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[Tue Jul 30 19:45:27 2024] Movable zone start for each node
[Tue Jul 30 19:45:27 2024] Early memory node ranges
[Tue Jul 30 19:45:27 2024]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[Tue Jul 30 19:45:27 2024] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[Tue Jul 30 19:45:27 2024] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[Tue Jul 30 19:45:27 2024] pcpu-alloc: [0] 0 
[Tue Jul 30 19:45:27 2024] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[Tue Jul 30 19:45:27 2024] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[Tue Jul 30 19:45:27 2024] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[Tue Jul 30 19:45:27 2024] Writing ErrCtl register=00000000
[Tue Jul 30 19:45:27 2024] Readback ErrCtl register=00000000
[Tue Jul 30 19:45:27 2024] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[Tue Jul 30 19:45:27 2024] mem auto-init: stack:off, heap alloc:off, heap free:off
[Tue Jul 30 19:45:27 2024] Memory: 56308K/65536K available (5000K kernel code, 584K rwdata, 1548K rodata, 1244K init, 201K bss, 9228K reserved, 0K cma-reserved)
[Tue Jul 30 19:45:27 2024] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[Tue Jul 30 19:45:27 2024] NR_IRQS: 256
[Tue Jul 30 19:45:27 2024] ------------[ cut here ]------------
[Tue Jul 30 19:45:27 2024] WARNING: CPU: 0 PID: 0 at drivers/clk/ralink/clk-mtmips.c:499 rt3883_bus_recalc_rate+0x98/0x138
[Tue Jul 30 19:45:27 2024] Modules linked in:
[Tue Jul 30 19:45:27 2024] CPU: 0 PID: 0 Comm: swapper Not tainted 6.6.43 #0
[Tue Jul 30 19:45:27 2024] Stack : 805e58d0 00000000 00000004 8004f950 00000000 00000004 00000000 00000000
[Tue Jul 30 19:45:27 2024]         80669c54 80830000 80700000 805ae570 80670068 00000001 80669bf8 00000000
[Tue Jul 30 19:45:27 2024]         00000000 00000000 805ae570 80669b38 00000020 804db7dc 00000000 00000000
[Tue Jul 30 19:45:27 2024]         203a6d6d 80669b78 80669e48 70617773 00000000 805ae570 00000000 00000009
[Tue Jul 30 19:45:27 2024]         00000000 00000001 00000004 00000001 00000000 00000000 83fe43b0 00000000
[Tue Jul 30 19:45:27 2024]         ...
[Tue Jul 30 19:45:27 2024] Call Trace:
[Tue Jul 30 19:45:27 2024] [<800065d0>] show_stack+0x64/0xf4
[Tue Jul 30 19:45:27 2024] [<804bca14>] dump_stack_lvl+0x38/0x60
[Tue Jul 30 19:45:27 2024] [<800218ac>] __warn+0x94/0xe4
[Tue Jul 30 19:45:27 2024] [<8002195c>] warn_slowpath_fmt+0x60/0x94
[Tue Jul 30 19:45:27 2024] [<80259ff8>] rt3883_bus_recalc_rate+0x98/0x138
[Tue Jul 30 19:45:27 2024] [<80254530>] __clk_register+0x568/0x688
[Tue Jul 30 19:45:27 2024] [<80254838>] of_clk_hw_register+0x18/0x2c
[Tue Jul 30 19:45:27 2024] [<8070b910>] rt2880_clk_of_clk_init_driver+0x18c/0x594
[Tue Jul 30 19:45:27 2024] [<8070b628>] of_clk_init+0x1c0/0x23c
[Tue Jul 30 19:45:27 2024] [<806fc448>] plat_time_init+0x58/0x18c
[Tue Jul 30 19:45:27 2024] [<806fdaf0>] time_init+0x10/0x6c
[Tue Jul 30 19:45:27 2024] [<806f9bc4>] start_kernel+0x458/0x67c

[Tue Jul 30 19:45:27 2024] ---[ end trace 0000000000000000 ]---
[Tue Jul 30 19:45:27 2024] CPU Clock: 500MHz
[Tue Jul 30 19:45:27 2024] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041786 ns
[Tue Jul 30 19:45:27 2024] sched_clock: 32 bits at 250MHz, resolution 4ns, wraps every 8589934590ns
[Tue Jul 30 19:45:27 2024] Calibrating delay loop... 249.44 BogoMIPS (lpj=1247232)
[Tue Jul 30 19:45:27 2024] pid_max: default: 32768 minimum: 301
[Tue Jul 30 19:45:27 2024] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[Tue Jul 30 19:45:27 2024] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[Tue Jul 30 19:45:27 2024] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1.
[Tue Jul 30 19:45:27 2024] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[Tue Jul 30 19:45:27 2024] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[Tue Jul 30 19:45:27 2024] pinctrl core: initialized pinctrl subsystem
[Tue Jul 30 19:45:27 2024] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[... cut ...]
[Tue Jul 30 19:47:13 2024] rt2800pci 0000:01:00.0 phy0-ap0: entered allmulticast mode
[Tue Jul 30 19:47:13 2024] rt2800pci 0000:01:00.0 phy0-ap0: entered promiscuous mode
[Tue Jul 30 19:47:13 2024] br-lan: port 2(phy0-ap0) entered blocking state
[Tue Jul 30 19:47:13 2024] br-lan: port 2(phy0-ap0) entered forwarding state
[Tue Jul 30 19:47:14 2024] br-lan: port 2(phy0-ap0) entered disabled state
[Tue Jul 30 19:47:14 2024] br-lan: port 2(phy0-ap0) entered blocking state
[Tue Jul 30 19:47:14 2024] br-lan: port 2(phy0-ap0) entered forwarding state

DragonBluep, 2024-08-03:
@paraka It seems that there are some clock issues about clock probe sequence.

paraka, 2024-08-04:
It looks like you got the warning because parent rate for the cpu clock (xtal) is zero...Even that the CPU clock of 500 MHz seems correct so the init code in 'arch/mips/ralink/clk.c' seems ok for this soc. xtal for this soc is a fixed clock which should be 40 MHz (drivers/clk/ralink/clk-mtmips.c#L266). I have reviewed the code and it is not obvious why this is happening and I don't have the hardware to test in my side. I also do believe the dts files are correct. Can we please get some debug to try to figure out what might be happening there? Please check '/sys/kernel/debug/clk' related clocks frequency and parent to see which one is wrong. I guess the radios for this are comming from wmac so "xtal" should be the parent with 40 MHz?
So check as follows adding the output for the clock of the radios:
```
root@gnubee:/sys/kernel/debug/clk# cat /sys/kernel/debug/clk/xtal/clk_rate                                                                                                                                                                                                     
40000000
root@gnubee:/sys/kernel/debug/clk# cat /sys/kernel/debug/clk/cpu/clk_rate 
900000000
root@gnubee:/sys/kernel/debug/clk# cat /sys/kernel/debug/clk/cpu/clk_parent 
xtal
```

DragonBluep, 2024-08-04:
We are trying to register cpu and bus clocks before xtal clock. However, xtal is the parent clock of the bus clock. I think that's why it failed. CPU clock is not calculated from xtal so it's still correct. I guess rt305x series have the similar issue.

paraka, 2024-08-04:
'bus' clock parent is the 'cpu' clock so it should not be failing because of that (drivers/clk/ralink/clk-mtmips.c#L694) ... Am I missing something? Anyway it is worth to get all the rates for all the clocks in '/sys/kernel/debug/clk/'

ascendbeing, 2024-08-04:
Hey thanks for showing interest in this @paraka . Please feel free to let me know if it would help to pull any real data off a live system running this hardware. I have two of these (openwrt only officially supports f9k1109, but I also have a f9k1103, which is functionally, if not completely, identical; runs off the f9k1109 build, presenting the same problems).

paraka, 2024-08-05:
@ascendbeing Can you please share your SoC's clock plan debug in /sys/kernel/debug/clk/? We need the clock, its parent and the rate. So the following command should give what I need:
```
root@gnubee:~# for clk in `ls /sys/kernel/debug/clk/`; do echo "Clock: $clk -> rate: `cat /sys/kernel/debug/clk/$clk/clk_rate` parent `cat /sys/kernel/debug/clk/$clk/clk_parent`"; done
Clock: 125m -> rate: 125000000 parent xtal
Clock: 150m -> rate: 150000000 parent xtal
Clock: 250m -> rate: 250000000 parent xtal
Clock: 270m -> rate: 270000000 parent xtal
Clock: 50m -> rate: 50000000 parent xtal
Clock: bus -> rate: 225000000 parent cpu
cat: /sys/kernel/debug/clk/clk_dump/clk_rate: Not a directory
cat: /sys/kernel/debug/clk/clk_dump/clk_parent: Not a directory
Clock: clk_dump -> rate:  parent 
cat: /sys/kernel/debug/clk/clk_orphan_dump/clk_rate: Not a directory
cat: /sys/kernel/debug/clk/clk_orphan_dump/clk_parent: Not a directory
Clock: clk_orphan_dump -> rate:  parent 
cat: /sys/kernel/debug/clk/clk_orphan_summary/clk_rate: Not a directory
cat: /sys/kernel/debug/clk/clk_orphan_summary/clk_parent: Not a directory
Clock: clk_orphan_summary -> rate:  parent 
cat: /sys/kernel/debug/clk/clk_summary/clk_rate: Not a directory
cat: /sys/kernel/debug/clk/clk_summary/clk_parent: Not a directory
Clock: clk_summary -> rate:  parent 
Clock: cpu -> rate: 900000000 parent xtal
Clock: crypto -> rate: 250000000 parent 250m
Clock: eth -> rate: 50000000 parent 50m
Clock: fe -> rate: 250000000 parent 250m
Clock: gdma -> rate: 225000000 parent bus
Clock: hsdma -> rate: 150000000 parent 150m
Clock: i2c -> rate: 50000000 parent 50m
Clock: i2s -> rate: 270000000 parent 270m
Clock: nand -> rate: 125000000 parent 125m
Clock: pcie0 -> rate: 125000000 parent 125m
Clock: pcie1 -> rate: 125000000 parent 125m
Clock: pcie2 -> rate: 125000000 parent 125m
Clock: pcm -> rate: 270000000 parent 270m
Clock: pio -> rate: 50000000 parent 50m
Clock: shxc -> rate: 50000000 parent 50m
Clock: sp_divtx -> rate: 270000000 parent 270m
Clock: spi -> rate: 225000000 parent bus
Clock: timer -> rate: 50000000 parent 50m
Clock: uart1 -> rate: 50000000 parent 50m
Clock: uart2 -> rate: 50000000 parent 50m
Clock: uart3 -> rate: 50000000 parent 50m
cat: /sys/kernel/debug/clk/xtal/clk_parent: No such file or directory
Clock: xtal -> rate: 40000000 parent 
```

paraka, 2024-08-05:
You can also try the following patch to see if there is a real order setup issue:
```
diff --git a/drivers/clk/ralink/clk-mtmips.c b/drivers/clk/ralink/clk-mtmips.c
index 50a443bf79ec..8a53287585d1 100644
--- a/drivers/clk/ralink/clk-mtmips.c
+++ b/drivers/clk/ralink/clk-mtmips.c
@@ -263,10 +263,6 @@ static int mtmips_register_pherip_clocks(struct device_node *np,
                .rate = _rate            \
        }
 
-static struct mtmips_clk_fixed rt305x_fixed_clocks[] = {
-       CLK_FIXED("xtal", NULL, 40000000)
-};
-
 static struct mtmips_clk_fixed rt3352_fixed_clocks[] = {
        CLK_FIXED("periph", "xtal", 40000000)
 };
@@ -366,6 +362,12 @@ static inline struct mtmips_clk *to_mtmips_clk(struct clk_hw *hw)
        return container_of(hw, struct mtmips_clk, hw);
 }
 
+static unsigned long rt2880_xtal_recalc_rate(struct clk_hw *hw,
+                                            unsigned long parent_rate)
+{
+       return 40000000;
+}
+
 static unsigned long rt5350_xtal_recalc_rate(struct clk_hw *hw,
                                             unsigned long parent_rate)
 {
@@ -677,10 +679,12 @@ static unsigned long mt76x8_cpu_recalc_rate(struct clk_hw *hw,
 }
 
 static struct mtmips_clk rt2880_clks_base[] = {
+       { CLK_BASE("xtal", NULL, rt2880_xtal_recalc_rate) },
        { CLK_BASE("cpu", "xtal", rt2880_cpu_recalc_rate) }
 };
 
 static struct mtmips_clk rt305x_clks_base[] = {
+       { CLK_BASE("xtal", NULL, rt2880_xtal_recalc_rate) },
        { CLK_BASE("cpu", "xtal", rt305x_cpu_recalc_rate) }
 };
 
@@ -690,6 +694,7 @@ static struct mtmips_clk rt3352_clks_base[] = {
 };
 
 static struct mtmips_clk rt3883_clks_base[] = {
+       { CLK_BASE("xtal", NULL, rt2880_xtal_recalc_rate) },
        { CLK_BASE("cpu", "xtal", rt3883_cpu_recalc_rate) },
        { CLK_BASE("bus", "cpu", rt3883_bus_recalc_rate) }
 };
@@ -746,8 +751,8 @@ static int mtmips_register_clocks(struct device_node *np,
 static const struct mtmips_clk_data rt2880_clk_data = {
        .clk_base = rt2880_clks_base,
        .num_clk_base = ARRAY_SIZE(rt2880_clks_base),
-       .clk_fixed = rt305x_fixed_clocks,
-       .num_clk_fixed = ARRAY_SIZE(rt305x_fixed_clocks),
+       .clk_fixed = NULL,
+       .num_clk_fixed = 0,
        .clk_factor = rt2880_factor_clocks,
        .num_clk_factor = ARRAY_SIZE(rt2880_factor_clocks),
        .clk_periph = rt2880_pherip_clks,
@@ -757,8 +762,8 @@ static const struct mtmips_clk_data rt2880_clk_data = {
 static const struct mtmips_clk_data rt305x_clk_data = {
        .clk_base = rt305x_clks_base,
        .num_clk_base = ARRAY_SIZE(rt305x_clks_base),
-       .clk_fixed = rt305x_fixed_clocks,
-       .num_clk_fixed = ARRAY_SIZE(rt305x_fixed_clocks),
+       .clk_fixed = NULL,
+       .num_clk_fixed = 0,
        .clk_factor = rt305x_factor_clocks,
        .num_clk_factor = ARRAY_SIZE(rt305x_factor_clocks),
        .clk_periph = rt305x_pherip_clks,
@@ -779,8 +784,8 @@ static const struct mtmips_clk_data rt3352_clk_data = {
 static const struct mtmips_clk_data rt3883_clk_data = {
        .clk_base = rt3883_clks_base,
        .num_clk_base = ARRAY_SIZE(rt3883_clks_base),
-       .clk_fixed = rt305x_fixed_clocks,
-       .num_clk_fixed = ARRAY_SIZE(rt305x_fixed_clocks),
+       .clk_fixed = NULL,
+       .num_clk_fixed = 0,
        .clk_factor = NULL,
        .num_clk_factor = 0,
        .clk_periph = rt5350_pherip_clks,
```

ascendbeing, 2024-08-05:
I will get that to you today. I've been able to go to and from Padawan stock and openwrt via the uboot web UI.
Settings for Padavan survived flashing openwrt to Padavan and back. I didn't know that when I submitted the bug report.
so I'll reset the padavan settings and flash same or later openwrt. I'll do this for f9k1103 too. I'll post the output for both devices today when I have it and I'll try to point to any differences I see in dmesg which potentially may correlate to having padavan settings not cleared originally.

ascendbeing, 2024-08-06:
@paroka here's what I got on my f9k1109:

``root@OpenWrt:~# for clk in `ls /sys/kernel/debug/clk/`; do echo "clock: $clk ->
 rate: `cat /sys/kernel/debug/clk/$clk/clk_rate` parent `cat /sys/kernel/debug/
clk/$clk/clk_parent`"; done
clock: 10000100.timer -> rate: 166000000 parent bus
clock: 10000120.watchdog -> rate: 166000000 parent bus
clock: 10000500.uart -> rate: 0 parent 
clock: 10000900.i2c -> rate: 0 parent 
clock: 10000a00.i2s -> rate: 0 parent 
clock: 10000b00.spi -> rate: 166000000 parent bus
clock: 10000b40.spi -> rate: 166000000 parent bus
clock: 10000c00.uartlite -> rate: 0 parent 
clock: 10100000.ethernet -> rate: 166000000 parent bus
clock: 10180000.wmac -> rate: 40000000 parent xtal
clock: bus -> rate: 166000000 parent cpu
cat: can't open '/sys/kernel/debug/clk/clk_dump/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_dump/clk_parent': Not a directory
clock: clk_dump -> rate:  parent 
cat: can't open '/sys/kernel/debug/clk/clk_orphan_dump/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_orphan_dump/clk_parent': Not a directory
clock: clk_orphan_dump -> rate:  parent 
cat: can't open '/sys/kernel/debug/clk/clk_orphan_summary/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_orphan_summary/clk_parent': Not a directory
clock: clk_orphan_summary -> rate:  parent 
cat: can't open '/sys/kernel/debug/clk/clk_summary/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_summary/clk_parent': Not a directory
clock: clk_summary -> rate:  parent 
clock: cpu -> rate: 500000000 parent xtal
cat: can't open '/sys/kernel/debug/clk/xtal/clk_parent': No such file or directory
clock: xtal -> rate: 40000000 parent``

paraka, 2024-08-06:
> @paroka here's what I got on my f9k1109:
> 
> root@OpenWrt:~# for clk in `ls /sys/kernel/debug/clk/`; do echo "clock: $clk -> rate: `cat /sys/kernel/debug/clk/$clk/clk_rate` parent `cat /sys/kernel/debug/ clk/$clk/clk_parent`"; done clock: 10000100.timer -> rate: 166000000 parent bus clock: 10000120.watchdog -> rate: 166000000 parent bus clock: 10000500.uart -> rate: 0 parent clock: 10000900.i2c -> rate: 0 parent clock: 10000a00.i2s -> rate: 0 parent clock: 10000b00.spi -> rate: 166000000 parent bus clock: 10000b40.spi -> rate: 166000000 parent bus clock: 10000c00.uartlite -> rate: 0 parent clock: 10100000.ethernet -> rate: 166000000 parent bus clock: 10180000.wmac -> rate: 40000000 parent xtal clock: bus -> rate: 166000000 parent cpu cat: can't open '/sys/kernel/debug/clk/clk_dump/clk_rate': Not a directory cat: can't open '/sys/kernel/debug/clk/clk_dump/clk_parent': Not a directory clock: clk_dump -> rate: parent cat: can't open '/sys/kernel/debug/clk/clk_orphan_dump/clk_rate': Not a directory cat: can't open '/sys/kernel/debug/clk/clk_orphan_dump/clk_parent': Not a directory clock: clk_orphan_dump -> rate: parent cat: can't open '/sys/kernel/debug/clk/clk_orphan_summary/clk_rate': Not a directory cat: can't open '/sys/kernel/debug/clk/clk_orphan_summary/clk_parent': Not a directory clock: clk_orphan_summary -> rate: parent cat: can't open '/sys/kernel/debug/clk/clk_summary/clk_rate': Not a directory cat: can't open '/sys/kernel/debug/clk/clk_summary/clk_parent': Not a directory clock: clk_summary -> rate: parent clock: cpu -> rate: 500000000 parent xtal cat: can't open '/sys/kernel/debug/clk/xtal/clk_parent': No such file or directory clock: xtal -> rate: 40000000 parent Edit: that came out really ugly looking let's see if I can just set it to normal format and if that does anything. edit2: yeah. anyway I'm gonna do my 1103 now. then I'll upload dmesg for both as attachments either in another edit or another reply here edit3: 1103 gives me identical output to 1109 for that command loop. I'm gonna upload the dmesges in a second.

Ok, There are some bads with this clock plan but I see what is happening. There is no "periph" clock defined for your SoC so there are some clocks which are invalid and getting no parent and rate 0. "uart", i2c, i2s and uartlite". Can you try the following patch?

```
diff --git a/drivers/clk/ralink/clk-mtmips.c b/drivers/clk/ralink/clk-mtmips.c
index 50a443bf79ec..d6e1947ad9ba 100644
--- a/drivers/clk/ralink/clk-mtmips.c
+++ b/drivers/clk/ralink/clk-mtmips.c
@@ -264,7 +264,8 @@ static int mtmips_register_pherip_clocks(struct device_node *np,
        }
 
 static struct mtmips_clk_fixed rt305x_fixed_clocks[] = {
-       CLK_FIXED("xtal", NULL, 40000000)
+       CLK_FIXED("xtal", NULL, 40000000),
+       CLK_FIXED("periph", "xtal", 40000000)
 };
```

paraka, 2024-08-06:
> [dm_1103.txt](https://github.com/user-attachments/files/16506533/dm_1103.txt) [dm_1109.txt](https://github.com/user-attachments/files/16506534/dm_1109.txt) Here are full dmesges for stock openwrt HEAD for my 1103 and 1109 respectively. I did a cursory `icdiff -U0` comparison in comparing the two. Main things that stood out were the ErrCtrl point to different addresses or locations (000 vs ca0). A line following that one up also seems to differ like that too (...9o vs ...b0). Other than that they look the same

AFAICS your traces look good (apart from the clk driver warning) and are properly aligned with the documented in https://openwrt.org/toh/belkin/f9k1109_v1. Let's apply the proposed patch and see how it performs. I'd like you to execute the for loop for the clocks with the driver's patch applied if possible. Once I get the correct and expected clock plan I will send the patch to upstream to get this issue fixed. Take into account that we could not test this SoC when the driver was mainlined since no users at all appear, so we only tested mt7620 mt7628 and rt5030 :). It is awesome to have your debug now :-).

Thanks at all for your time and sorry for inconvenince!!!

paraka, 2024-08-06:
@DragonBluep we should increase dts's clocks indexes also to add one to all clocks for peripherals. If you have time, can you maybe do a branch with this patch and dts changes to make easier the testing? If not I will try to do myself sometime next week.

DragonBluep, 2024-08-06:
> @DragonBluep we should increase dts's clocks indexes also to add one to all clocks for peripherals. If you have time, can you maybe do a branch with this patch and dts changes to make easier the testing? If not I will try to do myself sometime next week.

Sure. Thanks for your help. @ascendbeing Please try this branch: https://github.com/DragonBluep/openwrt/tree/fix-belkin-rt3883. Or you can download testing firmware here: https://github.com/DragonBluep/Actions/actions/runs/10267096316

@paraka BTW, would you mind add a 48M sdcard clock for MT7620 and MT7628? We are now using the upstream mmc driver for these two SoCs, see: https://github.com/openwrt/openwrt/pull/15896/files.

paraka, 2024-08-06:
@DragonBluep Thanks a lot! Let's wait for @ascendbeing feedback. Good to know that openwrt people switch to use upstream mtk-sd driver! Awesome. Will add 48M sdcard clock sonner than later to avoid that ugly fixed clock node in dts :-)

paraka, 2024-08-06:
BTW, patch sent to upstream: https://lore.kernel.org/linux-clk/20240806142902.224164-1-sergio.paracuellos@gmail.com/T/#u

ascendbeing, 2024-08-06:
Hi just saw this. I have a half decent build PC (ryzen 5700x w/ 64gb ddr4 ECC 3200mhz). So if there are any new commits I can probably build it if it's fairly straightforward (trying arbitrary patches isn't terrifying to me it's just i am not knowledgeable about how to implement them and have them stay like that in the owrt build ecosystem)
 I can go ahead and try the artifacts build though. Is there anything in particular I should be looking at when I boot? dmesg and same command output? or look at the 2.4ghz and whether it works now? should I see if the second wifi registers as being 2 4 or 5 switchable (Dropdown pickable in Luci) despite it only ever working as 5 on any build I've tried?

DragonBluep, 2024-08-07:
> **Here's dmesg from f9k1103v1 running your artifacts build:** [dm_1103_withpatch.txt](https://github.com/user-attachments/files/16515284/dm_1103_withpatch.txt) **Here's a screenshot of the command output running the earlier command with the build:** ![Screenshot_20240806_131702_1](https://private-user-images.githubusercontent.com/119428197/355591115-29d32261-c4bf-4beb-a374-47f519b871d7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI5OTIwMTEsIm5iZiI6MTcyMjk5MTcxMSwicGF0aCI6Ii8xMTk0MjgxOTcvMzU1NTkxMTE1LTI5ZDMyMjYxLWM0YmYtNGJlYi1hMzc0LTQ3ZjUxOWI4NzFkNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwN1QwMDQ4MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NGMwOWYxZWFkMzg3YzQ5ODY0NmYxNDRmOTQxNDFhN2Y4NWQzNjUyMTgzOTBkNTkzNDQzMDdhYmUzOWNhZmMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XIATthegX1MEdMEkKQV2E74wAWVgFQl96pTou43bB5s) 

You uploaded the wrong update image. This image is not downloaded from my link, my version number is `r0-d5ef6c2`.

>^ edits 2-3 Hi just saw this. I have a half decent build PC (ryzen 5700x w/ 64gb ddr4 ECC 3200mhz). So if there are any new commits I can probably build it if it's fairly straightforward (trying arbitrary patches isn't terrifying to me it's just i am not knowledgeable about how to implement them and have them stay like that in the owrt build ecosystem) I can go ahead and try the artifacts build though. 

If you are interested in building your own image, you can take a look on https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem.

>Is there anything in particular I should be looking at when I boot? dmesg and same command output? or look at the 2.4ghz and whether it works now? should I see if the second wifi registers as being 2 4 or 5 switchable (Dropdown pickable in Luci) despite it only ever working as 5 on any build I've tried? EDIT currently flashing the build onto the f9k1103v1 (had that one plugged in) edit4 pinging @DragonBluep

Yes for all questions.

paraka, 2024-08-07:
@ascendbeing I do believe that you upload the wrong image as @DragonBluep said. The expected would be the warning dissapears and the i2c, i2s, uart and uartlite clocks having 'periph' as parent with a rate of 40 MHz. Regarding the radios, I don't know if they will work but if we don't have a correct clock plan first we are not in a good shape at all to test anything else. What is clear is that the clock plan is correct and the  radios are still not working, this is not a clock issue.

paraka, 2024-08-07:
> @DragonBluep Thanks a lot! Let's wait for @ascendbeing feedback. Good to know that openwrt people switch to use upstream mtk-sd driver! Awesome. Will add 48M sdcard clock sonner than later to avoid that ugly fixed clock node in dts :-)

@DragonBluep, Can you please test the following patch?

[0001-clk-ralink-mtmips-add-mmc-related-clocks-for-SoCs-MT.PATCH](https://github.com/user-attachments/files/16519581/0001-clk-ralink-mtmips-add-mmc-related-clocks-for-SoCs-MT.PATCH)

DragonBluep, 2024-08-07:
> > @DragonBluep Thanks a lot! Let's wait for @ascendbeing feedback. Good to know that openwrt people switch to use upstream mtk-sd driver! Awesome. Will add 48M sdcard clock sonner than later to avoid that ugly fixed clock node in dts :-)
> 
> @DragonBluep, Can you please test the following patch? (rebased on the top of the rt3383 patch fixes)
> 
> [0001-clk-ralink-mtmips-add-mmc-related-clocks-for-SoCs-MT.PATCH](https://github.com/user-attachments/files/16520313/0001-clk-ralink-mtmips-add-mmc-related-clocks-for-SoCs-MT.PATCH)

Thanks. It works. I find another small issue. For MT7628, 10000b40.spi clock is useless as it only has one spi port, the same as MT7621.
<details>
<summary>MT7620:</summary>

```
Clock: 10000100.timer -> rate: 40000000 parent periph
Clock: 10000120.watchdog -> rate: 40000000 parent periph
Clock: 10000500.uart -> rate: 40000000 parent periph
Clock: 10000900.i2c -> rate: 40000000 parent periph
Clock: 10000a00.i2s -> rate: 40000000 parent periph
Clock: 10000b00.spi -> rate: 193333333 parent bus
Clock: 10000b40.spi -> rate: 193333333 parent bus
Clock: 10000c00.uartlite -> rate: 40000000 parent periph
Clock: 10130000.mmc -> rate: 48000000 parent sdhc
Clock: 10180000.wmac -> rate: 20000000 parent xtal
Clock: bbppll -> rate: 480000000 parent xtal
Clock: bus -> rate: 193333333 parent cpu
Clock: clk_dump -> rate:  parent
Clock: clk_orphan_dump -> rate:  parent
Clock: clk_orphan_summary -> rate:  parent
Clock: clk_summary -> rate:  parent
Clock: cpu -> rate: 580000000 parent pll
Clock: periph -> rate: 40000000 parent xtal
Clock: pll -> rate: 580000000 parent xtal
Clock: sdhc -> rate: 48000000 parent bbppll
Clock: xtal -> rate: 20000000 parent
```
</details>

<details>
<summary>MT7628:</summary>

```
Clock: 10000100.timer -> rate: 40000000 parent periph
Clock: 10000120.watchdog -> rate: 40000000 parent periph
Clock: 10000900.i2c -> rate: 40000000 parent periph
Clock: 10000a00.i2s -> rate: 480000000 parent pcmi2s
Clock: 10000b00.spi -> rate: 191666666 parent bus
Clock: 10000b40.spi -> rate: 191666666 parent bus
Clock: 10000c00.uart0 -> rate: 40000000 parent periph
Clock: 10000d00.uart1 -> rate: 40000000 parent periph
Clock: 10000e00.uart2 -> rate: 40000000 parent periph
Clock: 10130000.mmc -> rate: 48000000 parent sdhc
Clock: 10300000.wmac -> rate: 20000000 parent xtal
Clock: bbppll -> rate: 480000000 parent xtal
Clock: bus -> rate: 191666666 parent cpu
Clock: clk_dump -> rate:  parent
Clock: clk_orphan_dump -> rate:  parent
Clock: clk_orphan_summary -> rate:  parent
Clock: clk_summary -> rate:  parent
Clock: cpu -> rate: 575000000 parent xtal
Clock: pcmi2s -> rate: 480000000 parent bbppll
Clock: periph -> rate: 40000000 parent xtal
Clock: sdhc -> rate: 48000000 parent bbppll
Clock: xtal -> rate: 20000000 parent
```
</details>

paraka, 2024-08-07:
@DragonBluep awesome, thanks for testing. Can I get your Tested-by for when I send the patch to upstream? Will do once the fix is accepted.

Regarding the 10000b40.spi it is in original clock plan:
- https://elixir.bootlin.com/linux/v5.4.281/source/arch/mips/ralink/mt7620.c#L566

But will remove it also if it is not needed.

DragonBluep, 2024-08-07:
> @DragonBluep awesome, thanks for testing. Can I get your Tested-by for when I send the patch to upstream? Will do once the fix is accepted.

`Tested-by: Shiji Yang <yangshiji66@outlook.com>`

> 
> Regarding the 10000b40.spi it is in original clock plan:
> 
> * https://elixir.bootlin.com/linux/v5.4.281/source/arch/mips/ralink/mt7620.c#L566
> 
> But will remove it also if it is not needed. Since mt7628 and 7688 share the same clocks as mt76x8, is it possible that it exists because of that?

The datasheet said both of them have two spi interfaces. But it seems that we only register them as one device. So maybe just leave the unused clock. Annyway, it's harmless.

paraka, 2024-08-07:
> > @DragonBluep awesome, thanks for testing. Can I get your Tested-by for when I send the patch to upstream? Will do once the fix is accepted.
> 
> `Tested-by: Shiji Yang <yangshiji66@outlook.com>`
> 

Thanks!

> The datasheet said both of them have two spi interfaces. But it seems that we only register them as one device. So maybe just leave the unused clock. Annyway, it's harmless.

Yes, just maintain what it was already there initially... As yousaid it is harmless.

ascendbeing, 2024-08-10:
Actually I was able to flash the .bin stock fw. Gonna try padavan again

> @ascendbeing I do believe that you upload the wrong image as @DragonBluep said. The expected would be the warning dissapears and the i2c, i2s, uart and uartlite clocks having 'periph' as parent with a rate of 40 MHz. Regarding the radios, I don't know if they will work but if we don't have a correct clock plan first we are not in a good shape at all to test anything else. What is clear is that if the clock plan is correct and the radios are still not working, this is not a clock issue.

I don't think I uploaded the wrong image. What I did was flash the f9k1103v1 with dragons f9k1109v1 artifact build. That was what I got. So perhaps it isn't done correctly for f9k1103v1 in that build? I didn't try it against 09 but I can if y'all want. Or if you have a newer build I could try it on the 03.

ascendbeing, 2024-08-10:
> @ascendbeing I do believe that you upload the wrong image

Not possible. I did not grab that from any other posted screenshot.

There is a possibility that since i flashed it via sysupgrade rather than web recovery, it may have not taken dragons build.

I made no attempt to verify if the build took.

The other possibility is the code addressed the 1109 and not the 1103. As mentioned, I flashed it (via sysupgrade), to my 1103.

I would rate my former explanation as likely what happened: it may have not taken the build because of whatever checking sysupgrade may do (signature? version lower maybe? etc).

i will try to flash the build (or a newer one of your choosing if preferred) via web recovery rather than sysupgrade. probably will have that today for y'all.

ascendbeing, 2024-08-11:
@DragonBluep @paraka The former theory proved correct (sysupgrade didn't take). I just ran it on the 1109. here:

root@OpenWrt:~# for clk in `ls /sys/kernel/debug/clk/`; do echo "clock: $clk -> rate: `cat /sys/kernel/debug/clk/$clk/clk_rate` parent `cat /sys/kernel/debu
g/clk/$clk/clk_parent`"; done
clock: 10000100.timer -> rate: 166000000 parent bus
clock: 10000120.watchdog -> rate: 166000000 parent bus
clock: 10000500.uart -> rate: 40000000 parent periph
clock: 10000900.i2c -> rate: 40000000 parent periph
clock: 10000a00.i2s -> rate: 40000000 parent periph
clock: 10000b00.spi -> rate: 166000000 parent bus
clock: 10000b40.spi -> rate: 166000000 parent bus
clock: 10000c00.uartlite -> rate: 40000000 parent periph
clock: 10100000.ethernet -> rate: 166000000 parent bus
clock: 10180000.wmac -> rate: 40000000 parent xtal
clock: bus -> rate: 166000000 parent cpu
cat: can't open '/sys/kernel/debug/clk/clk_dump/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_dump/clk_parent': Not a directory
clock: clk_dump -> rate:  parent 
cat: can't open '/sys/kernel/debug/clk/clk_orphan_dump/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_orphan_dump/clk_parent': Not a directory
clock: clk_orphan_dump -> rate:  parent 
cat: can't open '/sys/kernel/debug/clk/clk_orphan_summary/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_orphan_summary/clk_parent': Not a directory
clock: clk_orphan_summary -> rate:  parent 
cat: can't open '/sys/kernel/debug/clk/clk_summary/clk_rate': Not a directory
cat: can't open '/sys/kernel/debug/clk/clk_summary/clk_parent': Not a directory
clock: clk_summary -> rate:  parent 
clock: cpu -> rate: 500000000 parent xtal
clock: periph -> rate: 40000000 parent xtal
cat: can't open '/sys/kernel/debug/clk/xtal/clk_parent': No such file or directory
clock: xtal -> rate: 40000000 parent

ascendbeing, 2024-08-11:
Ok the 1109 and 1103 give the same clock data from the commands
Since the dmesg I posted last time was not useful here's my 1109 running @DragonBluep firmware build:
[dmesg_dbluep_1109.txt](https://github.com/user-attachments/files/16576386/dmesg_dbluep_1109.txt)

paraka, 2024-08-12:
@ascendbeing I don't really understand why the warning is still appearing on initial trace BUT it is just cosmetic since all the clock plan is correct now. Are the radios still don't working? If not I do believe it is not, I do believe this is not a clock issue. Checking traces and comparing to https://openwrt.org/toh/belkin/f9k1109_v1 I can see two differences. MDIO is complaining in you trace as follows:
```
[    2.778637] rtl8367b rtl8367b: cannot find mdio node phandle
```
And the following phy0 and phy1 'minstrel_ht' rate algorithm selection which is not in original trace also:
```
[   37.495722] NET: Registered PF_PPPOX protocol family
[   37.658029] rt2800pci 0000:01:00.0: enabling device (0000 -> 0002)
[   37.670687] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3071, rev 021c detected
[   37.689696] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 0001 detected
[   37.705787] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   37.857078] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 3883, rev 0400 detected
[   37.873270] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 3853 detected
[   37.889422] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
```

ascendbeing, 2024-08-12:
I didn't check yet. I'm not sure what all is compiled in, but since this is an arbitrary fork build, if it doesn't have the stuff to connect to wifi like wpa_supplicant or the web GUI, I take it it will require clone the fork and try to build them from there?

ascendbeing, 2024-08-13:
with @DragonBluep build, the second wifi device is correctly detected as a/n not a/b/g/n. it correctly scans for a/n. So that one's fine

however, @paraka with 2.4g if I scan, or if I host an AP, nothing. Nothing in logs. I'll try to up the log level.

would it help if I grab the dmesg and command output from a 23.05-SNAPSHOT build? That one has working 2.4g. meaning something changed in between 23.05 and snapshot to adversely cause that.

ascendbeing, 2024-08-13:
Here's the 23.05 dmesg. broken looking at the start but the 2.4g works and the 5 works as long as you don't try to use it as a 2.4g

paraka, 2024-08-14:
@ascendbeing It is normal that your 23.05 snapshot trace cannot show the clocks with the four loop. Clock code was on kernel's arch files before and it has been moved into a new driver. However your clock plan is CORRECT and aligned with the one before in 5.15 (https://elixir.bootlin.com/linux/v5.15.164/source/arch/mips/ralink/rt3883.c) with the @DragonBluep 's build. Which radio is the one that is not working with @DragonBluep 's build? The one connected to the pci or the one which uses wmac directly? I cannot see anything that makes me thing something is wrong with your kernel traces... @DragonBluep Any idea of something that is wrong with this now?

ascendbeing, 2024-08-14:
@paraka in every build, the 3883 (a/n) is always phy1 and works great unless you choose to set it to 2.4g which is possible in every build except @DragonBluep . A good change and nothing broken in exchange for that fix.

On snapshot based on main/HEAD/trunk/6.6, phy0 (2.4g, it's an addon chip) is essentially a no-op, printing messages but not functioning in any way.

To be able to tell them apart easily, phy1 is named rt3883, which is also the name of the sub target or target (ramips/rt3883). Phy0 is a chip I think Belkin added to the platform. In every dmesg I posted it always goes in order phy0 (2.4) then phy1 (rt3883). Phy0 is the one that is at issue and issue isn't dragons fault it's something happened in between 5.15 and 6.6

DragonBluep, 2024-08-14:
I think I've found the phy0 PCIe NIC issue, it's caused by https://github.com/openwrt/openwrt/pull/13697. I can work a patch to fix it on weekend. But we still need to solve the clock driver warning.

ascendbeing, 2024-08-15:
I reviewed TD wiki and according to them, 1109 has flash "Macronix MX25L6405D" and RAM "Winbond W9751G6JB-25"; and 1103 has flash "Macronix MX25L6406EM2I-12G" and RAM "EtronTech EM68B16CWPA-25H".
unsure how accurate, because same pages show phy1 and phy0 ordering for 1103 and phy0 and phy1 ordering for 1109.
I just booted up 1103 to verify orderings, and it's in phy0 (pcie) phy1 (rt3883 5ghz) order, same as the 1109's I most recently shared here.

ascendbeing, 2024-08-17:
@DragonBluep This issue with phy0 2.4ghz appears to be more complex. I started playing with 1109 tonight: flashed stock 23.05-SNAP.
The first time I booted up, scanning 2.4 yielded nothing. scanning 5 yielded results. 
I got 2.4 to work again after that somehow (reboot? restart device a few times? something), but eventually, after I got it to work, after a reboot it happened again. 2.4 won't connect to the AP I configured, so I switched it to auto, nothing. Scanned, nothing.
So this issue isn't as binary as 23.05 good 6.6 bad. It's possible there's something that triggers it to work or not, which affects both 23.05 and main. 
This was the soonest I could confirm this with certainty. I hope this helps to delineate what is actually occurring, and if there's any commands I should try to provide anything tangible, please feel free to let me know. 
When this most recently happened the last two lines in the log were:
```[  111.730934] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2860.bin'```
```[  115.400558] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.40```

DragonBluep, 2024-08-17:
@ascendbeing Please try this version and upload kernel log.
https://github.com/DragonBluep/Actions/actions/runs/10267096316

ascendbeing, 2024-08-17:
Got it, will do shortly, or else later today.

ascendbeing, 2024-08-17:
@DragonBluep 
[latest_build_on_f9k1103v1_082024.txt](https://github.com/user-attachments/files/16645602/latest_build_on_f9k1103v1_082024.txt)

ascendbeing, 2024-08-17:
@DragonBluep I did one reboot post install. Scan didn't work (2.4). Scanned 5, then thought to try channel analysis. It pulled stuff up the first try, and same boot now the 2.4 scan is working. 
I'll poke around a little bit more, to see if there's a reliable way to always reproduce or always work around it, from a user point of view. I will also need to edit my OP to make clear that it's a random problem affecting at least 23.05.x and higher.

DragonBluep, 2024-08-17:
> @DragonBluep I did one reboot post install, after I'd said it didn't work. On reboot Scan still didn't work (2.4). Scanned 5, then thought to try channel analysis. It pulled stuff up the first try, and after that now the 2.4 scan is working. I'll poke around a little bit more, to see if there's a reliable way to always reproduce or always work around it, from a user point of view. I will also need to edit my OP to make clear that it's a random problem affecting at least 23.05.x and higher.

I think it just needs some time to initialize the hardware, about two minutes.

DragonBluep, 2024-08-17:
@paraka Based on your patch, I set the parent clock of the "cpu" to NULL, and it fixes the kernel warning. So I think we still need to register the "xtal" clock first, even if we don't use it to calculate the CPU clock. I didn't track the clock source code but it seems that it will check if parent clock is uninitialized and then it will try again after parent clock is ready. That's why we can get the correct clock frequency but the kernel still throw the warning. The "cpu" and "bus" clock have been deferred.

ascendbeing, 2024-08-18:
@DragonBluep when I was trying 23.05-SNAP, (stock, so none of your patches), when I made my new hypothesis, it had been stuck in that state for at least an hour, if not longer. So it was well past two minutes, and those lines (or 5.15 branch equivalent) were the last lines in the log.
I haven't determined whether the indefinite 2.4 no-op occurs in HEAD. It's possible it doesn't.

paraka, 2024-08-19:
> @paraka Based on your patch, I set the parent clock of the "cpu" to NULL, and it fixes the kernel warning. So I think we still need to register the "xtal" clock first, even if we don't use it to calculate the CPU clock. I didn't track the clock source code but it seems that it will check if parent clock is uninitialized and then it will try again after parent clock is ready. That's why we can get the correct clock frequency but the kernel still throw the warning. The "cpu" and "bus" clock have been deferred.
> 
> ref: https://github.com/DragonBluep/openwrt/tree/fix-belkin-rt3883

@DragonBluep Thanks, I think you are right. We have three ways of fixing this issue:
- Declare the xtal in the clock base group and just return fixed 40 MHz in its related recalc() callback (I was told to declare it as fixed in driver first upstream submit so I don't know, but the explanation about the order probably makes sense to Stephen too).
- Get rid of the WARN_ON_ONCE since we know that the clock plan is correct in any case at the end (no other WARN_ON_ONCE on driver code).
- Assume cpu clock has no parent as you did since we don't know how to derive it from the xtal clock properly (would be different from the rest of the SoCs).

In any case it is a different issue from the one fixing the peripheral stuff, so it will be a different patch. Please, let me know your thoughts on this.

DragonBluep, 2024-08-20:
I prefer the first way, moving `xtal` to the base group. `xtal` should be the physical root clock node of all clocks, even if there are no registers here to configure the relevant PLLs.

paraka, 2024-08-20:
Agreed. So the following should work (rebased on the top of my previous two patches: the fix and the mmc):
```
diff --git a/drivers/clk/ralink/clk-mtmips.c b/drivers/clk/ralink/clk-mtmips.c
index c716cfbd9404..898be1dbdf11 100644
--- a/drivers/clk/ralink/clk-mtmips.c
+++ b/drivers/clk/ralink/clk-mtmips.c
@@ -270,7 +270,6 @@ static struct mtmips_clk_fixed rt305x_fixed_clocks[] = {
 };
 
 static struct mtmips_clk_fixed rt3383_fixed_clocks[] = {
-       CLK_FIXED("xtal", NULL, 40000000),
        CLK_FIXED("periph", "xtal", 40000000)
 };
 
@@ -472,6 +471,12 @@ static unsigned long rt305x_cpu_recalc_rate(struct clk_hw *hw,
        }
 }
 
+static unsigned long rt3883_xtal_recalc_rate(struct clk_hw *hw,
+                                            unsigned long parent_rate)
+{
+       return 40000000;
+}
+
 static unsigned long rt3883_cpu_recalc_rate(struct clk_hw *hw,
                                            unsigned long xtal_clk)
 {
@@ -711,6 +716,7 @@ static struct mtmips_clk rt3352_clks_base[] = {
 };
 
 static struct mtmips_clk rt3883_clks_base[] = {
+       { CLK_BASE("xtal", NULL, rt3883_xtal_recalc_rate) },
        { CLK_BASE("cpu", "xtal", rt3883_cpu_recalc_rate) },
        { CLK_BASE("bus", "cpu", rt3883_bus_recalc_rate) }
 };
```

paraka, 2024-08-20:
> Do we need to apply the similar change to rt2880 and rt305x?
> 
> I'll build a testing firmware later.

@DragonBluep No warning will appear and clock plan will be correct, but yes we should do the same to be coherent. I have updated the diff in previous comment.

DragonBluep, 2024-08-21:
@ascendbeing Please try this version, disable 5GHz radio && reboot && upload kernel log.
https://github.com/DragonBluep/Actions/actions/runs/10267096316

ascendbeing, 2024-08-23:
Dragon I will get that done and sorted for you today, just saw this.

ascendbeing, 2024-08-23:
is there anything radically different with the build @DragonBluep ? 1. tried flashing via chrome on web recovery, destination unreachable thereafter 2. same with Firefox 3. successfully flashed latest 23.05 snapshot (official, no image builder, no fwselecta), did ```sysupgrade -v -i``` after ```scp```ing the upgrade image, typed ```n```, been waiting like 5-10 min same thing. these flashes are all on the f9k1109.

ascendbeing, 2024-08-24:
@DragonBluep if ya want a build tried it'll have to be another 1. something went wrong. all 3 ways I tried flashing (2 browsers via web recovery ; 1 cli sys upgrade) soft bricked.
No urgency just trying to summarize in fewer words.

DragonBluep, 2024-08-25:
@ascendbeing Please try https://github.com/DragonBluep/Actions/actions/runs/10267096316.

DragonBluep, 2024-08-26:
> @DragonBluep I tried flashing the build 3x. Firefox and chrome timed out during the flash which I consider unusual. Then it didn't work so I flashed 23.05 stock. then I did flash 3 (`sysupgrade -v -i`) and it also didn't work. There's a contrast between this build and the attempt I had at flashing a custom built padavan the other day. Flashing the custom padavan didn't take. This build took, but doesn't work. While the padavan custom dropped me into a web recovery page reload, this leaves me with a soft brick until I flash something else. I don't have hardware to connect lower level than standard Ethernet I have to conclude this is a **Bad** bulld, maybe per luck, maybe too radical of changes. I will check back here, can't use this build, and indeed I checked the digests.

It seems that some recent commits break the DHCP service. My MT7620 device also have the similar issue. Please retry this **Bad** version with static IP `192.168.1.2` and server IP `192.168.1.1`.

ascendbeing, 2024-08-27:
@DragonBluep how about this: can you spin up a new build, but with WiFi AP preconfigured? 
cause I tried that. not with IP ending in 2. it was probably 6 thru around 30 that I tried. 
i believe that the router was up but I couldn't access 192.168.1.1. if you can spin up that build with WiFi maybe I could try static via WiFi and get the log back to ya.

DragonBluep, 2024-08-27:
> [096316](https://github.com/DragonBluep/Actions/actions/runs/10267096316).

Just try this final build. It should fixes the net issue. The WiFi still default off on this build.

ascendbeing, 2024-09-07:
@DragonBluep Been away, booted up the build I said soft bricks (already on 1109, no reflash): 192.168.1.2: no rx pkts; 192.168.0.2: no rx pkts. I'm not caught up with the the thread here, but I'm probably gonna try using 1 of these, as dumb AP/dumb STA+AP, so I will most likely update with how that goes. mostly will be using USB WiFi though.

ascendbeing, 2024-09-09:
@DragonBluep the build before I was away and the build you last said to try, both of them soft brick. The last build that didn't soft brick was 2 builds ago. I don't think I tried the 2nd to last on the 1103. I only tried your latest on 1109, and result is soft brick. I can try 1103 if you would like.

DragonBluep, 2024-09-09:
Please re-download the firmware from the link. It has been refreshed. This build should enabled wifi by default.

ascendbeing, 2024-09-09:
thanks dragon. I also just tried flashing the latest snapshot. which also soft bricks
 I'll try yours rn

[4 more replies]