openwrt/openwrt #16055: Explanation for my IP ban

Kcajminer2312, 2024-08-02:
### Describe the bug

This is not a bug repoort, I have been IP banned after I deleted my account and created it again
is there anyway of me contacting slh or peter sherman

slh ip banned me, but did not give me time to read there message
I want my account back
I left myself, I was not banned
so why did I get banned

### OpenWrt version

N\A

### OpenWrt release

N\A

### OpenWrt target/subtarget

N\A

### Device

N\A

### Image kind

Self-built image

### Steps to reproduce

N\A

### Actual behaviour

N\A

### Expected behaviour

N\A

### Additional info

N\A

### Diffconfig

```text
N\A
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vincejv, 2024-08-03:
Did you try OFTC Irc? They are online there

thess, 2024-08-03:
Not a dev issue - User has been banned from forum.

closed 2024-08-03 by thess