openwrt/openwrt #16060: No WiFI in Master mode

iqdba, 2024-08-03:
### Describe the bug

Hello!
I installed the latest firmware 23.05.4, but it does not work the wifi in access points mode
Luci shows that radio0 is active and shows that the SSID is working, but there is no wireless network itself

Client mode working correct

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link CPE605 v1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-08-03:
Please attach system log.

brada4, 2024-08-03:
Add output of `iwinfo` (strike out last 3 groups in MAC address and make SSID name nonsensical) pertaining the defective AP.