openwrt/openwrt #16066: bcm27xx: kernel panic in __udp_gso_segment and reboot

schuettecarsten, 2024-08-04:
### Describe the bug

I get these kind of crashed from time to time:

```
[47827.577605] Unable to handle kernel read from unreadable memory at virtual address 00000000000000c0
[47827.586687] Mem abort info:
[47827.589479]   ESR = 0x0000000096000005
[47827.593228]   EC = 0x25: DABT (current EL), IL = 32 bits
[47827.598549]   SET = 0, FnV = 0
[47827.601603]   EA = 0, S1PTW = 0
[47827.604743]   FSC = 0x05: level 1 translation fault
[47827.609622] Data abort info:
[47827.612500]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[47827.617986]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[47827.623038]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[47827.628352] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000045f53000
[47827.634805] [00000000000000c0] pgd=08000000534ea003, p4d=08000000534ea003, pud=08000000534ea003, pmd=0000000000000000
[47827.645449] Internal error: Oops: 0000000096000005 [#1] SMP
[47827.651022] Modules linked in: brcmfmac_wcc(O) pppoe ppp_async brcmfmac(O) wireguard pppox ppp_generic nft_fib_inet nf_flow_table_inet lz4 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211(O) cdc_ncm cdc_ether zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connlabel xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet usbhid ts_fsm ts_bm tcp_bbr slhc sch_cake r8169 r8152 poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tables nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp(O) nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog
[47827.651335]  nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp(O) nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo_rle lzo lz4_decompress lz4_compress libcurve25519_generic libchacha lan78xx iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hid_generic crc_ccitt compat(O) cls_flower brcmutil(O) br_netfilter asn1_decoder act_vlan fuse ntfs3 em_ipset cls_bpf act_bpf act_ctinfo act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs hid evdev raspberrypi_hwmon gpio_fan hwmon i2c_dev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port
[47827.740980]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfs ip6_gre ip_gre gre ifb dummy ip6_udp_tunnel udp_tunnel ip6_tunnel oid_registry tunnel6 ip_tunnel veth lockd sunrpc grace autofs4 dns_resolver nls_utf8 bfq xxhash_generic md5 blake2b_generic crypto_acompress nls_iso8859_1 nls_cp437 ehci_pci fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd vfat fat exfat btrfs xor zstd_decompress zstd_compress zstd_common zlib_inflate zlib_deflate xxhash xor_neon raid6_pq lzo_decompress lzo_compress libcrc32c realtek microchip
[47827.889924] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O       6.6.43 #0
[47827.897325] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[47827.903155] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[47827.910119] pc : __udp_gso_segment+0x364/0x550
[47827.914575] lr : __udp_gso_segment+0x334/0x550
[47827.919020] sp : ffffffc0800030a0
[47827.922331] x29: ffffffc0800030a0 x28: 0000000000000011 x27: 0000000000040000
[47827.929476] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff809c9bfb00
[47827.936619] x23: 0000000000000000 x22: 0000000000000000 x21: ffffff809c9bfb00
[47827.943761] x20: 0200000000005000 x19: 0000000000000000 x18: 0000000000000070
[47827.950903] x17: ffffffc07ac5f000 x16: ffffffc080000000 x15: 0a0d646578696d2f
[47827.958045] x14: 7472617069746c75 x13: 0a0d30203a687467 x12: 6e654c2d746e6574
[47827.965187] x11: 0a0d0a0d30203a68 x10: 000000000000002d x9 : 0000000000000000
[47827.972329] x8 : ffffffc080003140 x7 : 0000000000000200 x6 : 0000000000000732
[47827.979471] x5 : ffffff809b89cf40 x4 : ffffff809b89d7c0 x3 : 0000000000000000
[47827.986612] x2 : 0000000000000002 x1 : ffffff809b89cf40 x0 : ffffff809b89cfe4
[47827.993754] Call trace:
[47827.996196]  __udp_gso_segment+0x364/0x550
[47828.000294]  udp4_ufo_fragment+0x138/0x16c
[47828.004392]  inet_gso_segment+0x108/0x378
[47828.008405]  skb_mac_gso_segment+0xa0/0xf8
[47828.012506]  __skb_gso_segment+0x64/0x14c
[47828.016519]  0xffffffc0793f4fec
[47828.019664]  __dev_queue_xmit+0x498/0xda8
[47828.023674]  neigh_direct_output+0x18/0x24
[47828.027772]  ip_finish_output2+0x194/0x540
[47828.031868]  __ip_finish_output+0xa8/0x168
[47828.035962]  ip_finish_output+0x48/0x11c
[47828.039883]  ip_output+0x54/0xbc
[47828.043109]  ip_forward_finish+0x68/0x88
[47828.047035]  ip_forward+0x42c/0x458
[47828.050526]  ip_rcv_finish+0x44/0x5c
[47828.054103]  0xffffffc0794325ac
[47828.057244]  nf_hook_slow+0x48/0xec
[47828.060731]  ip_rcv+0x5c/0xd4
[47828.063701]  __netif_receive_skb_one_core+0x48/0x58
[47828.068578]  netif_receive_skb+0x58/0xf4
[47828.072500]  br_pass_frame_up+0x140/0x198
[47828.076512]  br_handle_frame_finish+0x4fc/0x59c
[47828.081043]  br_nf_hook_thresh+0xe8/0x1410 [br_netfilter]
[47828.086449]  br_nf_hook_thresh+0x268/0x1410 [br_netfilter]
[47828.089863] banIP/fwd-lan/reject/voipv4: IN=br-lan OUT=pppoe-wan PHYSIN=br-eth0.1 MAC=d8:3a:dd:83:61:f9:5c:e4:2a:2a:81:22:08:00 SRC=192.168.179.121 DST=149.154.167.99 LEN=52 TOS=0x00 PREC=0x00 TTL=127 ID=11521 DF PROTO=TCP SPT=53743 DPT=443 WINDOW=64240 RES=0x00 SYN URGP=0
[47828.091935]  br_nf_hook_thresh+0x910/0x1410 [br_netfilter]
[47828.121667]  br_handle_frame+0x16c/0x414
[47828.125591]  __netif_receive_skb_core.constprop.0+0x24c/0xe74
[47828.131337]  __netif_receive_skb_one_core+0x2c/0x58
[47828.136215]  netif_receive_skb+0x58/0xf4
[47828.140135]  br_pass_frame_up+0x140/0x198
[47828.144146]  br_handle_frame_finish+0x4fc/0x59c
[47828.148677]  br_nf_hook_thresh+0xe8/0x1410 [br_netfilter]
[47828.154083]  br_nf_hook_thresh+0x268/0x1410 [br_netfilter]
[47828.159575]  br_nf_hook_thresh+0x910/0x1410 [br_netfilter]
[47828.165066]  br_handle_frame+0x16c/0x414
[47828.168989]  __netif_receive_skb_core.constprop.0+0x24c/0xe74
[47828.174735]  __netif_receive_skb_one_core+0x2c/0x58
[47828.179611]  process_backlog+0x9c/0x1ec
[47828.183445]  __napi_poll+0x38/0x178
[47828.186932]  net_rx_action+0x130/0x278
[47828.190680]  handle_softirqs+0xf8/0x22c
[47828.194518]  __do_softirq+0x14/0x20
[47828.198005]  ____do_softirq+0x10/0x1c
[47828.201668]  call_on_irq_stack+0x24/0x4c
[47828.205590]  do_softirq_own_stack+0x1c/0x28
[47828.209773]  irq_exit_rcu+0x90/0xc8
[47828.213262]  el1_interrupt+0x38/0x54
[47828.216839]  el1h_64_irq_handler+0x18/0x24
[47828.220935]  el1h_64_irq+0x68/0x6c
[47828.224335]  default_idle_call+0x28/0x3c
[47828.228259]  do_idle+0x1dc/0x228
[47828.231488]  cpu_startup_entry+0x38/0x3c
[47828.235410]  kernel_init+0x0/0x1e8
[47828.238813]  arch_post_acpi_subsys_init+0x0/0x8
[47828.243344]  start_kernel+0x45c/0x520
[47828.247006]  __primary_switched+0xb4/0xbc
[47828.251022] Code: 79000813 3707fb56 f9400313 f9406301 (f9406260)
[47828.257117] ---[ end trace 0000000000000000 ]---
[47828.261733] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[47828.268604] SMP: stopping secondary CPUs
[47828.272526] Kernel Offset: disabled
[47828.276010] CPU features: 0x0,80000000,34020000,0000400b
[47828.281321] Memory Limit: none
[47828.284374] Rebooting in 3 seconds..
```

### OpenWrt version

r26997-a67b20e3d5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

unterd0g, 2024-08-04:
Could this be related to commit b5c5384 and another manifestation of bug report #15821 ?

schuettecarsten, 2024-08-05:
> Could this be related to commit b5c5384 and another manifestation of bug report #15821 ? (this time on send instead of receive?)

Is seems to be related to https://github.com/openwrt/openwrt/commit/b5c53848c3792355a2dd4ff0e06bf4d230e1f5fb, but #15821 is a different issue.

ErwanMAS, 2024-08-05:
I have a crash that involve vxlan and wifi and bridge  . So i re-build  in a viritualized context .  And now i have a crash that produce a similar trace that  this report ( https://github.com/openwrt/openwrt/issues/16026  ) 

The crash is almost instant when using virtio device .

ErwanMAS, 2024-08-09:
@schuettecarsten  so in my case, i use vxlan ( tunnel ip over udp ) .

On x86_64 and on ramips/mt7621 ( EA7500v2 )  , i don't have anymore crash with the patch . 

So this patch fix my 2 differents crash from https://github.com/openwrt/openwrt/issues/16026

on x86_64 i saw on dmesg this message
```
[  119.212844] eth0: bad gso: type: 1, size: 36
[  120.909823] eth0: bad gso: type: 1, size: 36
[  120.910390] eth0: bad gso: type: 1, size: 36
```

BrainSlayer, 2024-08-15:
read this. this describes the exakt occurence of the crash and also includes a patch proposal. its a upstream bug https://lore.kernel.org/lkml/65f5a2e1ed02e_6ef3e29463@willemb.c.googlers.com.notmuch/T/

ErwanMAS, 2024-08-15:
@nbd168 did you see this https://lore.kernel.org/lkml/65f5a2e1ed02e_6ef3e29463@willemb.c.googlers.com.notmuch/T/ ?

nbd168, 2024-08-15:
Please test if this commit from my staging tree helps:
https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=3c9b7ee55123a22965dd87f27c9864f675e4ccd5

nbd168, 2024-08-15:
Please also try this one: https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=f003e0b8ce7b4ea0210bfbedcd9a405b216a3322

ErwanMAS, 2024-08-16:
@nbd168 FYI  https://github.com/openwrt/openwrt/issues/16026#issuecomment-2292901886 

That's work , will run more intensive tests this week-end

BrainSlayer, 2024-08-16:
same for me @ErwanMAS consider that this fix nbd made has mainly a effect if you are using tunnels like wireguard, ovpn dco, vxlan etc. so make sure that your original configuration is still based on such things

nbd168, 2024-08-16:
Pushed the fixes to master. Please let me know if this issue re-appears and post updated logs for me.

BrainSlayer, 2024-08-16:
@nbd168 still crashes. no change

unterd0g, 2024-08-16:
(@nbd168 appears to have fixed #15821)

BrainSlayer, 2024-08-19:
@unterd0g okay for you the problem might be solved. the issue on my side which still triggered the bug is a corner case which doesnt belong to openwrt at all.

unterd0g, 2024-08-19:
@BrainSlayer reported here that the patch set solved my issue (#15821) because @nbd168 only introduced the patches in this issue. Therefore, even if the intended issue is not fixed, something else got fixed in the process _(i.e., please do not start reverting the commits without having that in mind)_.

I still hope this issue (and others related to GSO/GRO) can be solved.

nbd168, 2024-09-26:
Fixed in 774badd8a840be9888583266e12ba9d267201ee9