openwrt/openwrt #16071: kmod-sched-bpf missing act_bpf.ko

hack3ric, 2024-08-05:
### Describe the bug

When building OpenWrt from main with BPF and DEBUG_INFO_BTF enabled, kmod-sched-bpf failed to build because act_bpf.ko was missing. However, cls_bpf.ko exists and builds successfully.

```
ERROR: module '/home/hacker/Projects/3rdparty/openwrt/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/linux-6.6.43/net/sched/act_bpf.ko' is missing.
make[3]: *** [modules/netsupport.mk:807: /home/hacker/Projects/3rdparty/openwrt/bin/targets/armsr/armv8/packages/kmod-sched-bpf_6.6.43-r1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/hacker/Projects/3rdparty/openwrt/package/kernel/linux' 
time: package/kernel/linux/compile#1.56#0.22#1.76
    ERROR: package/kernel/linux failed to build.
```

### OpenWrt version

d604b2699b6015f99ff194b34f532958bde7d442

### OpenWrt release

d604b2699b6015f99ff194b34f532958bde7d442

### OpenWrt target/subtarget

armsr/armv8

### Device

N/A

### Image kind

Self-built image

### Steps to reproduce

1. Clone the project
2. Apply the diffconfig below
3. `make -j$(nproc)`

### Actual behaviour

Image building failed at Linux and kmod-sched-bpf

### Expected behaviour

Image builds successfully

### Additional info

`mimic` and `kmod-mimic` in the following diffconfig refers to my own package, which depends on kmod-sched-bpf.

### Diffconfig

```text
CONFIG_TARGET_armsr=y
CONFIG_TARGET_armsr_armv8=y
CONFIG_TARGET_armsr_armv8_DEVICE_generic=y
CONFIG_DEVEL=y
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH=""
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_DWARVES=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_KERNEL_DEBUG_INFO_BTF=y
CONFIG_KERNEL_DEBUG_INFO_BTF_MODULES=y
# CONFIG_KERNEL_DEBUG_INFO_REDUCED is not set
CONFIG_KERNEL_MODULE_ALLOW_BTF_MISMATCH=y
CONFIG_PACKAGE_kmod-mimic=y
CONFIG_PACKAGE_kmod-sched-bpf=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libffi=y
CONFIG_PACKAGE_mimic=y
CONFIG_PACKAGE_zlib=y
CONFIG_USE_LLVM_HOST=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hack3ric, 2024-08-07:
Then it should be added to sched-bpf's dependencies? If it is I'll submit a PR.

grische, 2025-07-27:
Is this still a problem?