openwrt/openwrt #16081: IPV6 MAP-E, mapcalc binary bogus on aarch64

leo5593, 2024-08-05:
### Describe the bug

When using full stack IPV4 MAP-E, mapcalc returns too many PORTSETS with port above 65535, see below.

When I move from Linksys WRT1900ACS to GL.iNet GL-MT3000, I have my MAP-E configure manualy to get the full stack IPV4 given by my provider, this need some tweaks, but when applying them to my new router at first it seams it did work but in facts configuration script was still running taking way too much time, because of the way too long variable PORTSETS.

MAP-E settings recevied by the provider with 1/4 of the port range was working normaly.

The only diference I found is **arm** and **aarch64**. I cross compiled the source file mapcalc.c and run it on both platform to be sure it is the same code. Since it is mosly bit shifting architecture matters. 

Changing the variable start and end to uint16_t could be a way since the represent ports and should be between 0 and 65535. I have tried it with my parameters and the behaviour was as on the 32bit system. But there is still  (1<<16) loop for nothing. Further investigation will be made.  

```
mapcalc wan6 type=map-e,ipv6prefix=2a01:***,prefix6len=60,ipv4prefix=82.***,prefix4len=32,psidlen=2,offset=16,ealen=32,br=2a01:***

RULE_1_FMR=0
RULE_1_EALEN=32
RULE_1_PSIDLEN=2
RULE_1_OFFSET=16
RULE_1_PREFIX4LEN=32
RULE_1_PREFIX6LEN=60
RULE_1_IPV4PREFIX=82.****
RULE_1_IPV6PREFIX=2a01:****
RULE_1_IPV6PD=2a01:****
RULE_1_PD6LEN=64
RULE_1_PD6IFACE=wan6
RULE_1_IPV6ADDR=2a01:***
RULE_BMR=1
RULE_1_IPV4ADDR=82.***
RULE_1_ADDR4LEN=32
RULE_1_PORTSETS='1-1073741824 2-1073741825 3-1073741826 4-1073741827 5-1073741828 6-1073741829 7-1073741
830 8-1073741831 9-1073741832 10-1073741833 11-1073741834 12-1073741835 13-1073741836 14-1073741837 15-1
073741838 16-1073741839 17-1073741840 18-1073741841 19-1073741842 20-1073741843 21-1073741844 22-1073741
845 23-1073741846 24-1073741847 25-1073741848 26-1073741849 27-1073741850 28-1073741851 29-1073741852 30
-1073741853 31-1073741854 32-1073741855 33-1073741856 34-1073741857 35-1073741858 36-1073741859 37-10737
41860 38-1073741861 39-1073741862 40-1073741863 41-1073741864 42-1073741865 43-1073741866 44-1073741867 
45-1073741868 46-1073741869 47-1073741870 48-1073741871 49-1073741872 50-1073741873 51-1073741874 52-107
3741875 53-1073741876 54-1073741877 55-1073741878 56-1073741879 57-1073741880 58-1073741881 59-107374188
2 60-1073741883 61-1073741884 62-1073741885 63-1073741886 64-1073741887 65-1073741888 66-1073741889 67-1
073741890 68-1073741891 69-1073741892 70-1073741893 71-1073741894 72-1073741895 73-1073741896 74-1073741

...
```

Code of [openwrt](https://github.com/openwrt/openwrt/tree/main)/[package](https://github.com/openwrt/openwrt/tree/main/package)/[network](https://github.com/openwrt/openwrt/tree/main/package/network)/[ipv6](https://github.com/openwrt/openwrt/tree/main/package/network/ipv6)/[map](https://github.com/openwrt/openwrt/tree/main/package/network/ipv6/map)/[src](https://github.com/openwrt/openwrt/tree/main/package/network/ipv6/map/src)/[mapcalc.c](https://github.com/openwrt/openwrt/blob/main/package/network/ipv6/map/src/mapcalc.c)

```

		if (psidlen > 0 && psid >= 0) {
			printf("RULE_%d_PORTSETS='", rulecnt);
			for (int k = (offset) ? 1 : 0; k < (1 << offset); ++k) {
				int start = (k << (16 - offset)) | (psid >> offset);
				int end = start + (1 << (16 - offset - psidlen)) - 1;

				if (start == 0)
					start = 1;

				if (start <= end)
					printf("%d-%d ", start, end);
			}
			printf("'\n");
		}

```

I checked the code above on a x86_64 and it gave the same output as aarch64. Couldn't try on x86. When casting the start and end variable on uint16_t, the behavior was the same as armhf.

Regards,

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT3000

### Image kind

Official downloaded image

### Steps to reproduce

MAP-E config

```
config interface 'MAP'
	option proto 'map'
	option maptype 'map-e'
	option ipaddr '82.***'
	option ip4prefixlen '32'
	option tunlink 'WAN6'
	option ealen '32'
	option psidlen '2'
	option encaplimit 'ignore'
	option mtu '1500'
	option peeraddr '2a01:***'
	option ip6prefix '2a01:***::'
	option ip6prefixlen '60'
	option offset '16'
```

You need to have aarch64 and an interface with a ipv6 prefix delegation and set it "ipv6prefix="

mapcalc "*" type=map-e,ipv6prefix=2a01:***,prefix6len=60,prefix4len=32,psidlen=2,offset=16,ealen=32,br=2a01::

### Actual behaviour

mapcalc binary output for variable PORTSETS bogus :

RULE_1_PORTSETS='1-1073741824 2-1073741825 3-1073741826 4-1073741827 5-1073741828 6-1073741829 7-1073741
830 8-1073741831 9-1073741832 10-1073741833 11-1073741834 12-1073741835 13-1073741836 14-1073741837  ...

### Expected behaviour

mapcalc binary with this ouput 

RULE_1_PORTSETS=''

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-06:
Close one of duplicate issues and retry with snapshot, x86/64 vm is valid 64bit too.

leo5593, 2024-08-06:
Hello,

This commands allows to reproduce easily :
mapcalc ethV6 type=map-e,prefix6len=64,ipv4prefix=1.1.1.1,prefix4len=32,psidlen=2,offset=16,ealen=32,br=2a01:b4::cafe,pdlen=60

Output on Linksys WRT1900ACS, armhf : 
```
RULE_1_FMR=0
RULE_1_EALEN=32
RULE_1_PSIDLEN=2
RULE_1_OFFSET=16
RULE_1_PREFIX4LEN=32
RULE_1_PREFIX6LEN=64
RULE_1_IPV4PREFIX=1.1.1.1
RULE_1_IPV6PREFIX=::
RULE_1_IPV6PD=::
RULE_1_PD6LEN=60
RULE_1_PD6IFACE=ethV6
RULE_1_IPV6ADDR=::101:101:0
RULE_BMR=1
RULE_1_IPV4ADDR=1.1.1.1
RULE_1_ADDR4LEN=32
RULE_1_PORTSETS=''
RULE_1_BR=2a01:b4::cafe
RULE_COUNT=1
```

I tried the snapshot version of x86-generic, x86-legacy and x86-64 on VMs and they all produce this output : 

```
RULE_1_FMR=0
RULE_1_EALEN=32
RULE_1_PSIDLEN=2
RULE_1_OFFSET=16
RULE_1_PREFIX4LEN=32
RULE_1_PREFIX6LEN=64
RULE_1_IPV4PREFIX=1.1.1.1
RULE_1_IPV6PREFIX=::
RULE_1_IPV6PD=::
RULE_1_PD6LEN=60
RULE_1_PD6IFACE=ethV6
RULE_1_IPV6ADDR=::101:101:0
RULE_BMR=1
RULE_1_IPV4ADDR=1.1.1.1
RULE_1_ADDR4LEN=32
RULE_1_PORTSETS='1-1073741824 2-1073741825 3-1073741826 4-1073741827 5-1073741828 6-1073741829 7-1073741830 8-1073741831 9-1073741832 10-1073741833 11-1073741834 12-1073741835 13-1073741836 14-1073741837 15-1073741838 16-1073741839 17-1073741840 18-1073741841 19-1073741842 20-1073741843 21-1073741844 22-1073741845 23-1073741846 24-1073741847 25-1073741848 26-1073741849 27-1073741850 28-1073741851 29-1073741852 30-1073741853 31-1073741854 32-1073741855 33-1073741856 34-1073741857 35-1073741858 36-1073741859 37-1073741860 38-1073741861 39-1073741862 40-1073741863 41-1073741864 42-1073741865 43-1073741866 44-1073741867 45-1073741868 46-1073741869 47-1073741870 48-1073741871 49-1073741872 50-1073741873 51-1073741874 52-1073741875 53-1073741876 54-1073741877 55-1073741878 56-1073741879 57-1073741880 58-1073741881 59-1073741882 60-1073741883 61-1073741884 62-1073741885 63-1073741886 64-1073741887 65-1073741888 66-1073741889 67-1073741890 68-1073741891 69-1073741892 70-1073741893 71-1073741894 72-1073741895 73-1073741896 74-1073741897 75-1073741898 76-1073741899 77-1073741900 78-1073741901 79-1073741902 80-1073741903 81-1073741904 82-1073741905 83-1073741906 84-1073741907 85-1073741908 86-1073741909 87-1073741910 88-1073741911 89-1073741912 90-1073741913 91-1073741914 92-1073741915 93-1073741916 94-1073741917 95-1073741918 96-1073741919 97-1073741920 98-1073741921 99-1073741922 100-1073741923 101-1073741924 102-1073741925 103-1073741926 104-1073741927 105-1073741928 106-1073741929 107-1073741930 108-1073741931 109-1073741932 110-1073741933 111-1073741934 112-1073741935 113-1073741936 114-1073741937 115-1073741938 116-1073741939

...

65531-1073807354 65532-1073807355 65533-1073807356 65534-1073807357 65535-1073807358 '
RULE_1_BR=2a01:b4::cafe
RULE_COUNT=1

```

So with this tests above, only **armhf** does not produce the kind of overflow seen with other architecture.

leo5593, 2024-08-06:
My bad parameters may not supposed to be used this way.

This test gives a working configuration :
mapcalc ethV6 type=map-e,prefix6len=60,ipv4prefix=1.1.1.1,prefix4len=32,br=2a01:b4::cafe,pdlen=60

There is still a kind of buffer overflow possible on some architecture.

Regards,

brada4, 2024-08-06:
Not buffer overflow but signed vs unsigned integer mismatch and overflow.