openwrt/openwrt #16083: The WAN functionality of the JCG-Q20 router is operational in version 23.05.3, but it does not work in the main branch

liudf0716, 2024-08-06:
### Describe the bug

23.05.3's dmesg
```
login as: root

BusyBox v1.36.1 (2024-03-22 22:09:42 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.3, r23809-234f1a2efa
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# dmesg
[    0.000000] Linux version 5.15.150 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23809-234f1a2efa) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Fri Mar 22 22:09:42 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is JCG Q20
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s17808 r8192 d23152 u49152
[    0.000000] pcpu-alloc: s17808 r8192 d23152 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00043100
[    0.000000] Readback ErrCtl register=00043100
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249104K/262144K available (7303K kernel code, 629K rwdata, 880K rodata, 1224K init, 225K bss, 13040K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008050] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066219] pid_max: default: 32768 minimum: 301
[    0.071638] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078850] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.090500] rcu: Hierarchical SRCU implementation.
[    0.095988] smp: Bringing up secondary CPUs ...
[    0.101397] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.101425] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.101440] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.101484] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.160772] Synchronize counters for CPU 1: done.
[    0.193370] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.193393] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.193404] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[... 575 lines cut ...]
When the JCG-Q20 uses the main firmware, the WAN cannot obtain an IP address from the DHCP server. Even with a static IP address, the network does not function properly.

### OpenWrt version

r27076-d604b2699b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

JCG Q20

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

coastdefence, 2024-08-06:
I also found the same problem，Looking forward to resolving it

brada4, 2024-08-06:
What negotiation parameter does `ethtool wan` show?

liudf0716, 2024-08-06:
@brada4
As the dmesg information from 23.05.3 shows, the other side has negotiated a speed of 100Mbps. It should not be gigabit.

brada4, 2024-08-06:
What is reported about autonegotiation?
Ie (f both ends think gigabit then gigabit it is...

DragonBluep, 2024-09-30:
I have the same issue on my MT7621 based `H3C TX1806`. When I connect WAN port to a 100 Mbps port modem, it cannot establish a stable connection. The log output shows the WAN port keeps link up and link down.
```
Mon Sep 30 10:13:22 2024 daemon.notice netifd: Network device 'wan' link is down
Mon Sep 30 10:13:22 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Sep 30 10:13:22 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Mon Sep 30 10:13:23 2024 daemon.notice netifd: wan (3835): udhcpc: received SIGTERM
Mon Sep 30 10:13:23 2024 daemon.notice netifd: Interface 'wan6' is now down
Mon Sep 30 10:13:23 2024 daemon.notice netifd: wan (3835): udhcpc: entering released state
Mon Sep 30 10:13:23 2024 daemon.notice netifd: wan (3835): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Mon Sep 30 10:13:23 2024 daemon.notice netifd: Interface 'wan' is now down
Mon Sep 30 10:13:24 2024 daemon.warn odhcpd[1517]: No default route present, overriding ra_lifetime to 0!
Mon Sep 30 10:13:25 2024 daemon.info dnsmasq[2574]: read /etc/hosts - 12 names
Mon Sep 30 10:13:25 2024 daemon.info dnsmasq[2574]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Mon Sep 30 10:13:25 2024 daemon.info dnsmasq[2574]: read /tmp/hosts/odhcpd - 2 names
Mon Sep 30 10:13:25 2024 daemon.info dnsmasq-dhcp[2574]: read /etc/ethers - 0 addresses
Mon Sep 30 10:13:29 2024 daemon.notice netifd: Network device 'wan' link is up
Mon Sep 30 10:13:29 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Mon Sep 30 10:13:29 2024 daemon.notice netifd: Interface 'wan' is setting up now
Mon Sep 30 10:13:29 2024 daemon.notice netifd: Interface 'wan6' has link connectivity
Mon Sep 30 10:13:29 2024 daemon.notice netifd: Interface 'wan6' is setting up now
Mon Sep 30 10:13:29 2024 kern.info kernel: [   72.113506] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control off
Mon Sep 30 10:13:29 2024 daemon.notice netifd: wan (4246): udhcpc: started, v1.36.1
Mon Sep 30 10:13:29 2024 daemon.notice netifd: wan (4246): udhcpc: broadcasting discover
Mon Sep 30 10:13:31 2024 kern.info kernel: [   74.193384] mtk_soc_eth 1e100000.ethernet wan: Link is Down
Mon Sep 30 10:13:31 2024 daemon.notice netifd: Network device 'wan' link is down
Mon Sep 30 10:13:31 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Sep 30 10:13:31 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Mon Sep 30 10:13:31 2024 daemon.notice netifd: wan (4246): udhcpc: received SIGTERM
Mon Sep 30 10:13:31 2024 daemon.notice netifd: wan (4246): udhcpc: entering released state
Mon Sep 30 10:13:31 2024 daemon.notice netifd: wan (4246): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Mon Sep 30 10:13:31 2024 daemon.notice netifd: Interface 'wan' is now down
Mon Sep 30 10:13:31 2024 daemon.notice netifd: Interface 'wan6' is now down
Mon Sep 30 10:13:32 2024 daemon.warn odhcpd[1517]: No default route present, overriding ra_lifetime to 0!
Mon Sep 30 10:13:33 2024 daemon.notice netifd: Network device 'wan' link is up
Mon Sep 30 10:13:33 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Mon Sep 30 10:13:33 2024 daemon.notice netifd: Interface 'wan' is setting up now
Mon Sep 30 10:13:33 2024 daemon.notice netifd: Interface 'wan6' has link connectivity
Mon Sep 30 10:13:33 2024 daemon.notice netifd: Interface 'wan6' is setting up now
Mon Sep 30 10:13:33 2024 kern.info kernel: [   76.273549] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control off
Mon Sep 30 10:13:33 2024 daemon.info dnsmasq[2574]: read /etc/hosts - 12 names
Mon Sep 30 10:13:33 2024 daemon.info dnsmasq[2574]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Mon Sep 30 10:13:33 2024 daemon.notice netifd: wan (4443): udhcpc: started, v1.36.1
Mon Sep 30 10:13:33 2024 daemon.info dnsmasq[2574]: read /tmp/hosts/odhcpd - 2 names
Mon Sep 30 10:13:33 2024 daemon.info dnsmasq-dhcp[2574]: read /etc/ethers - 0 addresses
Mon Sep 30 10:13:33 2024 daemon.notice netifd: wan (4443): udhcpc: broadcasting discover
Mon Sep 30 10:13:34 2024 kern.info kernel: [   77.313390] mtk_soc_eth 1e100000.ethernet wan: Link is Down
Mon Sep 30 10:13:34 2024 daemon.notice netifd: Network device 'wan' link is down
Mon Sep 30 10:13:34 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Sep 30 10:13:34 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Mon Sep 30 10:13:34 2024 daemon.notice netifd: wan (4443): udhcpc: received SIGTERM
Mon Sep 30 10:13:34 2024 daemon.notice netifd: wan (4443): udhcpc: entering released state
Mon Sep 30 10:13:34 2024 daemon.notice netifd: wan (4443): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Mon Sep 30 10:13:34 2024 daemon.notice netifd: Interface 'wan' is now down
Mon Sep 30 10:13:34 2024 daemon.notice netifd: Interface 'wan6' is now down
Mon Sep 30 10:13:35 2024 daemon.warn odhcpd[1517]: No default route present, overriding ra_lifetime to 0!
Mon Sep 30 10:13:36 2024 daemon.info dnsmasq[2574]: read /etc/hosts - 12 names
Mon Sep 30 10:13:36 2024 daemon.info dnsmasq[2574]: read /tmp/hosts/dhcp.cfg01411c - 4 names
[... cut ...]
Mon Sep 30 10:13:45 2024 daemon.info dnsmasq[2574]: read /etc/hosts - 12 names
Mon Sep 30 10:13:45 2024 daemon.info dnsmasq[2574]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Mon Sep 30 10:13:45 2024 daemon.info dnsmasq[2574]: read /tmp/hosts/odhcpd - 2 names
Mon Sep 30 10:13:45 2024 daemon.info dnsmasq-dhcp[2574]: read /etc/ethers - 0 addresses
```

Muxing wan port from gamc1 to the gmac0 switch can fix this issue. Things are still okay in February this year. Hi @arinc9, do you have any idea about this? I can help build&test if you have some patch.

Regards.

arinc9, 2024-09-30:
> When the JCG-Q20 uses the main firmware, the WAN cannot obtain an IP address from the DHCP server. Even with a static IP address, the network does not function properly.

That makes sense as the the link partner cannot operate in 1Gbps, so you won't have a functioning link.

> I have the same issue on my MT7621 based `H3C TX1806`.

@DragonBluep, yours doesn't look like the same issue. liudf0716 reports incorrect autonegotiation; `wan: Link is Up - 1Gbps/Full - flow control rx/tx` instead of `wan: Link is Up - 100Mbps/Full - flow control rx/tx`. Your issue is about the link constantly going down and up. And you have not mentioned whether or not this issue is there on 23.05.3.

> Muxing wan port from gamc1 to the gmac0 switch can fix this issue.

As `target/linux/ramips/dts/mt7621_h3c_tx180x.dtsi` describes that switch phy4 is connected to SoC gmac1, do you mean to describe switch phy0 being connected to SoC gmac1 instead? If so, give it a try. Also describe the wan port (switch phy4) under the switch ports node. Test if you see the down-up behaviour on the ports that are affine to switch phy4 and switch phy0. And make sure the MEDIATEK_GE_PHY driver is enabled. Reason:

https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/?id=4223f8651287

Regarding liudf0716's issue: Currently, `target/linux/ramips/dts/mt7621_jcg_q20.dts` describes that the wan port's phy (switch phy0) is connected to SoC gmac1. Incorrect autonegotiation is quite odd; I see multiple scenarios as to why we're experiencing this issue. Considering this issue has appeared on a newer kernel as liudf0716 reports, it is likely a driver issue and not something such as the switch phy reporting wrong abilities of the link partner using autonegotiation.

In any case, let's make sure that the link partner is not reporting 1Gbps capability. @liudf0716, connect the wan port to a computer that can run ethtool, limit the speed capability to 100M if needed, and test if you still see the device with the MT7530 switch reporting 1Gbps link negotiated.

DragonBluep, 2024-09-30:
I found this issue on today's master branch.  I‘ll take a look on openwrt v23 later.

> As `target/linux/ramips/dts/mt7621_h3c_tx180x.dtsi` describes that switch phy4 is connected to SoC gmac1, do you mean to describe switch phy0 being connected to SoC gmac1 instead?

No. phy0 and gamc1 are disabled on my working build. The modified dtsi file:
```
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>
#include <dt-bindings/leds/common.h>

/ {
	aliases {
		label-mac-device = &gmac1;
		led-boot = &led_status_amber;
		led-failsafe = &led_status_green;
		led-running = &led_status_green;
		led-upgrade = &led_status_amber;
	};

	chosen {
		bootargs = "console=ttyS0,115200 root=/dev/fit0 rootwait";
		bootargs-override = "console=ttyS0,115200 rootfstype=squashfs,ubifs root=/dev/fit0 rootwait";
		rootdisk = <&ubi_rootdisk>;
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 13 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RESTART>;
		};

		wps {
			label = "wps";
			gpios = <&gpio 14 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_WPS_BUTTON>;
		};
	};

	leds {
		compatible = "gpio-leds";

		led_status_amber: led-amber {
			color = <LED_COLOR_ID_AMBER>;
			function = LED_FUNCTION_STATUS;
			gpios = <&gpio 15 GPIO_ACTIVE_LOW>;
		};

		led_status_green: led-green {
			color = <LED_COLOR_ID_GREEN>;
			function = LED_FUNCTION_STATUS;
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
		};
	};
};

&ethernet {
	pinctrl-0 = <&mdio_pins>, <&rgmii1_pins>;
};

&gmac0 {
	nvmem-cells = <&macaddr_ubootenv_ethaddr 1>;
	nvmem-cell-names = "mac-address";
};

&nand {
	status = "okay";

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		partition@0 {
			label = "u-boot";
			reg = <0x0000000 0x0080000>;
			read-only;
		};

		partition@80000 {
			compatible = "fixed-partitions";
			label = "u-boot-env";
			reg = <0x0080000 0x0080000>;
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				compatible = "u-boot,env";
				label = "env-data";
				reg = <0x00000 0x20000>;

				macaddr_ubootenv_ethaddr: ethaddr {
					#nvmem-cell-cells = <1>;
				};
			};
		};

		partition@100000 {
			label = "exp";
			reg = <0x0100000 0x0080000>;
			read-only;
		};

		partition@180000 {
			label = "factory";
			reg = <0x0180000 0x0080000>;
			read-only;

			nvmem-layout {
				compatible = "fixed-layout";
				#address-cells = <1>;
				#size-cells = <1>;

				eeprom_factory_0: eeprom@0 {
					reg = <0x0 0xe00>;
				};
			};
		};

		partition@200000 {
			compatible = "linux,ubi";
			label = "ubi";
			reg = <0x0200000 0x7e00000>;

			volumes {
				ubi_rootdisk: ubi-volume-fit {
					volname = "fit";
				};
			};
		};
	};
};

&pcie {
	status = "okay";
};

&pcie1 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,disable-radar-background;
		nvmem-cells = <&eeprom_factory_0>;
		nvmem-cell-names = "eeprom";
		#address-cells = <1>;
		#size-cells = <0>;

		band@0 {
			reg = <0>;
			nvmem-cells = <&macaddr_ubootenv_ethaddr 2>;
			nvmem-cell-names = "mac-address";
			mac-address-bitwise-clear = <0x0000 0x00f00000>;
			mac-address-bitwise-set = <0x0200 0x00100000>;
		};

		band@1 {
			reg = <1>;
			nvmem-cells = <&macaddr_ubootenv_ethaddr 2>;
			nvmem-cell-names = "mac-address";
			mac-address-bitwise-clear = <0x0000 0x00f00000>;
			mac-address-bitwise-set = <0x0200 0x00700000>;
		};

		led {
			status = "disabled";
		};
	};
};

&state_default {
	gpio {
		groups = "jtag";
		function = "gpio";
	};
};

&switch0 {
	ports {
		port@1 {
			status = "okay";
			label = "lan1";
		};

		port@2 {
			status = "okay";
			label = "lan2";
		};

		port@3 {
			status = "okay";
			label = "lan3";
		};

		port@4 {
			status = "okay";
			label = "wan";

			nvmem-cells = <&macaddr_ubootenv_ethaddr 0>;
			nvmem-cell-names = "mac-address";
		};
	};
};

&xhci {
	status = "disabled";
};

```

> Also describe the wan port (switch phy4) under the switch ports node. Test if you see the down-up behaviour on the ports that are affine to switch phy4 and switch phy0.

Already done on my custom build. I‘m pretty sure WAN port is port4. I can see this log when I plug in wan cable:
`[  463.739090] mt7530-mdio mdio-bus:1f wan: Link is Up - 100Mbps/Full - flow control off`
[image]
Everything works well with this modified dts.

It looks like a link detection issue.

DragonBluep, 2024-09-30:
Tested on `ASUS RT-AC1200GU`, aka `ASUS RT-AC57U`. Based on today's master branch. The same auto link up and link down issue. I can reproduce it on my Windows computer (NIC: PCI\VEN_10EC&DEV_8168&SUBSYS_012310EC).
```
# Using auto speed on my computer eth port:
[  527.244998] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
# The link status didn't change and it won't link down automatically. Then I set computer eth port
# speed to force 100 Mbps full and re-plug the cable:
[  638.524911] mtk_soc_eth 1e100000.ethernet wan: Link is Down
# Bingo! It will link down and up every few seconds:
[  719.645033] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  720.684937] mtk_soc_eth 1e100000.ethernet wan: Link is Down
[  724.844992] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  725.884927] mtk_soc_eth 1e100000.ethernet wan: Link is Down
[  756.045021] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  757.084920] mtk_soc_eth 1e100000.ethernet wan: Link is Down
[  784.125028] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  785.164932] mtk_soc_eth 1e100000.ethernet wan: Link is Down
```

DragonBluep, 2024-09-30:
> This is a different issue than what is reported here by liudf0716.

I opened another issue to track it and cc to you.

> 
> Looking up the PCI ID, your link partner is RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller. I had a similar issue with the Killer E2500 Gigabit Ethernet Controller; the down-up issue appears on a 1Gbps negotiated link. After about 10 down-ups, the link speed becomes negotiated to 100Mbps, with which the down-up issue does not appear. See if you have your issue with another ethernet controller. With that said, I've got nothing else to add to your issue.

I waited several minutes but things didn't get better. Tested on my RTL8168 NIC computer and a 100 Mbps modem. On OpenWrt v23, it can always establish a stable connect on first attempt.