openwrt/openwrt #16096: Commit a37650821644 breaks Zyxel GS1900-24E

RussellSenior, 2024-08-06:
### Describe the bug

Since commit a37650821644, zyxel,gs1900-24e doesn't pass traffic. Reverting the commit on top of recent master branch commit b733b6acb5, and traffic passes again.

### OpenWrt version

r27082-b733b6acb5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

realtek/rtl838x

### Device

ZyXEL GS1900-24E

### Image kind

Self-built image

### Steps to reproduce

Install, in my case the bridge switch (sw) is configured to request DHCP, it doesn't get it, so boot with a serial console and notice that it doesn't get a DHCP lease and can't ping any other addresses.

### Actual behaviour

With the following /etc/config/network:
`config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config device 'switch'
        option name 'sw'
        option type 'bridge'
        option macaddr '5c:6a:80:xx:yy:zz'

config bridge-vlan 'pub_vlan'
        option device 'sw'
        option vlan '100'
        option ports 'lan1 lan2 lan3 lan4 lan5 lan6 lan7 lan8 lan9 lan10 lan11 lan12 lan13 lan14 lan15 lan16 lan17 lan18 lan19 lan20 lan21 lan22 lan23 lan24'

config device
        option name 'sw.100'
        option macaddr '5c:6a:80:xx:yy:zz'

config interface 'pub'
        option device 'sw.100'
        option proto 'dhcp'

config interface 'pub6'
        option device '@pub'
        option proto 'dhcpv6'
`
fails to get a lease, starting with commit a37650821644, and when the commit is reverted it starts getting a lease again.

### Expected behaviour

I expect sw.100 to pass traffic.

### Additional info

I bisected to locate the problematic commit, then tried reverting on top of recent master branch and that fixed my problem

### Diffconfig

```text
CONFIG_TARGET_realtek=y
CONFIG_TARGET_realtek_rtl838x=y
CONFIG_TARGET_realtek_rtl838x_DEVICE_zyxel_gs1900-24e=y
CONFIG_DEVEL=y
CONFIG_BUILD_LOG=y
CONFIG_PACKAGE_diffutils=y
# CONFIG_PACKAGE_firewall4 is not set
# CONFIG_PACKAGE_jansson is not set
# CONFIG_PACKAGE_kmod-crypto-crc32c is not set
# CONFIG_PACKAGE_kmod-crypto-hash is not set
# CONFIG_PACKAGE_kmod-lib-crc32c is not set
# CONFIG_PACKAGE_kmod-nf-conntrack is not set
# CONFIG_PACKAGE_kmod-nf-conntrack6 is not set
# CONFIG_PACKAGE_kmod-nf-flow is not set
# CONFIG_PACKAGE_kmod-nf-log is not set
# CONFIG_PACKAGE_kmod-nf-log6 is not set
# CONFIG_PACKAGE_kmod-nf-nat is not set
# CONFIG_PACKAGE_kmod-nf-reject is not set
# CONFIG_PACKAGE_kmod-nf-reject6 is not set
# CONFIG_PACKAGE_kmod-nfnetlink is not set
# CONFIG_PACKAGE_kmod-nft-core is not set
# CONFIG_PACKAGE_kmod-nft-fib is not set
# CONFIG_PACKAGE_kmod-nft-nat is not set
# CONFIG_PACKAGE_kmod-nft-offload is not set
# CONFIG_PACKAGE_libmnl is not set
CONFIG_PACKAGE_libnetsnmp=y
# CONFIG_PACKAGE_libnftnl is not set
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpci=y
CONFIG_PACKAGE_libpcre2=y
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_snmpd=y
CONFIG_PACKAGE_tcpdump=y
# CONFIG_PACKAGE_ucode-mod-ubus is not set
# CONFIG_PACKAGE_ucode-mod-uci is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jwh, 2024-08-07:
https://github.com/openwrt/openwrt/issues/14440

I didn't get any response, there just aren't enough people working on realtek targets, but reverting both of those patches resolves the issue.

0xharshal, 2024-08-08:
Hey @bmork I am the author of this commit, I would like to help out here, I don't have any rtl83xx device, but this commit is necessary for rtl930x and rtl931x devices on which I am working, these devices depend on bootloader to initialize phy and mac serdes and possibly other things.

If you have the same or other rtl83xx based device, can you check if following patch resolves the issue without reverting this commit,

```
diff --git a/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/dsa.c b/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/dsa.c
index 9eb444515f..c2fcacb2c6 100644
--- a/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/dsa.c
+++ b/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/dsa.c
@@ -125,7 +125,8 @@ static void rtl83xx_vlan_setup(struct rtl838x_switch_priv *priv)
        info.hash_uc_fid = false;       /* Do not build the L2 lookup hash with FID, but VID */
        info.hash_mc_fid = false;       /* Do the same for Multicast packets */
        info.profile_id = 0;            /* Use default Vlan Profile 0 */
-       info.tagged_ports = 0;          /* Initially no port members */
+       // info.tagged_ports = 0;               /* Initially no port members */
+       info.tagged_ports = BIT_ULL(priv->cpu_port); /* Initially only cpu port as member */
        if (priv->family_id == RTL9310_FAMILY_ID) {
                info.if_id = 0;
                info.multicast_grp_mask = 0;
```

bmork, 2024-08-08:
Tested now. Makes no difference unfortunately. 

Didn't really expect so either. If the problem had been related to the VID 1 configuration on either CPU or the untagged switch ports, then we would have been able to work around the issue in userspace by using "bridge vlan add/del" to reinitialise the table entries.

Note that the problem sending untagged packets from the CPU isn't limited to VID 1.  I can reproduce it with any untagged vlan.

I believe the problem really is the missing "VID 0" setup.  The assumption in your commit message is incorrect.  Configuring VID 0 is definitely NOT a no-op on these switches.  The vlan tables are 0 indexed and we do need to initialise index 0 too.  I don't know how the switches work internally, but I believe a reasonable assumption is that VID 0 is looked up for untagged packets from the CPU.  So if the port isn't member of VID 0 then the packet won't go out.

To be frank, I do not understand your justification for this change.  I do note that it seems to be comething completely different from the commit message.  That's not OK. Anyway, if this hack works around some missing initialision on rtl93xx then that's just a strange and unexpected side effect. You should not depend on something like that. I believe it makes more sense for you to debug the root cause of your problem.  Maybe you can even get the rtl93xx to work without the helping hand from bootloader if you find the real issues instead of papering over the symptoms with unexplainable code.

bmork, 2024-08-08:
Could be wrong.  I don't have any rtl93xx hardware to test the theory on.  But this code looks suspicious to me:
```
static void rtl931x_vlan_set_tagged(u32 vlan, struct rtl838x_vlan_info *info)
{
	u32 v, w, x, y;
	/* Access VLAN table (1) via register 0 */
	struct table_reg *r = rtl_table_get(RTL9310_TBL_0, 3);

	v = info->tagged_ports >> 25;
	w = (info->tagged_ports & 0x1fffff) << 7;
..
```

If you use any of the ports dropped from the bitmap, then I can understand how reinitialising vid 0 table breaks stuff.  Assuming the bootloader already set up this table correcly.

Not that I understand how that code can work for any other VIDs.  Ports 21-24 are always disabled?

0xharshal, 2024-08-09:
I agree with you, I am looking into it, I will check again and possibly fix the vlan table setup for rtl93xx

jwh, 2024-08-13:
@0xharshal sorry I haven't had time to test, but afaics the 838x bootloader does not do any setup before booting, I don't have any 93xx devices to check with - it seems @bmork has already got a handle on this though

Neustradamus, 2024-08-21:
@RussellSenior, @jwh, @bmork, @0xharshal: Can you look here:
- https://github.com/openwrt/openwrt/pull/16204

plappermaul, 2024-08-22:
5.15 will create the following final VLAN state:

```
[   50.816767] rtl83xx_vlan_add() Untagged ports, VLAN 0: 0
[   50.822822] rtl83xx_vlan_add() Tagged ports, VLAN 0: 1500ff00

[   51.023290] rtl83xx_vlan_add() Untagged ports, VLAN 1: 1500ff00
[   51.030022] rtl83xx_vlan_add() Tagged ports, VLAN 1: 1500ff00
```

6.6 will create the following final VLAN state:

```
[   64.866492] rtl83xx_vlan_add() Untagged ports, VLAN 0: 0
[   64.872464] rtl83xx_vlan_add() Tagged ports, VLAN 0: 1500ff00

[   64.940660] rtl83xx_vlan_add() Untagged ports, VLAN 1: 500ff00
[   64.947352] rtl83xx_vlan_add() Tagged ports, VLAN 1: 1500ff00
```

where tagged ports means "all ports in the bridge"

Let VLAN 0 aside we can remark VLAN 1 setup on 6.6 produces a different result. Difference comes how rtl83xx_vlan_add() is called.

- 5.15: rtl83xx_vlan_add port 28, VLAN 1, flags 4 ---> BRIDGE_VLAN_INFO_UNTAGGED
- 6.6: tl83xx_vlan_add port 28, VLAN 1, flags 21 ---> CPU BRIDGE_VLAN_INFO_BRENTRY | BRIDGE_VLAN_INFO_MASTER

plappermaul, 2024-08-22:
Just to remember. Maybe something like this in rtl83xx_vlan_add() might help

```
static int rtl83xx_vlan_add(struct dsa_switch *ds, int port,
			    const struct switchdev_obj_port_vlan *vlan,
			    struct netlink_ext_ack *extack)
{
  struct dsa_port *dp = dsa_to_port(ds, port);
  ...
  if (dsa_port_is_cpu(dp)) {
    /* do CPU extra magic */
  }
  ...
}
```

plappermaul, 2024-08-22:
Maybe try these three additions:

```
static int rtl83xx_vlan_add(struct dsa_switch *ds, int port,
			    const struct switchdev_obj_port_vlan *vlan,
			    struct netlink_ext_ack *extack)
{
	struct rtl838x_vlan_info info;
	struct rtl838x_switch_priv *priv = ds->priv;
	struct dsa_port *dp = dsa_to_port(ds, port); <-- ADD
	int err;

	pr_info("%s port %d, VLAN %d, flags %x\n",
		__func__, port, vlan->vid, vlan->flags);

	if(!vlan->vid) return 0;

	if (vlan->vid > 4095) {
		dev_err(priv->dev, "VLAN out of range: %d", vlan->vid);
		return -ENOTSUPP;
	}

	err = rtl83xx_vlan_prepare(ds, port, vlan);
	if (err)
		return err;

	mutex_lock(&priv->reg_mutex);

	if ((vlan->flags & BRIDGE_VLAN_INFO_PVID) || dsa_port_is_cpu(dp)) <-- ADD
		rtl83xx_vlan_set_pvid(priv, port, vlan->vid);
	else if (priv->ports[port].pvid == vlan->vid)
		rtl83xx_vlan_set_pvid(priv, port, 0);

	/* Get port memberships of this vlan */
	priv->r->vlan_tables_read(vlan->vid, &info);

	/* new VLAN? */
	if (!info.tagged_ports) {
		info.fid = 0;
		info.hash_mc_fid = false;
		info.hash_uc_fid = false;
		info.profile_id = 0;
	}

	/* sanitize untagged_ports - must be a subset */
	if (info.untagged_ports & ~info.tagged_ports)
		info.untagged_ports = 0;

	info.tagged_ports |= BIT_ULL(port);
	if ((vlan->flags & BRIDGE_VLAN_INFO_UNTAGGED) || dsa_port_is_cpu(dp)) <-- ADD
		info.untagged_ports |= BIT_ULL(port);
	else
		info.untagged_ports &= ~BIT_ULL(port);

```

bmork, 2024-08-23:
> if ((vlan->flags & BRIDGE_VLAN_INFO_PVID) || dsa_port_is_cpu(dp)) <-- ADD
> 		rtl83xx_vlan_set_pvid(priv, port, vlan->vid);

I can test like anyone else, but this looks odd to me.  There can obviously only ever be one PVID on a port.  Not entirely convinced that we want any PVID on the CPU port. But we most certainly don't want to make it the last added vlan.

bmork, 2024-08-23:
> ```
> [   51.023290] rtl83xx_vlan_add() Untagged ports, VLAN 1: 1500ff00
..
> [   64.940660] rtl83xx_vlan_add() Untagged ports, VLAN 1: 500ff00 (all ports untagged while CPU tagged)
> ```

Nice finding.  Looks like a reasonable explanation.  So something needs vid 1 confgured as pvid and/or untagged on the CPU port? 

Only problem is that the 6.6 output is the correct one.  That's exactly what we told the switch to configure.  The 5.15 version did something else, which is a bug.  If we wanted vid to be untagged or pvid then we would have configured it like so.

I think a good test on rtl93xx would be to see if the issues can be worked around in userspace by simply configuring vid 1 as untagged+pvid on the cpu port.

I wonder if this problem is somehow related to commit e691e2b302d9 ("rtl83xx: dsa: reset PVID to 1 instead of 0"), which also is very strange but otherwise harmless.  Maybe revert that as well and see if it makes the rtl93xx work?

Why do we ever want to receive untagged traffic on the CPU port?  Where does that come from?  And why should we send anything into the switch without a tag?

plappermaul, 2024-08-23:
Stupid me cannot help here. I do not even know how eth0 (CPU port I guess) and switch.1 (lan1-lan10 bridge VLAN 1) should interact.

plappermaul, 2024-08-26:
I cannot make progress without feedback. Maybe someone can clarify how VLAN config should work and what we would expect. To make it as simple as possible here a sample target config:

- lan1 & lan2 untagged internal VID 6 -> CPU port not connected
- lan 3 & lan4 untagged internal VID 7 -> CPU port connected and reachable by IP 192.168.9.1
- lan 5 & lan6 tagged internal VID 8 -> CPU port not connected
- lan 7 & lan8  tagged internal VID 9 -> CPU connected and reachable by IP 192.168.10.1

What brctl commands are needed for that and what do we expect when looking at the interfaces.

plappermaul, 2024-08-27:
@bmork Do you think this is a reasonable switch configuration for the 4 different cases to test with?

```
config device 'switch'
        option name 'switch'
        option type 'bridge'
        option macaddr 'd8:ec:5e:5b:7d:a1'

config bridge-vlan 'vlan_1'
        option device 'switch'
        option vlan '1'
        option ports 'lan1:t lan2:t'

config bridge-vlan 'vlan_2'
        option device 'switch'
        option vlan '2'
        option ports 'lan3:t lan4:t'

config bridge-vlan 'vlan_3'
        option device 'switch'
        option vlan '3'
        option ports 'lan5 lan6'

config bridge-vlan 'vlan_4'
        option device 'switch'
        option vlan '4'
        option ports 'lan7 lan8'

config device
        option name 'switch.1'
        option macaddr 'd8:ec:5e:5b:7d:a1'

config device
        option name 'switch.2'
        option macaddr 'd8:ec:5e:5b:7d:a2'

config device
        option name 'switch.3'
        option macaddr 'd8:ec:5e:5b:7d:a3'

config device
        option name 'switch.4'
        option macaddr 'd8:ec:5e:5b:7d:a4'

config interface 'lan1'
        option device 'switch.1'
        option proto 'static'
        option ipaddr '192.168.2.71'
        option netmask '255.255.255.0'

config interface 'lan3'
        option device 'switch.3'
        option proto 'static'
        option ipaddr '192.168.7.71'
        option netmask '255.255.255.0'
```

```
root@OpenWrt:/bin# bridge vlan
port              vlan-id
lan1              1
lan2              1
lan3              2
lan4              2
lan5              3 PVID Egress Untagged
lan6              3 PVID Egress Untagged
lan7              4 PVID Egress Untagged
lan8              4 PVID Egress Untagged
switch            1
                  2
                  3
                  4
```

```
ifconfig
switch    Link encap:Ethernet  HWaddr D8:EC:5E:5B:7D:A1
          inet6 addr: fe80::daec:5eff:fe5b:7da1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:555 errors:0 dropped:0 overruns:0 frame:0
          TX packets:27 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:99280 (96.9 KiB)  TX bytes:3018 (2.9 KiB)

switch.1  Link encap:Ethernet  HWaddr D8:EC:5E:5B:7D:A1
          inet addr:192.168.2.71  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::daec:5eff:fe5b:7da1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:1006 (1006.0 B)

switch.3  Link encap:Ethernet  HWaddr D8:EC:5E:5B:7D:A3
          inet addr:192.168.7.71  Bcast:192.168.7.255  Mask:255.255.255.0
          inet6 addr: fe80::daec:5eff:fe5b:7da3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:555 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:99280 (96.9 KiB)  TX bytes:1006 (1006.0 B)
```

bmork, 2024-08-27:
Looks like a good base to me.  Maybe add one vlan with a mix of tagged and untagged ports?  Or maybe not.  Better keep the number of combinations low, or no one will bother testing.

Personally I don't have any dedicated test switches.  So I test in the environment my switches are installed in, hoping that it will cover some ground.  And I often do remote testing so the cabling is what it is.  Reconfiguring connected equipment mostly isn't an option.  I assume others will have similar problems.  Most OpenWrt users buy their equipement to use it, not primarily for testing.

plappermaul, 2024-08-29:
> I think a good test on rtl93xx would be to see if the issues can be worked around in userspace by simply configuring vid 1 as untagged+pvid on the cpu port.

How would one do that?

bmork, 2024-08-29:
Maybe somwthing like
```
bridge vlan del vid 1 dev switch
bridge vlan add vid 1 pvid untagged dev switch self
```
assuming the brigde dev is "switch".  And then configure the switch interface with IP etc instead of switch.1

plappermaul, 2024-08-29:
Please try latest version of https://github.com/openwrt/openwrt/pull/16204

Neustradamus, 2024-09-07:
Dear all, have you tested this PR?
- https://github.com/openwrt/openwrt/pull/16204

plappermaul, 2024-10-06:
@RussellSenior still an issue with current head?

RussellSenior, 2024-10-07:
> @RussellSenior still an issue with current head?

I just built/flashed r27695-dae6a871de and it seems to be working.

closed 2024-10-07 by RussellSenior

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442