openwrt/openwrt #16105: attandedsysupgrade writes extraneous serial port info to grub.cfg

Fairalbion, 2024-08-07:
### Describe the bug

Upgrading from SR 23.05.2 to 23.05.4 on vanilla x64 system. Boot process stopped because attendedsysupgrade appears to have newly written serial port console info to grub.cfg on a system that has no serial port (keyboard/HDMI only mini PC).

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

Default string Default string

### Image kind

Official downloaded image

### Steps to reproduce

Run attendedsysupgrade from luci interface. No advanced parameters. 

### Actual behaviour

System hung on boot. Plugging in monitor & keyboard showed router paused at GRUB menu awaiting input. Choosing default menuentry allowed boot process to continue normally.

[grub_cfg_serial.txt](https://github.com/user-attachments/files/16532823/grub_cfg_serial.txt)

### Expected behaviour

grub.cfg should not include serial port info on device where serial port device is absent. 

Apologies I do not have a copy of the grub.cfg file from before the upgrade to 23.05.4

### Additional info

Editing grub.cfg to remove the reference to the serial port allowed the system to boot normally & unattended.

Note this behavior did not occur on previous upgrade to 23.0.2, which had been done manually (uploading a vanilla x64 image via luci  /System/Backup Flash Firmware/Flash image)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-07:
It includes serial an fb consoles at once. No impact on grub timeouts.

Fairalbion, 2024-08-08:
Thanks. As noted, editing/commenting out the serial port references has allowed the system to now boot without intervention.
[grub_cfg_post-edit.txt](https://github.com/user-attachments/files/16547998/grub_cfg_post-edit.txt)